Phil1988/FreeDi #84: Helper

Ferropluss, 2025-01-18:
Phil1988, здравствуйте.
Для многих пользователей, тяжело проходит установка. 
Но есть огромное желание, использовать Ваш продукт.
Помогите, Всем разобраться.

1. Принтер не подключается по IP кабелю. Проблема? Как можно решить?
2. EMMS MKS оригинал - понятно. Формат Fat32. После образа остаток памяти форматировать Fat32?
3. После установки, какой файл boot можно редактировать для Wi-Fi?
4. Какие файлы есть для редактирования через Raspberry Pi, чтоб упростить задачи на X-6 v1.0.

Пожалуйста, ответьте. Ваш продукт, мне очень интересен, хочу довести дело до конца.

Phil1988, 2025-01-18:
Please use english mate :)

Translated:
> Phil1988, hello. For many users, installation is challenging. However, there is a strong desire to use your product. Please help everyone understand.
> 
> 1. The printer does not connect via IP cable. What is the problem? How can this be resolved?
> 2. Is it possible to set up the connection via a Raspberry Pi?
> 3. EMMS MKS original - understood. Format Fat32. After flashing the image, should the remaining memory also be formatted as Fat32?
> 4. After installation, which boot file can be edited for Wi-Fi configuration?
> 5. What files are available for editing through the Raspberry Pi to simplify tasks on the X-6 v1.0?
> 
> [image]
> 
> All files are available for editing.
> 
> Please respond. Your product is very interesting to me, and I want to see this project through to the end.

1. The problem is, that a wired ethernet connection is required, see [chapter requirements](https://github.com/Phil1988/FreeDi/wiki/Requirements) under **2. Wired connection**

2. I dont get this part. Stock printers dont include a Raspberry Pi. I am missing information to understand your setup.

3. What is "EMMS MKS original"? If you are talking about the EMMC flash, just flash the latest matching image from the release section. If your disk is larger then the provided 8GB, see chapter [Expanding the filesystem](https://github.com/Phil1988/FreeDi/wiki/Expanding-the-filesystem-(for-larger-storage-devices))

4. Please start the install script to automatically detect and install every non standard component (from terminal as user mks):
```
cd ~/FreeDi && ./install.sh
```

5. I still dont get the part about the Raspberry Pi 😅

Ferropluss, 2025-01-18:
So, what file in boot is responsible for Wi-Fi? To write a line with my said and password. And what is the IP of the device registered in? Since you write that IP:10080 no longer exists.

Phil1988, 2025-01-18:
Set wifi via display. Its easier.
Otherwise:
WLAN data can be entered via nmtui (Network Manager Text User Interface) or with nmcli.
The configuration files are located in:
`/etc/NetworkManager/system-connections/`

Each connection has its own file. For example:
`/etc/NetworkManager/system-connections/<WLAN_NAME>.nmconnection`
These files contain the SSID, password and other connection details.
EG:
```
[connection]
id=<WLAN_NAME>
uuid=<UUID>
type=wifi

[wifi]
ssid=<SSID>

[wifi-security]
key-mgmt=wpa-psk
psk=<WLAN_PASSWORD>

[ipv4]
method=auto

[ipv6]
method=auto
```

For access via browser, its **just the IP** address (with the hidden standard port 80).
[image]

Ferropluss, 2025-01-18:
Tomorrow, I'll do everything. I will also rely on your wiki. I'll write the result.

Phil1988, 2025-01-23:
Please check this out and feedback 😉
https://youtu.be/_ZmSTJBWUus?feature=shared&t=2479

Phil1988, 2025-01-27:
Please see the wiki:
https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#2-your-mainboard-mcu-and-the-toolhead-are-not-on-klipper-v0120-289---and-you-want-it-the-easy-way

There is also a video that shows it
https://youtu.be/_ZmSTJBWUus?feature=shared&t=989

closed 2025-01-27 by Phil1988