Phil1988/FreeDi #85: [Help] Printing Error "The value 'chamber_heater' is not valid for HEATER" and more..

EisbacherG, 2025-01-18:
Hello Phil.
Thanks again for your time.
Since yesterday i'm able to start my Qidi x-plus 3 after solving the problem that mcu MKS_THR could not connect.
Everything seems to work perfectly.

Today I wanted to do a test print (calibration) and I got an error message.

I'm working with the Orca Slicer V 2.2.0
I set up the printer and the preview images as described in the wiki.

The following 2 errors occur after mesh bed leveling:
The value 'chamber_heater' is not valid for HEATER
SDCARD_RESET_FILE cannot be run from the sdcard

I can't find the entry chamber_heater in the printer.cfg.

Below is a screen with the errors:

[image]

Unfortunately, my Google search didn't produce any helpful results.
Perhaps you have an idea what could be going wrong.
I'm a bit confused at the moment.

Phil1988, 2025-01-18:
Please adjust yur M191 macro in macro.cfg.

rename "chamber_heater" to "chamber":
[image]

Error was introduced by a false positive of a user before I published the latest release and I didnt checked it myself.

You can always compare your config with my latest (fixed) config in the config section:
https://github.com/Phil1988/FreeDi/blob/master/config_section/X-Plus3/macros.cfg

Unfortunately there is no push message I can send to all users, so I expect others to have the same issue.

closed 2025-01-18 by Phil1988

EisbacherG, 2025-01-18:
Thanks again Phil.
Everything is working fine now.