Phil1988/FreeDi #87: [Bug] Smart-3: Unable to restart printing following manual interruption

Algorus21, 2025-01-19:
**Describe the bug**
When I manually interrupt a print (stop button on the screen or mainsail), I cannot restart a print. To do this, I have to request a movement of the bed or the head beforehand (screen or mainsail) and then it works

**Expected behavior**
I would like to be able to restart a print without having to perform a prior manipulation to restart one.

**To reproduce**
Steps to reproduce the behavior:
1. start a print
2. Click on stop
3. restart a print that is in memory
4. nothing starts
5. change the position of the bed (or other) from the screen
6. restart the print
7. it works :)

**Screenshots or image**
I could make a video if needed (but I have a 10h print that is in progress :p )

**Base (please complete the following information):**
 - OS: [Armbian-unofficial 24.2.0-trunk bookworm]
 - Source: [FreeDi 1.30 (update from 1.21 image) + updated print.cfg]
 - Mainboard: [ stock Mainboard X-6 v1]

And otherwise...great job! The wiki of the time (1.21) allowed me to make the transition without worries :)

Phil1988, 2025-01-20:
Im currently busy, so can you please assist me and check if there is a "brush" symbol on mainsail, after you stopped the print? (it is a blue brush at the top bar or at the print job section if I recall correctly).

Might be a "not flush" state of the printjob, but I cant have a look right now.

Algorus21, 2025-01-20:
[image]

indeed there is the blue brush, as well as an error code "Move out of range" that I had not noticed.

Phil1988, 2025-01-21:
Thats a good catch ;)
Thanks for the screenshot.
So your problems seems to be caused by this "Move out of range" error.
Your printer wants to move more then the allowed z-height.

I changed some line break and spacing errors caused by windows style format.
Check if this print end and cancel print macro fixes it for you:
https://github.com/Phil1988/FreeDi/blob/master/config_section/X-Smart3/macros.cfg

If it doesnt fix it, I hope my explaination above helps you finding the cause (and please report back).

Algorus21, 2025-01-21:
I modified with the new macro.cfg and....it worked, no more errors. I am impressed by your mastery of cfg!!!

Phil1988, 2025-01-23:
Thanks for you kind words!
But also for your feedback and your patience 👍 

Didnt had the time yet to review everything as I was buys with the video guide 😅

closed 2025-01-23 by Phil1988