Phil1988/FreeDi #88: [Feature request] PRINT_END to lower the bed only a bit (similar to QIDI stock behaviour), instead of all the way down.

MaZZly, 2025-01-19:
**Is your feature request related to a problem? Please describe.**

When printing some small parts, it feels quite unnecessary that the bed is brought all the way down after finishing.

**Describe the feature you'd like**

Instead of bringing it all the way down, ~10cm (relative from end) would probably be enough?

Maybe there is some benefit to lowering it all down that I'm not understanding/seeing? 🤔 

---

I was looking at the `PRINT_END` macro and trying to figure out how to modify it to only lower it a bit after finishing..

I think that modifying the bit:
```
    G0 Z{z_safe} F3600       
```
to be:
```
    G91       ;  relative positioning
    G0 Z100 F3600       ;  bring bed 100mm down
    G90       ; reset to absolute positioning (maybe not needed?)
```
Should do what I want... Sadly I'm a gcode noob, so I'm not sure if that will work and be safe? 
i.e. what happens if the 100mm relative would go past `position_max`, does it stop at the max, or simply ignore the command?

Would appreciate some help with achieving similar behaviour as the stock printer had 😇

breathless19, 2025-01-19:
In my notes after updating my machine to FreeDi, I had the same thing to request.    It doesn't seem to always lower to the lowest possible position though, so I'm confused.   I believe that a failed / stopped print goes all the way to the bottom, while a successful doesn't necessary, but I'm not positive.   It would be nice to get an explanation of the current logic its using and possibly get an explanation of how to change it.

Phil1988, 2025-01-20:
> . what happens if the 100mm relative would go past `position_max`, does it stop at the max, or simply ignore the command?

It will ignore it in this case.
You can copy this part from your backup Qidi config, then you'll have the same behavior 😊

> It would be nice to get an explanation of the current logic its using and possibly get an explanation of how to change it.

The current logic is:
Always lower the bed all the way down for maximum convenience.
If it does not, could you check if your slicer is generating G-codes that call the PRINT_END macro on finish?
If it still doesn’t work, its not as I intended, and I'll need to look into it.
Right now, I have higher priority tasks to focus on.
Feel free to change it as you like, its your machine, and you are free to do whatever you want with it.

If I change it to your suggested logic, I will likely get feature requests asking for the bed to be lowered to the bottom at the end of the print, because some users prefer that.
I can't provide a universal setup that fits every taste, and I hope you understand that 😊

pmbroth, 2025-01-20:
When I print, and it is completed, the bed does not go all the way down, it goes approximately 65% down from the top.  I thought that was what was suppposed to occur.  I'm using Orca Slicer latest

breathless19, 2025-01-20:
> If I change it to your suggested logic, I will likely get feature requests asking for the bed to be lowered to the bottom at the end of the print, because some users prefer that. I can't provide a [...]

Sorry, I hope I didn't come off as rude.   I didn't mean your logic for making it do what its doing, I meant the logic in terms of the section of code in the printer.cfg necessary to change the behavior.   I'm a klipper noob.

Phil1988, 2025-01-21:
> Maybe once we have a working gcode that does what the stock does, it could be included to Wiki to "get the old behaviour back."

Thats would totally work and appreciated by others I think 👍 

> Sorry, I hope I didn't come off as rude. I didn't mean your logic for making it do what its doing, I meant the logic in terms of the section of code in the printer.cfg necessary to change the behavior. I'm a klipper noob.

No worries at all, mate! :)  
You can find the section youre looking for in the `macro.cfg` file.  

Check out the "print end" and "cancel print" sections.  
I dont have time to dive deeper into this at the moment, but I'm confident that with some testing and a bit of internet searching, you'll find the solution. 

Best of luck and feedback your outcome!

closed 2025-01-23 by Phil1988