Phil1988/FreeDi #90: [Help] Change a Displaymenu entry

cptcl, 2025-01-20:
Hello and first let me say,..
**Amazing and awesome work!**

Qidi X-Max 3 rev2

This worked like a charm.
Followed your guide, everything worked perfectly, except for flashing toolhead and mainboad.
There i had to compile and use my own files instead. (followed easy guide and switched to advance for compiling)
But no problem so far.

I just have a question

_Just an appearence thing_
i would like to change the name of the presets on the main display.. ABS to ASA and TPU to NYLON (cause i never print abs and rarely tpu, but more nylon) How can this be done? (i just dont want to try and error here)
i changed the names in mainsail settings, but i wish i could change it on the display too

Phil1988, 2025-01-21:
Hey @cptcl and welcome on board :)

I had an issue with flashing the toolhead and mcu 2 days ago when doing a video shoot.
It was caused by downloading the files via right-click and not as "raw file".
I added the hint in the wiki already:

[image]

As I have a dev environment, I never downloaded them via browser - so it was surprising to me.
Maybe this was also the reason for your issue?

Anyway: I happy you made it 👏

To your request a short answer:
You cant.

The presets being hard coded in the software.
**Its on the list to change that**, but its a very low priority thing currently.
Mainly because its time consuming and only a minor/cosmetic thing.

Fixing bugs, including daily use case features and working on the guide/wiki are currently more important.
I hope you understand that and can be patient.
Leave this open, so you get a notification when something changes in the future :)

cptcl, 2025-01-21:
Mybe thats it, ..
I was also wondering why both files where 256kb in size while the compiled was only 32kb (i.e. for the mainboard)

Thanks for the 2nd answer.
Will leave this open for future

Phil1988, 2025-06-24:
@cptcl 
FreeDi v2.00 was just released.
It allows you not only to change the preset names, and also not only to adjust the temperatures ...
It allows you to but ANY macro to these buttons you like - yes! This emoji was made for it: 🤯

To make use of it, install FreeDi v2.00 and head over to the freedi.cfg:
[image]

Have fun with it mate :)