Phil1988/FreeDi #91: [Bug] Shaper calibration error over display

cptcl, 2025-01-20:
Hi

X-Max 3 Rev.2

Did the install like an hour ago and everything worked so far.
Got the error about the heater_chamber and fixed it with the solved issue section post.

Wanted to do the Shaper calibration over the display and got...

> 14:20
> MCU 'mcu' shutdown: Timer too close
> This often indicates the host computer is overloaded. Check
> for other processes consuming excessive CPU time, high swap
> usage, disk errors, overheating, unstable voltage, or
> similar system problems on the host computer.
> Once the underlying issue is corrected, use the
> "FIRMWARE_RESTART" command to reset the firmware, reload the
> config, and restart the host software.
> Printer is shutdown
> 14:20
> SAVE_CONFIG
> 14:20
> MCU 'mcu' shutdown: Timer too close
> 14:20
> MCU 'mcu' shutdown: Timer too close
> This often indicates the host computer is overloaded. Check
> for other processes consuming excessive CPU time, high swap
> usage, disk errors, overheating, unstable voltage, or
> similar system problems on the host computer.
> Once the underlying issue is corrected, use the
> "FIRMWARE_RESTART" command to reset the firmware, reload the
> config, and restart the host software.
> Printer is shutdown
> 14:20
> PAUSE
> 14:20
> Internal error on command:"SHAPER_CALIBRATE"
> 14:20
> Internal error on command:"RESHAPER_CALIBRATE"
> 14:20
> Klipper state: Shutdown
> 14:20
> Testing frequency 34 Hz
> 14:20
> Testing frequency 33 Hz
> 14:20
> Testing frequency 32 Hz
> 14:20
> Testing frequency 31 Hz
> 14:20
> Testing frequency 30 Hz
> 14:20
> Disabled [input_shaper] for resonance testing
> 14:20
> Testing axis y
> 14:20
> Re-enabled [input_shaper]
> 14:20
> Testing frequency 150 Hz

Error in console.

Did the shaper calibration right after that over mainsail and it worked.

Is this known?

pmbroth, 2025-01-20:
I did the install last week, but updated today with Phils lastest configurations (Macro, Printer, Moonraker)  All is working perfectly.  I Just ran shaper and no issues.,  My Qidi X-max 3, has been upgraded with all of teh upgrades Qidi sent out.   I purchased it a few months after they released it.

Phil1988, 2025-01-21:
The display input shaper button does the same as the shaper calibration over mainsail.
So might it be just bad luck or can you replicate this?
On my end everything is working.

Please do it again and feedback.
And share the full output on mainsail when you do it over mainsail please :)

cptcl, 2025-01-21:
Seems to be that i just have been unlucky. Did run the shaper over display again and now it worked without error.

closed 2025-01-21 by cptcl