Phil1988/FreeDi #92: [Feature Suggestion]  First tap after screen dims to wake screen instead of initiate command

breathless19, 2025-01-20:
This is just a suggestion as it would seem more intuitive to me.   

On many touch screen devices, when the screen has significantly dimmed, the first tap to bring the screen back to full brightness would not initiate a command, but rather would just wake the screen up (similar to how its currently set in your firmware to wake the screen from full sleep).   My suggestion would be to apply that same setting to the dimmed screen as well.   

From dimmed screen =====> First tap initiates max brightness  =====> Second tap initiates command

Phil1988, 2025-01-21:
That sounds great, and I like this behavior!  
When I was creating the screen firmware, I couldn't find a solution to imitate it. 
Dimming only lowers the brightness, but the screen remains active.  

I'll take another look when I have some time (since it's not critical, this might take around two months). 
However, I don’t want to raise high hopes. :)

breathless19, 2025-01-21:
No worries, I appreciate your work and have already made a donation to your project.   You deserve it!  Take your time, and I have a few other suggestions that I'm going to post too, so just take them with a grain of salt.   Everything takes time, so just take everything slowly and keep up the good work dude.

Phil1988, 2025-01-23:
Thank you for that and also for your kind words!
Im very happy that appreciate and enjoy this project ☺️

breathless19, 2025-02-10:
I'd just like to point out that the behavior that I requested in this suggestion is kind of already implemented if you tap on the top portion of the screen.   There are no pushable buttons in the top middle of the screen, so to wake the screen up without risk of pushing something you shouldn't, you can just tap in the upper middle part and it wakes the screen up to full brightness.

Phil1988, 2025-06-24:
@breathless19 

This is now integrated at v2.00.
As of https://github.com/Phil1988/FreeDi/issues/93 this was quite some work - but I think we all see the benefit.
Thanks for making this suggestion ☺️