Phil1988/FreeDi #94: [Help] Probing the Bed directly over the bed screws

cptcl, 2025-01-21:
Hi...
U did a change to the screw tilt adjustment [here](https://github.com/Phil1988/FreeDi/commit/9b71692c58c951dcde5ec133099f13dc9a55afe5) and i was wondering...
On my X-Max 3 Rev 2 i positioned the Probe exactly above the 4 bed screws,
i got this positions for my printer. Maybe i have another layout of the screws/bed.

screw1: 5, 30
screw1_name: front left
screw2: 260, 30
screw2_name: front right
screw3: 260, 285
screw3_name: rear right
screw4: 5, 285
screw4_name: rear left

or is it not good/bad to have the probe exactly above the screw for probing?
Yes i was wondering if probing above the screw would realy impact the accuracy and/or consistency, while instead over the magnetic bed, slightly off of the screws.
 
Need to mention, i did a mod to my bed for tilt adjust system, with steel screws, silicon washers and click wheels. so no aluminum screws or something

Thanks for the info in advance

rosenrot00, 2025-01-21:
The adjustments correct an incorrect coordinate for screw 2, as evident from the changes made.

I can't verify your bed screw positions as I'm running a Carto and no BLTouch anymore, however, I trust your measurements.

Have you considered testing for accuracy errors around the screws by adjusting your bed mesh coordinates and resolution to take measurements near one of the screws? This approach could provide all the insights you’re looking for.

If you assume three screws are level and one is off by 1 mm (a significant deviation), the calculated distance between screw 1 and screw 3 is approximately 360 mm. This results in a slope of 0.00277 mm/mm. In short, even being off by 10 mm would have minimal impact on the leveling accuracy. But there would not be a need for deviating more than 5 mm at all.

cptcl, 2025-01-22:
Yes, i already noticed that the coorinates in the pinter.cfg was way off and just inserted mine before i did the first run.

in the next few days i will install a beacon 3D RevH scanner.

I use metal buildplates, pei coated, smooth and textured.
if i look towards the mesh i have good results <0,15mm diviation across the buildplate @80°
in my opinion the savest way, with most accuracy, could be to have the probe like 10mm from the screw, towards the middle of the printbed. So u dont have to deal with screwhole or inperfection there, and also have an even and exact messurepoint from each screw (and from the edges of the bed).

What i want to say is.. with your coordinates, where the nozzle is exactly above the screw, u would always messure +x and +y from every screw and i think it would be better to have it like... 
screw 1 +x +y, screw 2 -x +y, screw 3 -x -y, screw 4 +x -y 
(in perspective from the screws)

and,.. yes.. this is maybe me overthinking everything way to hard and maybe its not rly that important.

It was just a thought with "perfection" in mind that i had.
Thanks for you input there

Phil1988, 2025-01-23:
I will check this again today on my side and report back :)

closed 2025-01-23 by Phil1988