Phil1988/FreeDi #96: [Help] WIFI install not worked with TPLINK-T2U NANO

KIA1333, 2025-01-21:
Greetings,

Everything followed by wifi enabling guide.
After everything done and full shutdown - startup. In network settings i don`t find any wifi.
Tried to plug in on turned off machine, Tenda (What was included with printer) - after boot up - wifi works.
Tenda dongle as recovery option is ok, but it is too slow and rubbish signal.. 

Can you provide commands what i need to check or do with TPLINK dongle to make it detectable as wifi adapter?
As this dongle i think is wide spread.

rosenrot00, 2025-01-21:
Did you look into the first 3 Google results before posting here, and none worked?

It could be that this might be helpful. https://suraj.dev/blog/configuring-wireless-usb-adapter-archer-t2u-nano-on-linux

KIA1333, 2025-01-22:
Thank you for replay. Yes i checked, and in device menu i see it, using lsusb shows it.
It is detected correct. TP-LINK AC600 wireless RTL8811AU.
But on physical device screen, i don`t see any wifi broadcast and looks.

rosenrot00, 2025-01-22:
Sorry if I didn't get it. Did you try the guide I sent as well?

colin-small, 2025-01-22:
@KIA1333 Make sure the card actually works. I had a card die in my X Max 3 and ended up having to replace it. Maybe try the card in another system just to be sure that it is functional.

KIA1333, 2025-01-22:
yes it is working - before turning to freedi - it worked also it is  displayed in linux device, i understand from info what i see it has no working drivers at the moment.

when tried - https://suraj.dev/blog/configuring-wireless-usb-adapter-archer-t2u-nano-on-linux

Deprecated feature: REMAKE_INITRD (/usr/src/rtl8812au-5.3.4/dkms.conf)
Creating symlink /var/lib/dkms/rtl8812au/5.3.4/source -> /usr/src/rtl8812au-5.3.4
Sign command: /lib/modules/6.6.17-current-rockchip64/build/scripts/sign-file
Binary /lib/modules/6.6.17-current-rockchip64/build/scripts/sign-file not found, modules won't be signed
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/rtl8812au/5.3.4/source/dkms.conf)

Building module:
Cleaning build area...
'make' -j4 KVER=6.6.17-current-rockchip64 KSRC=/lib/modules/6.6.17-current-rockchip64/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.6.17-current-rockchip64 (aarch64)
Consult /var/lib/dkms/rtl8812au/5.3.4/build/make.log for more information.
Sign command: /lib/modules/6.6.17-current-rockchip64/build/scripts/sign-file
Binary /lib/modules/6.6.17-current-rockchip64/build/scripts/sign-file not found, modules won't be signed
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/rtl8812au/5.3.4/source/dkms.conf)

Building module:
Cleaning build area...
'make' -j4 KVER=6.6.17-current-rockchip64 KSRC=/lib/modules/6.6.17-current-rockchip64/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.6.17-current-rockchip64 (aarch64)
Consult /var/lib/dkms/rtl8812au/5.3.4/build/make.log for more information.
Finished running dkms install steps.

what next i will try https://github.com/aircrack-ng/rtl8812au
i dont have very deep linux knowladge, but i see others with similar kernel version use this mostly, but there are other of course...

Phil1988, 2025-01-23:
I would suggest you remove the current drivers first, to avaoid conflicts:

```
sudo dkms remove rtl8812au/5.3.4 --all
sudo dkms remove 8812au/5.6.4.2_35491.20191025 --all
```

then install the aircrack-ng driver, by doing something like this:
```
sudo apt-get install dkms
git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git
cd rtl8812au
make clean
make -j4
sudo make install
sudo modprobe 88XXau
```

Confirm the driver is loaded:
```
lsmod | grep 88XXau
```

Reset the  networkmanager and check if it can recognize it:
```
sudo systemctl restart NetworkManager
nmcli dev status
```

if its visible, try to use the display to connect to a wifi.

KIA1333, 2025-01-23:
Thank you tried and now i see this:
DEVICE           TYPE      STATE                   CONNECTION
end1             ethernet  connected               Wired connection 1
lo               loopback  connected (externally)  lo
wlxf0a7314b83ca  wifi      unavailable             --

lsmod
88XXau               2195456  0

So maybe lets leave at the moment like this, i will use tenda wifi dongle while it is working, but maybe other community members with more experience with linux and such dongle will check this and i could try other options

Phil1988, 2025-01-27:
Please open this at the discussions section as I am closing it now to get it off my focus :)

closed 2025-01-27 by Phil1988