Phil1988/FreeDi #100: [Bug] Wifi activation script failure

sparehead1, 2025-01-22:
**Describe the bug**
Script fails to restart usb-modeswitch

**Expected behavior**
usb-modeswitch.service should restart

**To Reproduce**
Steps to reproduce the behavior:
1. ssh in
2. run cd ~/FreeDi && ./install.sh
3. error message "Failed to restart usb-modeswitch.service: Unit usb-modeswitch.service not found." seen part way through

**Fix**

 line 348 needs to be sudo systemctl restart usb_modeswitch not sudo systemctl restart usb-modeswitch

Phil1988, 2025-01-23:
Thanks for your support 😊

Ive just fixed it 👍

closed 2025-01-23 by Phil1988

osvch, 2025-01-27:
Hi, 
for me "_" is not working:

[image]

osvch, 2025-01-27:
Hi @sparehead1 
for me "_" is not working:

[image]

It fails during `cd ~/FreeDi && ./install.sh` too. Did you manage to install successfully?

osvch, 2025-01-27:
Strange, I'm getting "permission denied". Downloaded file in the correct folder, SSH logged as mks user.

[image]

but 
`# Reload udev rules
sudo udevadm control --reload`
itself seems to be working