Phil1988/FreeDi #101: [Help] X plus 3 SCREWS_TILT_CALCULATE goes off the end of the bed on screw 2

sparehead1, 2025-01-22:
The values for Screw tilt adjust in printer.cfg has the probe go off the end of the bed when it goes to the screw 2 position on my X plus 3.

When I line up the probe with the screws on my X plus 3 I get:

screw1: 22, 45
screw1_name: front left screw
screw2: 202, 45
screw2_name: front right screw
screw3: 202, 224
screw3_name: rear right screw
screw4: 22, 224
screw4_name: rear left screw

Is this the right way to set those numbers?

Phil1988, 2025-01-22:
The coordinates being meant for the nozzle (not the probe).

Can you please verify the coordinates?

sparehead1, 2025-01-23:
If I take the coordinates of the nozzle over the screws I get

screw1: 49, 50
screw1_name: front left screw
screw2: 229, 50
screw2_name: front right screw
screw3: 229, 229
screw3_name: rear right screw
screw4: 49, 229
screw4_name: rear left screw

Phil1988, 2025-01-23:
Thanks for your report.
I can not verify this as I totally rely on plus3 users.

Your first line up is correct as this says
https://www.klipper3d.org/Manual_Level.html#adjusting-bed-leveling-screws-using-the-bed-probe

Problem for me is that I got 3 totally different coordinates from plus 3 users to the date.
Can you please tell me when you purchased this plus3 roughly and maybe also add a picture of your bed screws here?

I included your coordinates now, but I wouldnt be surprised if another user comes around with a different settings and I want to track that down and get it sorted :)

Thanks for your help and feedback 👍

sparehead1, 2025-01-23:
Happy to help :)

I got mine May 2024, directly from qidi.

[image]

Phil1988, 2025-01-23:
Thanks mate!

Pushed the corrected version to the [config section](https://github.com/Phil1988/FreeDi/tree/master/config_section) and will include it in future images 👍

closed 2025-01-23 by Phil1988