Phil1988/FreeDi #102: [Feature request] Subfolders/parent navigation support on LCD

theclue, 2025-01-22:
Having the subfolder support on the last release was a great improvement, but to me which I have a 32Gb storage and A LOT of files and folders inside it's still suboptimal, as they are simply too many to use the panel.

Additionally, subfolders are not in alphabetic ordering (perhaps it was a specific design choice?)

I was wondering if you could consider adding the possibility to show and navigate to subfolders and ofc go back to parent's one. This would make much more easier to organize the files themselves, as they would reflect the same way they are stored in the file system.

tnx allot!

Phil1988, 2025-01-23:
> Additionally, subfolders are not in alphabetic ordering (perhaps it was a specific design choice?)

Yes that is correct.
They are sorted by date (latest first) as this is the most common and usable order.

Your request is understandable and I want to include it.
But have mercy that it will not happen in the next ~6 weeks 😅

Phil1988, 2025-01-28:
Please check the [release v1.40](https://github.com/Phil1988/FreeDi/releases/tag/v1.40).

I'll close this now as its now featured.
If you have any problems, please answer here and I will reopen.

closed 2025-01-28 by Phil1988