Phil1988/FreeDi #103: [Bug] You select a file, another one is printed

theclue, 2025-01-22:
I think it was only due to https://github.com/Phil1988/FreeDi/issues/52

but in facts, the bug seems more articulated so I filed a new bug report:

- I printed a file, let's call it "A"
- I selected anothet one, let's call it "B"
- Filename is for "B", but thumbnails and other data being displayed are for "A"
- I started the print, "A" was printed

[image]

[image]

[image]

Phil1988, 2025-01-23:
Can you show me the file structure of you usb drive?
(Just to be sure)

There have been recent development on this part, but I didnt find the time to release it yet.

Phil1988, 2025-01-28:
@pmbroth 
Please check the [release v1.40](https://github.com/Phil1988/FreeDi/releases/tag/v1.40).

I'll close this now as its now.
If you have any problems, please answer here and I will reopen.

closed 2025-01-28 by Phil1988