Phil1988/FreeDi #99: [Bug] Borat Fan ?

ohyeah8000, 2025-01-22:
In your config section you describe a Broad Fan ;) [heater_fan borad_fan] 

Please spell it correctly. Borat is written with a T!

MarkusKgit, 2025-01-22:
Not a bug, that's just a reference to the original wrong spelling in the official Qidi config (most likely they meant Board Fan). The section in the wiki is therefor correct. In the FreeDi config its named `MCU_fan`

theclue, 2025-01-22:
I think he is ironic btw :D

[image]

Phil1988, 2025-01-22:
@ohyeah8000 Thank you for your comment. Just to clarify, are you being serious about this suggestion, or was it meant as a joke?

I want to ensure I address the report appropriately.

ohyeah8000, 2025-01-23:
Of course it's a joke! 😃

But the same time I'm serious and pointed out a little misspelling in your final modified printer.cfg to make it better. Board_fan it has to be of course...

[image]

Phil1988, 2025-01-23:
Thank you for your input!

However, I’d like to kindly remind you that bug reports should focus on genuine issues related to the project. 
Jokes or unrelated comments can make it harder to efficiently address real problems and take up valuable time.

I’m totally open to jokes, but I’d really appreciate it if you could place them anywhere else (like Discussions, Discord,...).

Thank you for your understanding, and I look forward to receiving more valuable feedback from you in the future!

closed 2025-01-23 by Phil1988