Phil1988/FreeDi #105: [Announcement] Temporary reduced availability

Phil1988, 2025-01-23:
To all the **FreeDi**ans!

Over the next six weeks, I will be focusing intensively on personal matters.<br>
I kindly ask for your understanding if I'm not as available as usual during this time.

I'm confident that 
with the Wiki 📑 , 
the video guide 📹 , 
and the amazing people ❤ in this community, 
most questions can be answered and issues resolved.

A huge thank you to everyone who contributes to this project and helps others along the way - it truly makes a difference!

I'm already looking forward to having more time to implement your suggestions and improvements, making this project an even better experience for everyone.

All the best,
C0co

pmbroth, 2025-01-23:
Phil - we wil try to hold down the fort with answering questions.  I hope all is well in your life.  Brian

Phil1988, 2025-01-23:
Thanks mate, I appreciate you ☺️

robonxt, 2025-01-24:
You take care of what you need to do.
Thank you for the hard work and time you put in FreeDi!

Phil1988, 2025-02-27:
I'm back :)

closed 2025-02-27 by Phil1988

pmbroth, 2025-02-27:
Welcome Back Phil!!

B-Kramer, 2025-03-01:
Welcome Back!