Phil1988/FreeDi #106: [Help] update question since I'm not 100% sure about

cptcl, 2025-01-23:
Hi

X-max 3 Rev. 2
Installed FreeDi with image

For updates.. Is it save to use kiauh?
can I do all updates there?
even Klipper (and afterwards compiling new files for Mainboard and tool head)?

Or should I only use updates from the webui mainsail?

Thx in advance.

Phil1988, 2025-01-23:
The first question that comes to my mind is, why would you update via kiauh when you can use the moonraker update manager?

Im not 100% sure on this, but I think kiauh sometimes installs non stable releases and grabs packages that might cause issues.
Except of some minor things like installing a non compatible python version, I think kiauh should work.
If you are experienced with this and might be able to fix this, sure, you can do that.

Bu I recommend doing it via the webui (mainsail) as this is the savest way.

Does this answer your question? 😊

cptcl, 2025-01-23:
I asked cause in kiauh I see a moonraker and a Klipper update.
in webui I don't see them.
And I was not sure if I can even update moonraker from kiauh.

So far I only did the system updates in Webui.

So yes, I this answers at least most of my question.

Phil1988, 2025-01-23:
Thanks for clarifying! 😊

The WebUI (Mainsail) shows updates managed by the Moonraker Update Manager of stable releases.

Kiauh can update Klipper and Moonraker, but it sometimes pulls experimental or unstable versions, which might cause issues (e.g., incompatible dependencies).

If updates appear in the WebUI, they are stable versions and safe to use.
Hope this helps! 😊