Phil1988/FreeDi #107: [Bug] Bed collides with bottom of printer when canceling and then homing

McSneaky, 2025-01-23:
**Describe the bug**
When you cancel print, bed goes to the bottom of printer, now when you try to start new print / home after it, it tries to move down some more in it's homing procedure, which makes bed collide with printer frame and z axis belt will skip steps

**Expected behavior**
Bed won't collide. Maybe not move bed to totally bottom of printer when colliding, or adding check into homing, so homing doesn't try to lower bed any more

**To Reproduce**
Steps to reproduce the behavior:
1. Start pring
2. Cancel print
3. Home printer
4. See the bed colliding

**Console output**
I'll try to get it next time it happens

**Base (please complete the following information):**
 - X3seriesLCD firmware version: v1.30-0-g388855ca
 - OS: Armbian-unofficial 24.2.0-trunk bookworm
 - Source: FreeDi image
 - Mainboard: Stock mainboard, I think was X-4

McSneaky, 2025-01-23:
If it happens again, I can try to get console output and ensure in steps

Idea I had is just to make cancel print to not bottom out. I can make PR for that

Phil1988, 2025-01-23:
please show me your 
[gcode_macro G28]
[gcode_macro CANCEL_PRINT]
from the macros.cfg
and your 
[stepper_z]
from the printer.cfg

and what printer do you have?

McSneaky, 2025-01-23:
Everything is stock that came with FreeDi 1.30, except Cartographer section

<details>
<summary>[gcode_macro G28]</summary>

```yml
[gcode_macro G28]
rename_existing: G28.1
gcode:
    {% set home_all = ('X' in rawparams.upper() and 'Y' in rawparams.upper() and 'Z' in rawparams.upper()) or 
                      ('X' not in rawparams.upper() and 'Y' not in rawparams.upper() and 'Z' not in rawparams.upper()) %}
    {% set home_current_x = printer.configfile.settings['tmc2209 stepper_x'].run_current * 0.8 %}
    {% set home_current_y = printer.configfile.settings['tmc2209 stepper_y'].run_current * 0.8 %}
    
    {% set init_XY_move = 30 %}
    #{% set y_init_move = 5 %}
    {% set z_clearance = 2 %}

    {% set home_all_final_position_x = printer.toolhead.axis_maximum.x / 2 %}
    {% set home_all_final_position_y = printer.toolhead.axis_maximum.y - 30 %}
    {% set home_all_final_position_z = 20 %}

    # If homing should at least move X or Y axis
    {% if home_all or 'X' in rawparams.upper() or 'Y' in rawparams.upper() %}     
        # Check if Z is homed
        {% if 'z' in printer.toolhead.homed_axes %}
            {% set Z_axis_was_homed = true %}                                                                       ; Set Z_axis_was_homed to true if Z is already homed
            G91                                                                                                     ; Use relative positioning
            G1 Z2 F600                                                                                              ; Move bed down
            G90                                                                                                     ; Use absolute positioning
        {% else %}
            {% set Z_axis_was_homed = false %}                                                                      ; Set Z_axis_was_homed to false if Z is not homed
            FORCE_MOVE STEPPER=stepper_z DISTANCE={z_clearance} VELOCITY=10                                         ; Move bed down
        {% endif %}
    {% endif %}

    {% if home_all or 'X' in rawparams.upper() or 'Y' in rawparams.upper() %} 
        G91                                                                                                         ; Use relative positioning
        FORCE_MOVE STEPPER=stepper_x DISTANCE={init_XY_move} VELOCITY=40                                            ; Move X a little before homing
        G90                                                                                                         ; Use absolute positioning
    {% endif %}

    # Home X
    {% if home_all or 'X' in rawparams.upper() %}        

        SET_TMC_CURRENT STEPPER=stepper_x CURRENT={home_current_x}                                                  ; Lower motor current
        G28.1 X                                                                                                     ; Home X
        SET_TMC_CURRENT STEPPER=stepper_x CURRENT={printer.configfile.settings['tmc2209 stepper_x'].run_current}    ; Set motor current to the previous value
        G1 X10 F1200                                                                                                ; Move X-axis slightly    
    {% endif %}

    # Home Y
    {% if home_all or 'Y' in rawparams.upper() %}
        SET_TMC_CURRENT STEPPER=stepper_y CURRENT={home_current_y}                                                  ; Lower motor current
        G28.1 Y                                                                                                     ; Home Y
        SET_TMC_CURRENT STEPPER=stepper_y CURRENT={printer.configfile.settings['tmc2209 stepper_y'].run_current}    ; Set motor current to the previous value
        G1 Y10 F1200                                                                                                ; Move Y-axis slightly    
    {% endif %}

    # Raise the bed again if only X or only Y axes were homed
    {% if not (home_all or 'Z' in rawparams.upper()) %}
        {% if Z_axis_was_homed %}
            G91                                                                                                     ; Use relative positioning
            G1 Z-2 F600                                                                                             ; Raise the bed back again
        {% else %}
            FORCE_MOVE STEPPER=stepper_z DISTANCE=-{z_clearance} VELOCITY=10                                        ; Raise the bed back again
        {% endif %}
    {% endif %}

    # Home Z
[... 78 lines cut ...]
algorithm: bicubic
```
</details>

<details>
<summary>[stepper_z]</summary>

```yml
[stepper_z]
step_pin:PC10
dir_pin:PA15
enable_pin: !PC11
microsteps: 16
rotation_distance: 8
full_steps_per_rotation: 200
endstop_pin:probe:z_virtual_endstop
position_max: 322
position_min: -2
homing_speed: 10
homing_retract_dist: 0.0 # Must be set to 0 by Carto
second_homing_speed: 8
homing_positive_dir:false
step_pulse_duration:0.000002
```
</details>

Phil1988, 2025-01-23:
Thanks. There we have it! 👍 

Please see and compare with the line up on your CANCEL_PRINT macro.
Its is a CR LF error coming from editing in windows (im sorry) and was fixed 3 days ago.

See here:
https://github.com/Phil1988/FreeDi/blob/master/config_section/X-Max3/macros.cfg
[image]

Please change it accordingly and feedback if this fixes it for you ☺️

colin-small, 2025-01-23:
@McSneaky I just received my first Cartographer probe. Can you share with me what is needed configuration wise to make it work on my X Max 3 with FreeDi installed?

McSneaky, 2025-01-23:
> [@McSneaky](https://github.com/McSneaky) I just received my first Cartographer probe. Can you share with me what is needed configuration wise to make it work on my X Max 3 with FreeDi installed?

Please open new help issue, can explain in there

KIA1333, 2025-01-23:
> > [@McSneaky](https://github.com/McSneaky) I just received my first Cartographer probe. Can you share with me what is needed configuration wise to make it work on my X Max 3 with FreeDi installed?
> 
> Please open new help issue, can explain in there

There is discussion, @McSneaky "Cartographer" install, could you take a look and share some details?

Phil1988, 2025-01-27:
Probably the same as https://github.com/Phil1988/FreeDi/issues/87
Lets wait for @McSneaky to be back and report if its solved.

McSneaky, 2025-01-28:
Nope, issue still exists

- Start any print
- Tap "Cancel print" on screen
- Tap "Home all axis" on screen
- Bed starts to collide with bottom of printer

Phil1988, 2025-01-28:
2 things:

1: Can you check if the `horizontal_move_z: 5` in your [bed_mesh] causes this?
2: Can you show me your PRINT_END macro pls?

McSneaky, 2025-01-30:
I'll do test with debug loging enabled, hopefully will get something interesting from there

Phil1988, 2025-01-30:
> I'll do test with debug loging enabled, hopefully will get something interesting from there

No need, its just a macro/config issue.

This should happen:
you cancel the print 
(CANCEL_PRINT gets called)
-> bed lowers 2mm
-> print head moving to back mid position
(PRINT_END gets called)
-> bed lowers 2mm again
-> bed lowers to (max-z-position) - 10mm

is that different for you guys?

there are things we need to check:
is your max-z-position really 322?

Please home your printer and move it down step by step (like in 10mm increments).
Do you reach 322?

If not, your `position_max: 322` ins stepper `[stepper_z]` needs to be adjusted.

If its ok, it can be caused by the last call of `BASE_CANCEL_PRINT` inside the `CANCEL_PRINT` macro.
Please delete this line then and check if the issue persists.

McSneaky, 2025-01-30:
> is your max-z-position really 322?

That's interesting, mine is actually about 320.5. Maybe that's the last some mm thing

Phil1988, 2025-01-30:
That's definitely a surprise :D 

But I think it wouldn't solve this completely?
Please play a bit with it and report the outcome.
Best of luck 🤞

sgghostrider, 2025-01-30:
I had the same problem (QIDI X Smart 3) but in me case the main problem was that i have replaced the solid metal studs with the spring adjustment system which has it's screws protruding a little bit from below, so after my prints ended or they were cancelled, it went to the bottom(with the original command from QIDI it always went to Z_MAX/2 instead of Z_MAX, so did not have that problem before) and was colliding with the metal bottom making marks, so i changed the "PRINT_END" to stop this and also put the bed in a better position as many times i need to cancel the print and was tired of waiting it to go ALL the way down to just have it go ALL THE WAY UP again.

I believe the real problem here is "PRINT_END" as well, because "CANCEL_PRINT" only moves Z for 2mm and only if it has space, and not sure about this, but i think that "CANCEL_PRINT" it's only called from the screen, and not from klipper, because as far as i know you can't actually "CANCEL_PRINT" from klipper web UI, you have to first PAUSE and then STOP(i have a long print running right now so can't test which command is the "stop" button calling, but i assume it's "PRINT_END" as changing it fixed my problem).

This is the start of my modified "PRINT_END", i have 3 stages, fist one is "i have space so just go to half of max height", second for 2/3 if has space and third is go to bottom with offset when no space(accounted for my screws protruding out).
```
[gcode_macro PRINT_END]
gcode:
    {% set max_x = printer.configfile.config["stepper_x"]["position_max"]|float %}
    {% set max_y = printer.configfile.config["stepper_y"]["position_max"]|float %}
    {% set max_z = printer.configfile.config["stepper_z"]["position_max"]|float %}

    {% if (printer.toolhead.position.z + 12) < (max_z / 2) %}
        {% set z_hop = 2  %}
        {% set z_safe = (max_z / 2) - 10 %}
    {% elif (printer.toolhead.position.z + 12) < ((max_z / 3) * 2) %}
        {% set z_hop = 2  %}
        {% set z_safe = ((max_z / 3) * 2) - 10 %}
    {% else %}
        {% set z_hop = 0 %}
        {% set z_safe = max_z - 10 %}
    {% endif %}
```

Note: You can change "{% set z_safe = max_z - 10 %}" to "{% set z_safe = max_z - X %}" being X whatever millimeters you want to offset from the bottom, but i don't recommend going too wild with it as if your print is using the entire height of you Z Axis(unlikely but possible) it might go back INTO your print, so keep that in mind.

McSneaky, 2025-01-30:
Found that my max position in printer.cfg was 226. Set it to 220, but still got the issue

It can go to the bottom without issues, but as soon as I hit "Home" it tries to move down 3 times during the homing procedure. Which makes it collide with bottom of printer 3 times

sgghostrider, 2025-01-31:
> Found that my max position in printer.cfg was 226. Set it to 220, but still got the issue
> 
> It can go to the bottom without issues, but as soon as I hit "Home" it tries to move down 3 times during the homing procedure. Which makes it collide with bottom of printer 3 times

I don't see which printer model you have, so can't make exact comparison, but mine does not make down movements to the bed when homing, and if it does, they are millimeters kind of small and pretty sure are made just once, i know because i have homed my printer at "almost touching the nozzle Z heights" after emergency shutdown(done to interrupt long instructions that i did not want to wait like bed meshing and bed heating) leaving the printer nozzle very close to the bedplate in many cases, and when homing it just moves few mm up before activating the probe(i always watch it just in case) no perceivable down movements.

In any case in few hours when my print finishes will make few tests and watch it carefully, but if anything, you might copy here your "[gcode_macro G28]" code from "macros.cfg" to check it, maybe it's different than mine, which i doubt, but at this point anything is better than nothing 😂

My setup is:
Printer: X-Smart 3
Board: X-6
Image: FreeDi_v1.30_X-Smart3_Armbian24.2.0_Bookworm6.6.17.rar

Note: I frankensteined my "printer.cfg" with the original QIDI config in which i had done some modifications, as the new printer.cfg looked too strange, everything looked out of place, i think was missing some stuff and some pins had the wrong pinouts, but i don't think that has anything to do with the homing, but i felt like mentioning it just in case.

Phil1988, 2025-01-31:
> Found that my max position in printer.cfg was 226. Set it to 220, but still got the issue

226?
That seems incorrect and I am confused because you showed `position_max: 322` earlier 😵

> as soon as I hit "Home" it tries to move down 3 times during the homing procedure

That must be something you added to it.
Have you changed the macros.cfg by yourself or may it be something comnig from the cartographer?
Please also try to see how much of a movement each of these "3 movements" are doing.
This might help finding the source. If its 5mm this might confirm suspicions against the carto 😉

McSneaky, 2025-01-31:
> 226?
> That seems incorrect and I am confused because you showed position_max: 322 earlier 😵

Me too :D Maybe I didn't have my glasses on and looked it wrong.. Anyway, I'll do some more testing with modifying Carto settings too and see where I end up with

Will also try to record video

Phil1988, 2025-02-04:
Any news on this @McSneaky ?

ChrisInMilwaukee, 2025-02-04:
Hi @Phil1988 and all,

Here have been my experiences with testing this afternoon.

- The absolute bottom of my Z axis is a little less than 323mm
- If `position_max: 322` when I attempt a home operation, either with front panel button push, Mainsail button push, or start a new print, the bed will try to z-hop -2mm (per code) which will run it against the stops and slip a belt cog

Test ### Cases

I set `position_max: 320` to see what movement looked like during homing and other button pushes:

- The front panel "Move Bed to Bottom" sent the bed to 310 and stopped (expected behavior per PRINT_END line 112 `z_safe` calculation in macros.cfg - `{% set z_safe = max_z - 10 %}`)
- When canceling an active print, the bed descended to 310 and stopped (expected behavior)
- I can manually move the bed down to 320 where it will go no further (expected behavior)
- When it is there, no homing operation will work because any z-hop will exceed the max z position (expected behavior)
- When it is there, I can start a new print job, but it will fail because it can't home (expected behavior) 

So the CANCEL_PRINT and PRINT_END will collectively under the worse conditions z-hop 2mm, which should drop the bed 2mm then bring it back up each time (net zero).  

So, for first tests, I set `position_max: 320` and then cancelled a print, manually sent the bed to the bottom with the display button, and forced the z-axis to its programmed limit the and have not been able to "grind the gears" as it were.

### Now, here's where things get spicy.

During typical operation, the following observation doesn't affect anything because **the toolhead knows where it is in 3D space**.  If the carriage is all the way at the bottom at `position_max: 320`, and I hit the E-stop *or* make a configuration change that restarts Klipper, the system loses track of its toolhead whereabouts.  When the system restarts, it thinks it's at 0,0,0.

When I initiate a G28 home after a restart (I have a Beacon probe, so your results may be slightly different due to system overrides) the following occurs:

- The bed drops 2mm and moves to its safe home position (middle of the bed for me)
- The bed drops 5mm and then performs X-home
- The bed drops another 5mm and then performs Y-home
- The bed drops another 5mm and then performs Z-home
- It then ran the bed all the way to the top where it stopped just before it crashed into the print head because it thinks it ran out of Z-axis (thankfully)
- Another G28 home gave the toolhead its bearings again, but still did the 2/5/5/5 thing before everything zeroed out correctly

That's a total of **17mm** and it "ground the gears" three times on me during that evolution because it bottomed out after the first drop.  

So although it's a fringe case, I set my `position_max: 306`, incrementally for each drop and a 1mm buffer.  No amount of poor planning on my part allowed the z-axis to bottom out after the change.  This for the safety peace of mind for the hardware.  I lose a little bit of Z-height (18mm), but when it's at the bottom of the bucket, at least I'm not grinding anything to bits.  If I need to print anything larger, I'll get a larger printer. :-)

I don't know if my use case describes what others may have experienced, but those have been my observations of the day.  I hope they can help illuminate a condition you may not have considered.

~Chris

McSneaky, 2025-02-04:
For me:
- Mid print tap "Stop"
- Tap "Comfirm"
- Wait for bed to go to bottom
- Tap "Home all axis" on screen
- Hear the bed collide with printer 3 times

https://youtube.com/shorts/ldEvPkRdK94?feature=share

ChrisInMilwaukee, 2025-02-04:
> For me:
> 
> * Mid print tap "Stop"
> * Tap "Comfirm"
> * Wait for bed to go to bottom
> * Tap "Home all axis" on screen
> * Hear the bed collide with printer 3 times
> 
> https://youtube.com/shorts/ldEvPkRdK94?feature=share

That aligns with my observations. :-)

sgghostrider, 2025-02-04:
@ChrisInMilwaukee I don't think changing the "position_max:" is advisable, as it will change your max usable printing volume(your slicer will not warn you that you are printing out of boundry/allowed space as the slicer has it's own printing volume setting). Imagine you set it to "position_max: 300" for "sake of safeness", and one day you slice something very tall, let's say 311mm, when the gcode reaches 300mm all "G1 Z" will start failing while "G1 X"s and "G1 Y"s will continue executing, which will make your final 11mm worth of layers to be printed in the same layer, which will make a big mess and fuk your print(or if unlucky break something off the machine).

My personal recommendation is to modify the "macros.cfg" instead, much safer, being it specifically "PRINT_END" and the "G28" override(homing).

ChrisInMilwaukee, 2025-02-04:
> [@ChrisInMilwaukee](https://github.com/ChrisInMilwaukee) I don't think changing the "position_max:" is advisable, as it will change your max usable printing volume(your slicer will not warn you that [...]

Thanks for your observation.  I'll definitely give it a look!

sgghostrider, 2025-02-04:
> > [@ChrisInMilwaukee](https://github.com/ChrisInMilwaukee) I don't think changing the "position_max:" is advisable, as it will change your max usable printing volume(your slicer will not warn you that you are printing out of boundry/allowed space as the slicer has it's own printing volume setting). Imagine you set it to "position_max: 300" for "sake of safeness", and one day you slice something very tall, let's say 311mm, when the gcode reaches 300mm all "G1 Z" will start failing while "G1 X"s and "G1 Y"s will continue executing, which will make your final 11mm worth of layers to be printed in the same layer, which will make a big mess and fuk your print(or if unlucky break something off the machine).
> > My personal recommendation is to modify the "macros.cfg" instead, much safer, being it specifically "PRINT_END" and the "G28" override(homing).
> 
> Thanks for your observation. I'll definitely give it a look. I did look at G28, but don't see where it's doing the z-5 move, so it's possible this is a Beacon thing. I'll keep looking, though. I do see that I can configure the print volume for the printer as well as you described. So I do have some options there. Appreciate you!

What i meant is if you want to modify the Z_MAX value you can do it from the macros, the line that fetches it from your printer config is `{% set max_z = printer.configfile.config["stepper_z"]["position_max"]|float %}` you could do math like this to add an offset `{% set max_z = (printer.configfile.config["stepper_z"]["position_max"]|float) - 19|float %}`(where "19" could be any offset you want to apply ) or just brute force it like this `{% set max_z = 317`(per example, 317 could be anything, per example "306" as you stated in previous comment)

ChrisInMilwaukee, 2025-02-04:
> > > [@ChrisInMilwaukee](https://github.com/ChrisInMilwaukee) I don't think changing the "position_max:" is advisable, as it will change your max usable printing volume(your slicer will not warn you that you are printing out of boundry/allowed space as the slicer has it's own printing volume setting). Imagine you set it to "position_max: 300" for "sake of safeness", and one day you slice something very tall, let's say 311mm, when the gcode reaches 300mm all "G1 Z" will start failing while "G1 X"s and "G1 Y"s will continue executing, which will make your final 11mm worth of layers to be printed in the same layer, which will make a big mess and fuk your print(or if unlucky break something off the machine).
> > > My personal recommendation is to modify the "macros.cfg" instead, much safer, being it specifically "PRINT_END" and the "G28" override(homing).
> > 
> > 
> > Thanks for your observation. I'll definitely give it a look. I did look at G28, but don't see where it's doing the z-5 move, so it's possible this is a Beacon thing. I'll keep looking, though. I do see that I can configure the print volume for the printer as well as you described. So I do have some options there. Appreciate you!
> 
> What i meant is if you want to modify the Z_MAX value you can do it from the macros, the line that fetches it from your printer config is `{% set max_z = printer.configfile.config["stepper_z"]["position_max"]|float %}` you could do math like this to add an offset `{% set max_z = (printer.configfile.config["stepper_z"]["position_max"]|float) - 19|float %}`(where "19" could be any offset you want to apply ) or just brute force it like this `{% set max_z = 317|float %}`(where "317" could be anything, per example "306" as you stated in previous comment)
> 
> Note: This line is mentioned in each command that uses the "position_max" of the "stepper_z", in this case "G28" and "PRINT_END", so to make it effective for both, you have to change it in both macros.

Ah, I see.  Much more elegant approach. :-)  Many thanks!

Phil1988, 2025-02-04:
Thanks for the observations and the video.
That explains, why I couldnt reproduce it.

Its linked to the [scanner]/[beacon]/etc (cartographer/beacon/eddy/bdsensor) -> (eg. horizontal_move_z: 5/home_z_hop: 5/etc)
If introduces additional bed movements after each axis is homed.

For now, I will not change the provided configs, as this is the "pure" config, for a stock system.
Anything added on user side, should be user maintained.

You can solve this in multiple ways:
change the `position_max: 322` in `[stepper_z]`
adjust the z-movement for your axis homings for your probes (if applicable)
change the G28 homing macro, so that no matter what axis you specified, it will always home all
Change the `PRINT_END` macro

I would recommend using @sgghostrider suggestion and use this (`19|float` not needed) :
```
[gcode_macro PRINT_END]
gcode:
    {% set max_x = printer.configfile.config["stepper_x"]["position_max"]|float %}
    {% set max_y = printer.configfile.config["stepper_y"]["position_max"]|float %}
    {% set max_z = (printer.configfile.config["stepper_z"]["position_max"]|float) - 19 %}

    {% if (printer.toolhead.position.z + 12) < max_z %}
        {% set z_hop = 2  %}
        {% set z_safe = max_z - 10 %}
    {% else %}
        {% set z_hop = 0 %}
        {% set z_safe = max_z %}
    {% endif %}

    G91                                                                                                             ; Relative positioning
    G0 Z{z_hop} E-4.0 F3600                                                                                         ; Move nozzle up and retract filament
    G90                                                                                                             ; Absolute positioning
    G0 X{max_x/2} Y{max_y-40} F20000                                                                                ; Move nozzle to remove stringing 
    G0 Z{z_safe} F3600                                                                                              ; Lower bed 
    M400                                                                                                            ; Wait for buffer to clear
    G92 E0                                                                                                          ; Zero the extruder

    M106 P0 S0                                                                                                      ; Turn off part cooling fan
    M106 P2 S0                                                                                                      ; Turn off side fan
    M106 P3 S0                                                                                                      ; Turn off activated charcoal fan

    M104 S0                                                                                                         ; Turn off hotend
    M140 S0                                                                                                         ; Turn off heated bed
    M141 S0                                                                                                         ; Turn off heated chamber

    M220 S100                                                                                                       ; Set feedrate (speed percentage) back to 100%
    M221 S100                                                                                                       ; Set flow percentage back to 100% 

    SET_IDLE_TIMEOUT TIMEOUT={printer.configfile.settings.idle_timeout.timeout}                                     ; Set timeout back to configured value
    CLEAR_PAUSE                                                                                                     ; Ensure pause state is cleared if applicable

    M84                                                                                                             ; Disable steppers
    BEEP I=2 DUR=500                                                                                                ; Alert at the end of the print (if sound is enabled)

```

closed 2025-02-04 by Phil1988