Phil1988/FreeDi #111: [Help] delayed_gcode KINEMATIC_POSITION

KIA1333, 2025-01-23:
In my stock qidi firmware i see line in printer.cfg

[delayed_gcode KINEMATIC_POSITION]
initial_duration:0.2
gcode:
    SET_KINEMATIC_POSITION X=150
    SET_KINEMATIC_POSITION Y=150
	SET_KINEMATIC_POSITION Z=150
    SET_FILAMENT_SENSOR SENSOR=fila ENABLE=0

On new set up it is missing - maybe it is better to add? or skip it?

pmbroth, 2025-01-23:
What machine do you have?

pmbroth, 2025-01-23:
My qidi x-max 3 does not have that in the original configs.

pmbroth, 2025-01-23:
Filamanet sensor is differnt name now    SET_FILAMENT_SENSOR SENSOR=filament ENABLE=1

Different names now
[resonance_tester]
accel_chip:adxl345
probe_points:
    160, 160, 10

KIA1333, 2025-01-23:
> What machine do you have?

MAX3

KIA1333, 2025-01-23:
> The naming conventions are differnt now..
> 
> Filament sensor --> SET_FILAMENT_SENSOR SENSOR=filament ENABLE=1
> 
> KINEMATIC_POSITION] [resonance_tester] accel_chip:adxl345 probe_points: 160, 160, 10

thank you for your time explaining

closed 2025-01-23 by KIA1333

pmbroth, 2025-01-23:
sorry I was still typing the filamanet sensor, and hit enter , then had to reedit, and you replied already..  :-)

pmbroth, 2025-01-23:
Qidi had some bad naming conventions, and mispellings..  I have the qidi x-max 3 and all is working wonderfully with the freedi

Phil1988, 2025-01-24:
I see this issue has already been closed, but I’m curious about the original question:
What exactly were you trying to understand or clarify?

I intentionally avoided using `SET_KINEMATIC_POSITION` and similar functions. 
While I know these are included in the stock configuration, they can be **dangerous**. 
These functions essentially set the current position, even if it **doesn’t reflect the actual position**, which can result in **crashes**.

It took considerable effort to achieve the same (or even better) behavior on FreeDi without relying on these potentially risky functions.

If your question is about the filament sensor:
This part is now (like everything else in FreeDi) properly organized and centrally located in one place.
This makes it easier to maintain an overview and to edit or adapt specific components as needed.
You can find it in the section:
```
########################################
Filament runout sensor
########################################
```

☺️

KIA1333, 2025-01-24:
Everything is more simple than You think. I am in CNC industry technologies 20+ years and just started 3dprinting 3year. I have a nature to check why stuff change and how it benefit user or machine or solution. No need to dig deeper than this :) I just found different commands, then checked for what it is needed and you wrote your approach - that it :)

Phil1988, 2025-01-24:
Ah ok, understandable.
That's how you understand first and then improve :)