Phil1988/FreeDi #113: [Help] What settings to use for network transfer from qidislicer

KIA1333, 2025-01-23:
If i choose in device tab - moonrack - i get connection refure, but test pass.
Or just manual upload possible? From local pc and then with using web interface? 
Qidislicer is not possible to use filetransfer?

robonxt, 2025-01-24:
Make sure you select `Klipper (Moonraker)` in the Physical Printer settings:

[image]

Phil1988, 2025-01-24:
Just added, see [slicer settings](https://github.com/Phil1988/FreeDi/wiki/Post-install-and-config-section#slicer-settings)

closed 2025-01-24 by Phil1988

KIA1333, 2025-01-24:
But even if test show ok - I get error in slicer when opening tab device.

[image]

[image]

KIA1333, 2025-01-24:
But file transfers correctly, maybe qidislicer checks other port than 80?

Phil1988, 2025-01-24:
It works for me.
Did you save the profile?

I can use it in orca and QIDI Slicer with no issues.
Qidi Studio has been reported to cause various issues lately.
What slicer and version are you using exactly?

pmbroth, 2025-01-25:
I just tried Qidi Slicer and get the same isssue. Setup to the printer and test works perfectly.  Try to connect after established printer connection, gets the same issue.  I use Orca Slicer no issues.

Phil1988, 2025-01-25:
QidiSlicer v1.1.6 works

[image]

QidiSlicer v1.2.1 doesnt work

[image]

QidiStudio doesnt work:
It defaults to port 10088 when entering no port
[image]

Attempting to force port 80, it reports a working connection, but its not showing the WebUI (see in background)
[image]

I reported it to both QidiSlicer and QidiStudio and hope that Qidi is fixing it.

Meanwhile: QidiSlicer v1.1.6 / OrcaSlicer / PrusaSlicer / SuperSlicer

pmbroth, 2025-01-25:
Thank you Phil..