Phil1988/FreeDi #115: [Help] Cant  connect to mainsail

Niceemil, 2025-01-24:
Hello i have trouble with my xmax3 install...

Did everything like Phill does in his video. but when I install the firmware with my new emmc i didnt get past the logo of the qidi xmax3 printer, i let i do that for over half an hour and then decided to powercycle ... then i get to the "klipper has an error" messange like in the tutorial. but when connencting to mainsail i get the massage in the browser "error connection timed out." 

i used the /cmd to ping my printer and get varied results 0% 25% 50% and 100% package loss. 

is there a way to reinstall the firmware ? 

/ Emil

Phil1988, 2025-01-24:
Is it connected via ethernet?

You have to finish the whole installation process before the display is functional

Niceemil, 2025-01-24:
it is connected through eathernet cable, 

 @pmbroth but thats not how Phil does it in the video.

Phil1988, 2025-01-24:
> ps. forgot ! thanks for qick respone! and great video [@Phil1988](https://github.com/Phil1988)

You are welcome ;)

So you have flashed the EMMC and you dont have access to the webUI?
Do you use the right IP (no port... no 10088) ?

Show and tell us as much as possible.
The more information we have, the faster and better we can help :)

Niceemil, 2025-01-25:
Hello again! 

Yes, i flashed a brand new EMMC with balena etcher and did go with the 1.30 firmware. 

i made a onedrive album with pictures taken, there one can see that i use the same ip adress like the one that was used with Fluid (the browser still think it leads to fluid hence the logo )  but when i do the arp -a command in cmd i see a different ip-adress. yet non of them works to connect to mainsail.

[https://1drv.ms/f/c/66d51b4da5072751/EuxZFTVdh9hFsIOUWvn8QkkBFUJ6gO841qA6VNf82p5DCg?e=RutLgW](url)

sorry for bad English 
/Emil

Phil1988, 2025-01-25:
For me it seems like you are using a dell docking station on a notebook right?
It looks a bit suspicious to me 😅
Why does the IP address change so often?
169.254.54.150
169.254.124.218
....

So you have a more stable option like a decent router?

rosenrot00, 2025-01-26:
Just connecting you printer to your Windows machine won't assign it an accessible address. What you see here "169...." is an APIPA. What you have to do is share you WiFi internet connection to ethernet so that Windows runs a DHCP server and assigns your printer an IP.

Short, turn on shared internet for ethernet, check which IP your Windows ethernet gets assigned "e.g., 192.168.124.1" and then use a network scanner so scan the 192.168.124 range for active devices. your printer will show up, use that IP address.

Phil1988, 2025-01-26:
All correct.
`arp -a` in cmd is sufficient to check the ip address (no ip scan on dedicated addresses needed)
Tested it last weekend on a win10 notebook. Connected to the printer via ethernet and a shared wifi connection of the notebook.

Niceemil, 2025-01-27:
Hello Everyone ! youre some really clever folks ! I did as @rosenrot00 told me and gott advanced ip scanner and enabled that devices could connect to internet through my computer and the ip-adress did show up !

I have now sucessfully installed FreeDi and i have 1 more question if you dont mind. 

when i try and print, it heats the bed then starts doing the print but it has not heatet the hotend, so i think the old qidi stuff in orcaslicer is wrong now that i´m running pure klipper ? 

this is the fault code i get after i try and print. "The value 'chamber_heater' is not valid for HEATER"

this is my machine start g-code:
PRINT_START
G28
M141 S0
G0 Z50 F600
M190 S[hot_plate_temp_initial_layer]
G28 Z
G29; mesh bed leveling ,comment this code to close it
G0 X0 Y0 Z50 F6000
M191 S{overall_chamber_temperature}
M109 S[nozzle_temperature_initial_layer]
M106 P3 S255
M83
G4 P3000
G0 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0)} Y{max((min(print_bed_max[1], first_layer_print_min[1] + 80) - 85),0)} Z5 F6000
G0 Z[initial_layer_print_height] F600
G1 E3 F1800
G1 X{(min(print_bed_max[0], first_layer_print_min[0] + 80))} E{85 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 Y{max((min(print_bed_max[1], first_layer_print_min[1] + 80) - 85),0) + 2} E{2 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0)} E{85 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 Y{max((min(print_bed_max[1], first_layer_print_min[1] + 80) - 85),0) + 85} E{83 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0) + 2} E{2 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 Y{max((min(print_bed_max[1], first_layer_print_min[1] + 80) - 85),0) + 3} E{82 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0) + 12} E{-10 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 E{10 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000

many thanks and i will buy you a coffee phil when the printer is upp and running :) 

also i used your code for cartographer probe , but i got some errors so I took some of your code and complied some from the cartographer wiki and some from openqidi and now i can do a mesh and a touch of the bedplate 

again sorry for a messy and confused text ! i´m way over my head with this ...

Phil1988, 2025-01-27:
"The value 'chamber_heater' is not valid for HEATER"
Is a name error in the M191 macro and can be easily fixed:
https://github.com/Phil1988/FreeDi/issues/85#issuecomment-2599862221

closed 2025-01-28 by Phil1988

Niceemil, 2025-01-30:
Printer works! Many many thanks!

Phil1988, 2025-01-30:
Great!
I'm very happy having you as a new member of the family and everything is working fine.
Happy printing ;)