Phil1988/FreeDi #116: [Feature request] LCD screen filament pushout screen no temp indicator

KIA1333, 2025-01-25:
**Is your feature request related to a problem? Please describe.**
When in filament extrude meniu - there is no temp indication of current nozzle temp.

**Describe the feature you'd like**
Small indication of nozzle temp in filament pushout screen.

Phil1988, 2025-01-25:
You mean from extruder menu or in the "move&extrude" menu while pausing?

KIA1333, 2025-01-26:
Here is video https://we.tl/t-UussISXu2j?utm_campaign=TRN_TDL_05&utm_source=sendgrid&utm_medium=email&trk=TRN_TDL_05

Here is picture where knowing temp would be great. Loading or unloading filament.
[image]

Phil1988, 2025-01-27:
I agree that it would be handy.
Problem is, that there is simply no space for it.

If you want to adjust the temp, there is the printjob page. 
Not the most confinient I have to admit, but in my opinion totally functional and not worth spending a few more days to include this.

closed 2025-01-27 by Phil1988