Phil1988/FreeDi #117: [Feature request] bet plates save as template

KIA1333, 2025-01-25:
**Is your feature request related to a problem? Please describe.**
I see heightmap save possibility, but as i understand it does not save Z nozzle to plate value - calibration.

**Describe the feature you'd like**
Would like to have possibility to save heightmap + Z calibration value = library of print scheets for quick change.

Phil1988, 2025-01-25:
I'm not aware of such a function existing in klipper, am I wrong?

I don't think the height map is needed at all, as kamp is doing a mesh prior to every print.

As for the z-offset: this might needed to be changed as the coating on each print bed has a different thickness.
The only solution I can think of is a preset on your slicer.
I know it doesn't give you the same flexibility, but I do not see a better solution.

You may (not tested) be able to set a macro for each bed and use it according to the bed.

Coming back to your feature request:
I don't see a possibility yet to implement something like this on the screen firmware.

I'm open to hear anybodies idea on this.

I hope my suggestions can help you

sgghostrider, 2025-01-26:
My recommendation is to save your z_offsets before the "SAVE_CONFIG" and swap them manually when you swap plates, it's the easiest and really the fastest way to do without messing with custom variables/functions, macros or other code, just copy&paste from your list to the "z_offset" inside "SAVE_CONFIG" and click "SAVE & RESTART" button on the top right, it literally takes less than 30 seconds to do, and every time you get a new plate, just calibrate the z_offset again, get the saved value from the generated section and add it to your list with a name you will remember.

Per example this is what my "SAVE_CONFIG" looks like;

`
# Original nozzle QIDI
# z_offset = 1.650 #-Chamaleon Black/Gold PEO
# z_offset = 1.600 #-Carbon Fiber PET
# z_offset = 1.550 #-Starry Sky PET
# z_offset = 2.100 #-Default Plate PEI-PEI (Rought both sides)

# New "upgraded" nozzle from aliexpress
# z_offset = 0.950 #-Carbon Fiber PEO

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [input_shaper]
#*# shaper_type_x = mzv
#*# shaper_freq_x = 66.0
#*# shaper_type_y = mzv
#*# shaper_freq_y = 57.4
#*# damping_ratio_x = 0.058
#*# damping_ratio_y = 0.057
#*#
#*# [probe]
#*# z_offset = 1.600
#*#
#*# [bed_mesh default]
`

IMPORTANT: Put your list as comments AND BEFORE the "SAVE_CONFIG" section, anything after the starting line is wiped and regenerated by klipper, so anything you put below will be deleted eventually.

Phil1988, 2025-01-26:
without the need to change the printer.cfg + restart, a macro would be possible.

Eg. print bed 1 is your standard bed. this is the reference.
You have print bed 2 which is thicker and the nozzle is 0.1mm closer to the bed.

You can create 2 macros like this:.

```
[gcode_macro PRINT_BED_1_REFERENCE]
description: "Sets the Z offset to 0 for bed 1 reference"
gcode:
  # Set the Z offset to 0 for bed 1 and move to apply the offset
  SET_GCODE_OFFSET Z=0 MOVE=1

[gcode_macro PRINT_BED_2]
description: "Sets the Z offset to 0.1 for bed 2"
gcode:
  # Set the Z offset to 0.1 for bed 2 and move to apply the offset
  SET_GCODE_OFFSET Z=0.1 MOVE=1
```

KIA1333, 2025-01-26:
> My recommendation is to save your z_offsets before the "SAVE_CONFIG" and swap them manually when you swap plates, it's the easiest and really the fastest way to do without messing with custom variables/functions, macros or other code, just copy&paste from your list to the "z_offset" inside "SAVE_CONFIG" and click "SAVE & RESTART" button on the top right, it literally takes less than 30 seconds to do, and every time you get a new plate, just calibrate the z_offset again, get the saved value from the generated section and add it to your list with a name you will remember.
> 
> Per example this is what my "SAVE_CONFIG" looks like;
> 
> ```
> # Original nozzle QIDI
> # z_offset = 1.650 #-Chamaleon Black/Gold PEO
> # z_offset = 1.600 #-Carbon Fiber PET
> # z_offset = 1.550 #-Starry Sky PET
> # z_offset = 2.100 #-Default Plate PEI-PEI (Rought both sides)
> 
> # New "upgraded" nozzle from aliexpress
> # z_offset = 0.950 #-Carbon Fiber PET
> 
> #*# <---------------------- SAVE_CONFIG ---------------------->
> #*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
> #*#
> #*# [input_shaper]
> #*# shaper_type_x = mzv
> #*# shaper_freq_x = 66.0
> #*# shaper_type_y = mzv
> #*# shaper_freq_y = 57.4
> #*# damping_ratio_x = 0.058
> #*# damping_ratio_y = 0.057
> #*#
> #*# [probe]
> #*# z_offset = 1.600
> #*#
> #*# [bed_mesh default]
> ```
> 
> IMPORTANT: Put your list as comments AND BEFORE the "SAVE_CONFIG" section, anything after the starting line is wiped and regenerated by klipper, so anything you put below will be deleted eventually.

I see, i have not though that it is not so easy :)

From my side it looks just variable filled in specific field. Let say.
Why it cannot be variable from interface:
[1] > # z_offset = x.xxx #-Original nozzle QIDI after check - SAVE_Restart - autoreboot
[2]> # z_offset = 1.650 #-Chamaleon Black/Gold PEO - SAVE_Restart - autoreboot
[3]> # z_offset = 1.600 #-Carbon Fiber PET - SAVE_Restart - autoreboot
[4]> # z_offset = 1.550 #-Starry Sky PET  - SAVE_Restart - autoreboot
[5]> # z_offset = 2.100 #-Default Plate PEI-PEI (Rought both sides) - SAVE_Restart - autoreboot

But it i think we can stop on this - as for qidi i have just 3 plates so to have value comment uncomment it can be a solution.
Latter i will go with cartographer i have it waiting on table and i see it now have Z offest calc possibilities, will see.

sgghostrider, 2025-01-26:
nono, i think there has been an misunderstanding, those lines actually doesn't do anything, i just use the generated format just for looks, what i actually do is just copy per example the "z_offset = 2.100" part from my own commented line and paste it over the value in the generated section, in this case "z_offset = 1.600".

The problem with what you suggest would be that someone has to program it, klipper doesn't have anything like that(as far as i know at least) so most likely only way would be modifying klipper itself or making a plugin of sorts, way out of the scope of this repo.

Maybe there would be a way with a macro(macros allow user input), that resets the offset to "0" and then sets it to the required value and then saves the settings, but the problem would be the "Set to 0" part, with my limited knowledge i don't know a way to read the current offset and substract it, and as far as i know you cannot just `SET_GCODE_OFFSET Z_ADJUST=0.00` or `SET_GCODE_OFFSET Z=0.00` those commands would do nothing, that functionality is meant for live adjustments, not to actually set the offset, so only positive and negative "offsets" are allowed, which adds or substracts from whatever you have currently.

Phil1988, 2025-01-26:
Why is my suggestion not applicable?

I will close this now to get it off my radar.
Definitely nothing that can be featured here, I'm sorry :)

closed 2025-01-26 by Phil1988