Phil1988/FreeDi #118: [Bug]Screen quit refreshing at start of print.

craftyham, 2025-01-26:
In short the screen quit refreshing & loading/switching to the "print mode" screen at the start of the print process & also no longer displays the title.

The issue is the screen no longer updates to the current "print mode" display. It stays stuck on the main home screen at the start of a print. However, if i pause the print from Orca (2.2.0) as the bed is heating up, it will pause(as it should) when the heating is complete. Then, the screen will update to the current print file & image will be displaied, & allow me to have full control( baby steps ect.). This, issue doesn't not seem to affect the printing process(only the users interaction with the current print. Also. after the print is paused & the display updates & resumed, the file name does not show up. 
Everything was working excellent after switching to FreeDi, so far i have logged almost 35H with zero issues. My last known print without issue was 1/25/25 @0551.
After, that print i proceeded to switch materials, & started to run a first layer test to set babysteps( switched from ASA to PLA). upon doing so, is when i noticed that at the display didn't change to the current print job. 
I have two videos as well which failed to upload.
Discord-@craftyham

[image]

[image]

Phil1988, 2025-01-26:
Okay, let's break this down first, because a lot of different information is coming together here.

1. The fact that the print only pauses after the bed has heated up is the normal and unavoidable behavior of Klipper. So, this is fine.
2. You mentioned starting from Orca, and there were reproducible errors in Orca at least in the dev version 2.2.0. Could you please try uploading the pressure from Orca alone and then start it via Mainsail (using the browser) and let me know if the problem occurs there as well?
3. When the pause is successful you are redirected to the pause. Does the thumbnail and filename display correctly there?

4. You first said, "The screen will update to the current print file & image will be displayed, & allow me to have full control," and later mentioned, "After the print is paused & the display updates & resumed, the file name does not show up." I’m not entirely sure which case happens when exactly.

5. Since everything worked fine up until 1/25/25, and you haven’t updated FreeDi since then (or have you?), the issue is probably related to something else. Please think again about what it could be. My best guess is Orca.

6. Prints that you start directly from the display are not affected, right?

7. When you encounter the error during your tests, please do the following:
```
sudo systemctl stop FreeDi.service
cd ~/FreeDi/FreeDiLCD && ~/klippy-env/bin/python3 start.py -debug
```
Please show me the output from that when you start the print and when you pause it.

Phil1988, 2025-01-28:
@craftyham 
Im still not sure what the cause if this, but you can check if the [latest release](https://github.com/Phil1988/FreeDi/releases/tag/v1.40) fixes this for you :)

craftyham, 2025-01-31:
> [@craftyham](https://github.com/craftyham) Im still not sure what the cause if this, but you can check if the [latest release](https://github.com/Phil1988/FreeDi/releases/tag/v1.40) fixes this for you :)

I apologize for the delay in response & the lack of clarification(I should of waited until more rested). 
Okay so for the first 35hrs it was great. Orca->upload & print--->machine display would update image & data--->happy printing:) Then out of the blue Orca->upload & print--->machine display QUIT updating image after upload & print.
No, if the file is a reprint from the upload fold on the screen it shows up right away as it should(& did from PC). 

Sadly no, i just got finished updating to 1.4.

Connecting to Moonraker at ws://127.0.0.1:80/websocket
Using selector: EpollSelector
TFT baudrate detection...Websocket connected
Moonraker connection opened at ws://127.0.0.1:80/websocket
G-Code file path is: ~/printer_data/gcodes
Found z_offset = 1.649 in variable "z_offset" in section "probe"
Detected baudrate: 921600
Display response: comok 1,1281-0,TJC8048X250_011R,165,10201,47B23401F1DFB95A,33554432-0
Display firmware: v1.40
Printer model: x-max3
Printer volume: 328x326x320
Klipper version: v0.12.0-418-g0114d72a6-dirty
Software version: v1.40 stable
FreeDiLCD start
FreeDi sw-version sent to screen!
self.lcd.data_update()
data.klipper_state != self.printer.klipper_state
klippy ready
Printer state: standby
Printer in standby
Forwarded to main page
_NewMainPage: len: 1 data[0]: ee
ActiveScreen: = MainPage
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
Gcode file .thumbs/2x2x4x_PLA+GT3_17m38s-300x300.png action 'modify_file'
Gcode file .thumbs/2x2x4x_PLA+GT3_17m38s-96x96.png action 'modify_file'
Gcode file .thumbs/2x2x4x_PLA+GT3_17m38s-32x32.png action 'modify_file'
Gcode file 2x2x4x_PLA+GT3_17m38s.gcode action 'modify_file'
Starting new HTTP connection (1): 127.0.0.1:80
http://127.0.0.1:80 "GET /server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode HTTP/11" 404 None
Error: 404 Client Error: Metadata not available for <2x2x4x_PLA GT3_17m38s.gcode> for url: http://127.0.0.1:80/server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode
self.lcd.data_update()
http://127.0.0.1:80 "GET /server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode HTTP/11" 404 None
Error: 404 Client Error: Metadata not available for <2x2x4x_PLA GT3_17m38s.gcode> for url: http://127.0.0.1:80/server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode
http://127.0.0.1:80 "GET /server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode HTTP/11" 404 None
Error: 404 Client Error: Metadata not available for <2x2x4x_PLA GT3_17m38s.gcode> for url: http://127.0.0.1:80/server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode
self.lcd.data_update()
http://127.0.0.1:80 "GET /server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode HTTP/11" 404 None
Error: 404 Client Error: Metadata not available for <2x2x4x_PLA GT3_17m38s.gcode> for url: http://127.0.0.1:80/server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode
http://127.0.0.1:80 "GET /server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode HTTP/11" 404 None
Error: 404 Client Error: Metadata not available for <2x2x4x_PLA GT3_17m38s.gcode> for url: http://127.0.0.1:80/server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode
self.lcd.data_update()
http://127.0.0.1:80 "GET /server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode HTTP/11" 404 None
Error: 404 Client Error: Metadata not available for <2x2x4x_PLA GT3_17m38s.gcode> for url: http://127.0.0.1:80/server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode
http://127.0.0.1:80 "GET /server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode HTTP/11" 404 None
Error: 404 Client Error: Metadata not available for <2x2x4x_PLA GT3_17m38s.gcode> for url: http://127.0.0.1:80/server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode
self.lcd.data_update()
http://127.0.0.1:80 "GET /server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode HTTP/11" 404 None
Error: 404 Client Error: Metadata not available for <2x2x4x_PLA GT3_17m38s.gcode> for url: http://127.0.0.1:80/server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode
http://127.0.0.1:80 "GET /server/files/metadata?filename=2x2x4x_PLA+GT3_17m38s.gcode HTTP/11" 404 None
[... 192 lines cut ...]
extract_printjob_values: self.printer.file_name: 2x2x4x_PLA+GT3_17m38s.gcode
Resetting dropped connection: 127.0.0.1
http://127.0.0.1:80 "GET /server/files/metadata?filename=2x2x4x_PLA%2BGT3_17m38s.gcode HTTP/11" 200 None
Filament Type: PLA+GT3
Estimated Time: 17m 38s
Filament Used (mm): 2524.66
Filament Used (cm): 252 cm
Filament Used (g): 8.2 g
37 ms: Loading GCODE
show_thumbnail: self.printer.file_name: 2x2x4x_PLA+GT3_17m38s.gcode
file: 2x2x4x_PLA%2BGT3_17m38s.gcode
http://127.0.0.1:80 "GET /server/files/thumbnails?filename=2x2x4x_PLA%2BGT3_17m38s.gcode HTTP/11" 200 None
http://127.0.0.1:80 "GET /server/files/gcodes/.thumbs/2x2x4x_PLA+GT3_17m38s-300x300.png HTTP/11" 200 5806
69 ms: Thumbnail Processing
Sending thumbnail to screen
Write thumbnail to LCD
198 ms: Sent thumbnail to LCD
(7200bytes) sent)
Write thumbnail to LCD done!
202 ms: Thumbnail transmission
Forwarded to printjob page
_NewPrintjobPage: len: 1 data[0]: ee
ActiveScreen: = PrintjobPage

I paused the print from orca as the nozzle was heating up, Once that was done the display updated. From there i can resume the print from the display & happy printing...

Phil1988, 2025-01-31:
Ok, this is a parsing issue of the file name. 

You can test it if you delete the "+" sigh from 5 the name.

Phil1988, 2025-01-31:
@craftyham Its fixed now.
Im not sure yet when to enroll it.
I will not include it back to the v1.40 as its too much work for this non critical bug.
For a v1.41 its simply not yet enough changes.

I kindy aks you to meanwhile prevent using the "+" sign in the file name.

closed 2025-01-31 by Phil1988