Phil1988/FreeDi #119: [Help]

KIA1333, 2025-01-26:
If you are facing issues (especially during the first installation), please share important information to let me understand your current status.

[image]

When starting print i get this error in console.
It does not stop printing. 
I am just curious - i am just alone or everyone else also get this?

pmbroth, 2025-01-26:
Check your start Gcode,  e filamant gcode and macros referncing T0, as there should not be aything  there.

Phil1988, 2025-01-26:
Hi KIA1333,  

I’d appreciate it if you could focus on opening issues related specifically to FreeDi.  
As much as I’d love to assist everyone, I simply cannot address or reply to every Klipper/Slicer issue you might encounter. 

For such issues, please consider using a search engine, the Discussions section or Discord.  
I hope you understand and can keep this in mind moving forward.  

### Regarding your problem:  
It seems to be caused by an incorrectly configured slicer.  
Unfortunately, I don’t know which one you’re using, as this crucial information is missing.  

This might help you:  
- [Reddit: Unknown command T0](https://www.reddit.com/r/VORONDesign/comments/qhap88/unknown_commandt0)  
- [GitHub Klipper Issue #2856](https://github.com/Klipper3d/klipper/issues/2856)  

So it's nothing to worry about and can be solved easily :)

closed 2025-01-26 by Phil1988

KIA1333, 2025-01-26:
Thanks for comment, i appreciate your work. 
If it not related to FreeDi - then for sure i will check your reference and search for info.
And yes I use Qidi Slicer 1.21. As you. wrote - it could be something in slicer code.
Just suggestion maybe as you done already some screenshots on wiki how to set up slicer, maybe post settings from your slicer G startup, pause etc. ? I could use as Free.di - good reference point and clear out or mod - what is not needed from qidi stock setup.

jensbodal, 2025-05-02:
I ran into this issue because apparently at some point I messed with multi-material settings in Orca slicer so when I followed the guide I (properly) turned it back off. If you have **Single Extruder Multi-Material** checked but _do not_ select manual filament change, then I assume _it_ assume you will be implementing the T0 macro for tool change.

After you select manual filament change it will comment it out or delete it depending on if you have verbose mode turned on. I recommend turning on verbose mode if you are having issues like this since it makes it trivial to troubleshoot

[image]

```
; MANUAL_TOOL_CHANGE T0 ; change extruder
```

I am not sure if the verbose code causes any issues with the machine parsing the gcode but it will definitely result in larger file sizes so I can't imagine it's something you would want to leave enabled after solving the issue.

Phil1988, 2025-05-02:
Thanks for sharing this and the screenshot will definetly help others thaty might stumble across this too 👍