Phil1988/FreeDi #124: 1.4 Update - Frozen update, data mismaatch error

pmbroth, 2025-01-28:
Hello Phil I updated via Fluid, 

Update Froze, Rebooted and have data error, and then model does not match.  See pic.  What should I do next?

[image]

[image]

[image]

Thank You Brian

pmbroth, 2025-01-28:
i downloaded the stable 1.4 and out on micro sd card to flash the display.  It is flashing  now, will let you know the outcome.  What is the difference between stable and beta?

Phil1988, 2025-01-28:
I am fighting since 6h with git, please be patient I have it soon :D

pmbroth, 2025-01-28:
I flashed manually and back up and running..   Should.  Not an issue.

pmbroth, 2025-01-28:
Do you want me to test the 1.4o Beta,?

pmbroth, 2025-01-28:
When you done, I will have to do a repo sync, since now mine is dirty.  When you get timne, I know you are swampewd - What command should I do later to resync?  Thank you again...

Phil1988, 2025-01-28:
Please use this command line here:
https://github.com/Phil1988/FreeDi/issues/127

Apologies for the delay, but it was quite a bit of work with all the changes.
Thanks for your patience!

Please report back, if it worked (and close with comment of so).

pmbroth, 2025-01-29:
Do I need to do the update per your new release instructions or will
I be fine after running the command line?  As I ran the update per fluid earlier

pmbroth, 2025-01-29:
Gotcha was in men's bible study reading the email.  , I will try now.   :-)

pmbroth, 2025-01-29:
Hello Phil, I ran the update, and validated the printer,cfg and macro.cfg, this is what I get:

[image]

Phil1988, 2025-01-29:
If you did the command linke from my post, it should clean it.
Did you refresh the Update Manager?

pmbroth, 2025-01-29:
@Phil1988   GM Phil, Yes I refreshed, shutdown and rebooted machine.   The command executed.  Should I run it again?  Thank you.

pmbroth, 2025-01-29:
while running got this error, 

[image]

pmbroth, 2025-01-29:
After Rebooting, again, all is good this time in the  updates, repo is clean.    Do I need todo anything for the error that occurred in the execution of the update script?

Brian

Phil1988, 2025-01-29:
> Do I need to do anything for the error that occurred in the execution of the update script?

No this is completely OK.
The wifi module just reports something back that is unknown for usb_modeswitch
You can ditch it, It works anyway.

closed 2025-01-29 by Phil1988

pmbroth, 2025-01-29:
Great - Thank you again!!!

Phil1988, 2025-01-29:
Just edited to say thanks to you too ;)