Phil1988/FreeDi #125: [Bug] "Resume print" button on screen seems to not work correctly

McSneaky, 2025-01-28:
**Describe the bug**
When filament runs out, you load in new filament and then tap "Resume" (play button) on screen, it doesn't seem to actually "resume" printing. It does one movement and then pauses again. Tapping on play button once more makes printer do another movement and pauses again. Going to web UI and clicking "resume" in there worked all good

**Expected behavior**
Resume print work from touchscreen or from first "Resume" click

**To Reproduce**
Steps to reproduce the behavior:
1. Cut the filament before sensor, so printer pauses (haven't tested, if "pause" button also gives same error)
2. Load in new filament
3. Tap on "Resume" (play button) on touchscreen
4. Printer does one movement and then pauses again (can see paused state in Mainsail)
5. Tap on "Resume" again on touchscreen
6. Printer does another movement and then pauses again
7. Click on "Resume" in Mainsail and everything starts to work correctly

**Screenshots or picture**

[image]

**Console output**
If applicable, add the debug console output here (here is a [how to](https://github.com/Phil1988/FreeDi/wiki/Bugs!))

**Base (please complete the following information):**
 - Printer model: X-Max 3
 - X3seriesLCD firmware version: v1.30-0-g388855ca
 - Source: FreeDI
 - Mainboard: Stock Mainboard X-4

McSneaky, 2025-01-28:
I'll try to get it tomorrow :)

KIA1333, 2025-01-29:
I can confirm that i have the same issue.

rosenrot00, 2025-01-30:
I just tried to pause this gcode and it works without an issue. Could you check that one on your end? It just moves the toolhead randomly in a 200x200 space.

[Simulation_PLA_5m.gcode.zip](https://github.com/user-attachments/files/18600865/Simulation_PLA_5m.gcode.zip)

McSneaky, 2025-01-30:
> I just tried to pause this gcode and it works without an issue. Could you check that one on your end? It just moves the toolhead randomly in a 200x200 space.
> 
> [Simulation_PLA_5m.gcode.zip](https://github.com/user-attachments/files/18600865/Simulation_PLA_5m.gcode.zip)

Seems to be issue only when filament runs out. I'll do test with debug logging enabled, will get back in about hour or so

breathless19, 2025-01-31:
I had this issue as well, and yes, it happened when filament ran out too.

Phil1988, 2025-01-31:
I never had a filament runout and cant test it currently.

For me it seems that the stock macro that I included is wrong:

[image]

Please try this and report back:
```
########################################
# Filament runout sensor
########################################
[filament_switch_sensor filament]
pause_on_runout: True
runout_gcode:
            PAUSE
            SET_FILAMENT_SENSOR SENSOR=filament ENABLE=1
event_delay: 3.0
pause_delay: 0.5
switch_pin: !PC1
```

colin-small, 2025-01-31:
On my X max 3 I have a run out sensor and ran out of filament the other day during a long print. From the screen I was able to resume without a problem. I had to heat the nozzle to 215 then remove the old filament, then loaded the new filament and hit play/resume on the screen and it was flawless.

Phil1988, 2025-01-31:
Good point.
Possibly all the reporters were too impatient?

Phil1988, 2025-02-01:
Any news on this @McSneaky @breathless19 ?

breathless19, 2025-02-01:
I haven't run out of filament again yet, I'll let you know :)

Phil1988, 2025-02-04:
Any news on this @McSneaky @breathless19 @KIA1333 ?

breathless19, 2025-02-04:
I'll go ahead and purposefully run out of filament, see what happens, and get back to you soon.  Stay tuned...

McSneaky, 2025-02-04:
Alrighty, just made video of it. Play button in filament management screen is not working

Steps to reproduce:
- Start print
- Pull out filament
- Wait for print to pause
- Go to filament management screen
- Put filament back in
- Tap "Load" on filament managment screen
- Tap "play" on filament managment screen (bug, not working)
- See toolhead just move slightly to left 
- Go back to current print view
- Tap "Play" in there. Everything works correctly

Since I tried to get many bugs into one video. It will take me some minutes to cut out the correct place. Will add video in some minutes

breathless19, 2025-02-04:
Here you go: https://youtu.be/8rXLZWQPO7Y

A few things: 

I thought I remembered it giving some sort of filament run out page, but I was confused when I didn't see that.   I think there should be a filament run out page which includes the current nozzle temp integrated into it, and also a few quick select buttons for temperatures, namely 200, 220, 240. 

I initially had the tiny leftward movement that McSneaky had I believe, but this time it was a nice full lurch to the left.   It doesn't crash into the left side thankfully though if you keep pushing it, otherwise I would have grinded some gears :)   I think after it goes to the left its waiting for it to get to temperature, but I could be wrong.