Phil1988/FreeDi #127: FreeDi v1.40 released - Update instructions

Phil1988, 2025-01-28:
### v1.40 has just been released!

This update includes several "under the hood" changes that can't be updated through Moonraker Update Manager.
I'm really sorry about this - unfortunately, there’s no other way, and I hope you understand
This update changes many things that you will not even notice, but it was needed to enable the desired features.

To take advantage of these, you can either flash the images from the release section or manually add the new features.

This is a How-To for all of you currently using FreeDi:

## I recommend this step for everyone:
The installation requires using the SSH terminal and sudo privileges.
This can be tricky for beginners and prone to errors.
That’s why I’ve bundled everything together.
It includes, among other things:

- Klipper Auto-Flasher
- USB Drive Support
- Enabling Wi-Fi
- Fixing Git Errors
- Cleaning FreeDi
- Cleaning Klipper

Enter this line to the ssh terminal
```
cd ~ && rm -rf FreeDi && git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh
```

You have to confirm the use or the stock (original from Qidi) mainboard and to enter the user password 2 times.
I cant make it easier and I hope its fine for you.

While it installs you can get an idea if whats being done and it may take a while.
I tested it thoroughly (multiple times... coming from different software states) and I hope everything works smoothly.

---

### 📣 To enable announcements 
Add this anywhere to your `moonraker.conf`:
```
[announcements]  
subscriptions:  
  FreeDi  
```

---

### 🌐 To enable Custom Port Feature
Add this to your `[freedi]` section of your `printer.cfg`:
```
# Port of the printer service
port: 80
```
https://github.com/Phil1988/FreeDi/issues/40#issuecomment-2569085096

---

### 🎛️ Adding PID Tuning macros:
[image]
Add this  of your `macros.cfg`:

**Nozzle PID Tuning**
  ```
[gcode_macro NOZZLE_PID_TUNE]  
description: "Perform PID tuning for the nozzle heater"  
`PID_CALIBRATE HEATER=extruder TARGET=220`
```

**Bed PID Tuning**  
```
[gcode_macro BED_PID_TUNE]  
description: "Perform PID tuning for the bed heater"  
`PID_CALIBRATE HEATER=heater_bed TARGET=80`
```
Get the matching macros from [here](https://github.com/Phil1988/FreeDi/blob/master/config_section).  

---

### 🧹 CR+LF/Space Errors
Resolved errors in **PRINT_END** and **CANCEL_PRINT** macros.  

Replace your macro from [here](https://github.com/Phil1988/FreeDi/blob/master/config_section).

---

### 🔄 Chamber Heater Macro Fix
Corrected the **M191 macro** chamber heater reference:

Change the`[gcode_macro M191]` macro of your `macros.cfg`:
Find it [here] [here](https://github.com/Phil1988/FreeDi/blob/master/config_section).

Solves https://github.com/Phil1988/FreeDi/issues/85 https://github.com/Phil1988/FreeDi/issues/115

---

### 🎚️ Screws Tilt Adjust
Updated coordinates for Plus3 and Max3.  

Change the`[screws_tilt_adjust]` section of your `printer.cfg`:
Find it [here](https://github.com/Phil1988/FreeDi/blob/master/config_section).

---

leckert123, 2025-01-28:
Since this is a manual flash update, is it possible to go from a pre 1.20 version directly to this 1.40?

Phil1988, 2025-01-28:
Yes 😊

pmbroth, 2025-01-29:
@Phil,  I think folks will get confused with the Nozzle PID Tuning and Bed PID Tuning above, due to the copy capability of what you have, and that what is there will cause boot up error   In the box,  the bottom code lines  (PID_Calibrate....)  have single parenthesis, see below to see what I mean, highlighted in yellow.

[image]

robonxt, 2025-01-29:
I've updated to 1.40, and then updated FreeDi through moonraker, however the display is still running v1.300. am I missing a step here? (or do I run the install/update.sh in FreeDi folder? I can do that, but just confused since instructions didn't mention anything about it)

Phil1988, 2025-01-29:
@pmbroth 
Yes, thanks.
Edited the post ;)

@robonxt 
Where does it say to run update.sh?
Are you on `stable` in the [freedi] section?

robonxt, 2025-01-29:
> Are you on `stable` in the [freedi] section?

beta. Haven't changed it since last time when testing out v1.300.

Phil1988, 2025-01-29:
> beta. Haven't changed it since last time when testing out v1.300.

Please go to stable :)
Beta is not tested for 1.40

robonxt, 2025-01-29:
No change. TFT firmware is still stuck at v1.300.

Phil1988, 2025-01-29:
@robonxt please open an issue report with a debug console output and background information of the current status.

Phil1988, 2025-01-29:
> @ phil

That's not me btw ;)

robonxt, 2025-01-29:
> please open an issue report with a debug console output and background information of the current status.

@Phil1988 Seems to be working now. I just needed to kill FreeDI.service and run it with debug for it to update the TFT firmware. Weird how it didn't do that after installing v1.4. Will make new issue and post the logs there once it's done.

Phil1988, 2025-01-29:
Reason was a path error because of switched users. Its fixed now and will be pushed when compiling is done (~15min).

breathless19, 2025-01-30:
Is it fair to say that the only changes made to the macros.cfg and printer.cfg files in this update were the changes that you listed here?  Or are there other changes since 1.30 that would make it worthwhile to just replace the whole file(s) with yours?

colin-small, 2025-01-31:
I just upgraded to 1.40 from the previous version and it all went well it seems except for an odd issue where the screen firmware was stuck at 15% and I was forced to shutdown and power on again. After that, it updated as expected. Might be worth mentioning that if people experience this to try shutting down and powering back on again before they open a new issue etc.

Phil1988, 2025-02-01:
> Is it fair to say that the only changes made to the macros.cfg and printer.cfg files in this update were the changes that you listed here?

Yes mainly.
There are only a few spaces I deleted, so of the start, end and cancel macro are working, then its nothing more to do for you :)
Otherwise take the while config.

> except for an odd issue where the screen firmware was stuck at 15%

Yes its known and probably caused by a to weak cpu/ to many start processes (freedi, moonraker, klipper, mainsail, etc. all at the same time) / short overheating after updating. The install script is also quite mighty already. 
Im adding a note to the guide. 
Thanks

Phil1988, 2025-02-01:
Update:
This release - including the images - had 2 small mistakes with a big impact.

No1: A path error caused to not finding the right display firmware. See here https://github.com/Phil1988/FreeDi/issues/127#issuecomment-2621237416 and
https://github.com/Phil1988/FreeDi/issues/128#issuecomment-2621232890
Its now fixed.

No2: Git is not my friend 😅 
While the permissions for klipper_auto_flasher.sh were set correctly:
[image]
https://github.com/Phil1988/FreeDi/issues/130#issuecomment-2624660327
git bash seems to not have it synced despite `git state` showed it was OK.
Permissions have been reset and doesnt look different on my end, bit its now OK:
[image]
Klipper Auto updater should work now.

**I recreated and uploaded images where both issues being fixed.**

**Questions to you:**
Creating images is a time consumung process, as I am doing it live on the machine for every printer.
I am:

1. Flashing the image to the EMMC
2. Changing every setting according to the printer
3. verify it
4. backing up the EMMC
5. compress it as rar
6. uploading it to GitHub
7. Rinse and repeat

If anyone knows a better way to (semi-) automate and accelearate this, I appreciate any hints.

Phil1988, 2025-02-01:
New install instructions added for KAF:
https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#2-your-mainboard-mcu-and-the-toolhead-are-not-on-klipper-v0120-289---and-you-want-it-the-easy-way

[image]

MaZZly, 2025-02-02:
Hi,

I ran the command and it only asked for password once (X-max 3) then finished and told me to restart system.

Screen was showing flashing process so I waited for it to finish, and it started showing 1.40.

Then I restarted the system, and when I check the Machine section I see that FreeDi shows "30 commits available"

Is there something that didn't happen in the install script which it should have? How to proceed?

Phil1988, 2025-02-02:
Did you hit the refresh icon at the update manager?

I tested the script over 100 times and I always had to enter the password twice.
One at the beginning and the second in the middle of this process.

MaZZly, 2025-02-03:
> Did you hit the refresh icon at the update manager?

That did it! Thanks, sorry for the confusion 🙈 (Hopefully this helps someone else...)

> I tested the script over 100 times and I always had to enter the password twice.
One at the beginning and the second in the middle of this process.

I had done a manual `sudo apt upgrade` before, so maybe the initial sudo "password request" was within the timeout for having to type it again.

TopheC, 2025-02-06:
> New install instructions added for KAF: https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#2-your-mainboard-mcu-and-the-toolhead-are-not-on-klipper-v0120-289---and-you-want-it-the-easy-way
> 
.../...

Luckily, i switch to FreeDi my Xmax3 JUST after (02/02) you updated install instructions !! ;-)
So, after my mistake on a reused IP address on my DHCP, everything was OK, FreeDi went flawlessly.
Though my toolhead is different model, maybe more recent, so I put an image here, if you want to add it to your useful wiki : 

[image]

My2cents :-)

Phil1988, 2025-02-06:
Thanks @TopheC for the pic.
I'm not sure if if will every possible component in the wiki cause there are a few ;)
"It might look different for your toolhead, as different revisions have been installed in the machines."

Yours is the same as in my [video](https://youtu.be/_ZmSTJBWUus?t=1220)
Thanks for sharing and I'm glad everything went flawless for you 👍

KIA1333, 2025-02-15:
if i run update connected with putty and login, then do 'sudo su', and paste command line > i get no klipper found.
Need to run update command from first level login, with no 'sudo su'. [Just my notice]

Phil1988, 2025-02-15:
Execute as user, not as su.

Phil1988, 2025-03-01:
As v1.41 has just been released, I'm closing this now 😊

closed 2025-03-01 by Phil1988