Phil1988/FreeDi #128: [Logs for possible bug] Firmware Update not automatic

robonxt, 2025-01-29:
**Describe the bug**
On my X-Smart3, once I followed the upgrade instructions for updating from v1.3X to v1.4, the TFT firmware did not automatically update as mentioned in #127 . I've followed the debug instructions and it was able to update the TFT firmware. 

**Expected behavior**
It should be able to autoupdate, right? unless v1.4 was major changing, the service should had updated the firmware by itself, right?

**To Reproduce**
Steps to reproduce the behavior:
1. Be on v1.30 beta
2. Follow upgrade instructions using the command line.
3. Reboot printer completely
4. Notice TFT firmware is still stuck at v1.300 beta

**Console output**
```command
mks@roboqxs3p:~$ cd ~/FreeDi/FreeDiLCD && ~/klippy-env/bin/python3 start.py -debug
  Connecting to Moonraker at ws://127.0.0.1:80/websocket
  Using selector: EpollSelector
  TFT baudrate detection...Websocket connected
  Moonraker connection opened at ws://127.0.0.1:80/websocket
  Detected baudrate: 921600
  Display response: comok 1,1073-0,TJC4827X243_011R,165,10201,A3B03401CD3CAD50,16777216-0
  G-Code file path is: ~/printer_data/gcodes
  Found z_offset = 2.205 in variable "z_offset" in section "probe"
  Display firmware v1.300 is below required v1.4. Searching for updates...
  Updating firmware using /home/mks/FreeDi/screen_firmwares/x-smart3_stable_v1.40.tft
  Updating firmware... 100%
  156.089 s: Firmware upload complete.
  TFT baudrate detection...Detected baudrate: 9600
  TFT baud will be changed to user setting 921600
  Display response: comok 1,1073-0,TJC4827X243_011R,165,10201,A3B03401CD3CAD50,16777216-0
  Display firmware: v1.40
  Printer model: x-smart3
  Printer volume: 177x185x175
  Klipper version: v0.12.0-289-g08a1c9f12-dirty
  Software version: v1.40 stable
  FreeDiLCD start
  FreeDi sw-version sent to screen!
  self.lcd.data_update()
  data.klipper_state != self.printer.klipper_state
  klippy ready
  Printer state: standby
  Printer in standby
  Forwarded to main page
  _NewMainPage: len: 1 data[0]: ee
  ActiveScreen: = MainPage
  self.lcd.data_update()
  self.lcd.data_update()
  self.lcd.data_update()
[...]

```

**Base (please complete the following information):**
 - Printer model: X-Smart3
 - X3seriesLCD firmware version: v1.4 stable
 - Source: FreeDi image
 - Mainboard: stock X-Smart3 motherboard X-6 V1.0

Phil1988, 2025-01-29:
I found the bug and fixed it.
Compiling right now to push it.

Thanks for your report :)

closed 2025-01-29 by Phil1988

bronsn, 2025-01-30:
Hello, i´m also still on v1.30 at the Display and v1.4 on the System.
Should i do the update process again to fix it ?

regards

Phil1988, 2025-01-30:
@bronsn yes, please do the update again.
The version you currently have cant find the right firmware on your system.

when you enter that line in the terminal the screen firmware should be transmitted.
Occationally this gets stuck, but a simple reboot will solve it.

I recomment to reboot in all cases.