Phil1988/FreeDi #129: [SCREWS_TILT_CALCULATE] for x-smart 3

Algorus21, 2025-01-29:
The screw coordinates are included in the printer.cfg of the plus3 and Max3 but not for the Smart-3, which does not allow the use of the SCREWS_TILT_CALCULATE command

Here are the ones I found on the smart 3, using the location of the original probe (when it is located to the right of the nozzle):

[screws_tilt_adjust]
horizontal_move_z: 5
screw_thread: CCW-M4
speed: 300

screw1: 4, 29
screw1_name: front left screw
screw2: 124, 29
screw2_name: front right screw
screw3: 124, 149
screw3_name: rear right screw
screw4: 4, 149
screw4_name: rear left screw

Phil1988, 2025-01-29:
Thanks mate!

I was looking and asking for this many times without success 👍 

Just [added](https://github.com/Phil1988/FreeDi/commit/9b766e8a10cd963b19c4b27bf88aac1158c6429b) and included it on the [latest release](https://github.com/Phil1988/FreeDi/releases/tag/v1.40)

Unfortunately this is not included in the X-Smart3 image, as my time machine is broken 😅

closed 2025-01-29 by Phil1988

Algorus21, 2025-01-29:
You're welcome. This is my very modest contribution to Freedi ;)