Phil1988/FreeDi #131: Shutdown on every print after upgrade

bronsn, 2025-01-30:
Hello,

i upgrade my FreeDi sys from 1.3 to 1.4 like u descripe in your Update Guide.

If i upload a print from my slicer and try to print, the printer (X Smart 3) shuts down with following Error (.

[image]

rosenrot00, 2025-01-30:
Can you share your gcode here to have it tested?

Phil1988, 2025-01-30:
can you share how many being "many"?

might be related to this?
https://github.com/Phil1988/FreeDi/issues/89

bronsn, 2025-01-30:
sure :)

[boalded_rope_parts_PETG_6h10m.zip](https://github.com/user-attachments/files/18603043/boalded_rope_parts_PETG_6h10m.zip)

Phil1988, 2025-01-30:
I can only see one object:
boalded_rope_parts.stl_id_0_copy_0

please share a screenshot of the slicers print bed or better the slicer saved project file (and name the slicer + the version) please

bronsn, 2025-01-30:
[boalded_rope_parts_OrcaSlicer2.2.0.zip](https://github.com/user-attachments/files/18603415/boalded_rope_parts_OrcaSlicer2.2.0.zip)

bronsn, 2025-01-30:
hmm ok so i found a mistake, perhaps the hardware from the smart is too weak, i printed only on my Vorons the last years ).

The file i send too you is one print file with 150 small objects , the first try i uploaded the same file but with 150 single Objects of these small parts, so perhaps the printer cant handle this.

Sry for your time, i´m sure you have enough work.

Phil1988, 2025-01-30:
> Sry for your time, i´m sure you have enough work.

No problem at all, I really appreciate you thinking about my time! 😊

I was  just wondering that you said "Yesterday, with System v1.3 i did `the same prin`t without trouble".

Im glad you found it was not the same (but similar) and I totally understand that hardware can be a bit tricky sometimes. 
If you need help or want to test anything in the future, just let me know!

closed 2025-01-30 by Phil1988