Phil1988/FreeDi #132: [Bug]

bronsn, 2025-01-31:
Hello Phil, (XSmart 3)

since the Update tp 1.4 the Heater and Bed wont Cooldown after a Print or after i cancel a Print.
I use your Macros.

[gcode_macro PRINT_END]
gcode:
    {% set max_x = printer.configfile.config["stepper_x"]["position_max"]|float %}
    {% set max_y = printer.configfile.config["stepper_y"]["position_max"]|float %}
    {% set max_z = printer.configfile.config["stepper_z"]["position_max"]|float %}

    {% if (printer.toolhead.position.z + 12) < max_z %}
        {% set z_hop = 2  %}
        {% set z_safe = max_z - 10 %}
    {% else %}
        {% set z_hop = 0 %}
        {% set z_safe = max_z %}
    {% endif %}
    
    G91                                                                                                             ; Relative positioning
    G0 Z{z_hop} E-4.0 F3600                                                                                         ; Move nozzle up and retract filament
    G90                                                                                                             ; Absolute positioning
    G0 X{max_x/2} Y{max_y-40} F20000                                                                                ; Move nozzle to remove stringing 
    G0 Z{z_safe} F3600                                                                                              ; Lower bed 
    M400                                                                                                            ; Wait for buffer to clear
    G92 E0                                                                                                          ; Zero the extruder
    M106 P0 S0                                                                                                      ; Turn off part cooling fan
    M106 P2 S0                                                                                                      ; Turn off side fan
    M106 P3 S0                                                                                                      ; Turn off activated charcoal fan
    M104 S0                                                                                                         ; Turn off hotend
    M140 S0                                                                                                         ; Turn off heated bed
    M220 S100                                                                                                       ; Set feedrate (speed percentage) back to 100%
    M221 S100                                                                                                       ; Set flow percentage back to 100% 
    SET_IDLE_TIMEOUT TIMEOUT={printer.configfile.settings.idle_timeout.timeout}                                     ; Set timeout back to configured value
    CLEAR_PAUSE                                                                                                     ; Ensure pause state is cleared if applicable
    M84                                                                                                             ; Disable steppers
    BEEP I=2 DUR=500                                                                                                ; Alert at the end of the print (if sound is enabled)
    SET_SKEW CLEAR=1

Phil1988, 2025-01-31:
Looks good to me.
The update didn't touch the macros or your config.

So it's caused by something different.

Please heat up the bed and use this in the console in mainsail:                                                                                     
M140 S0                                                                   

Show and tell me what it does and pleas give this issue a better name.

bronsn, 2025-01-31:
Hey ty for answering that fast )

I try the Commands manually and it works , only after a print it didnt.

The Xsmart3 do everything in the PRINT_END Command except cooldown Heater and Bed so i´m a bit Confused.
Before the Update it worked and i did nothing else only updated to 1.4

pmbroth, 2025-01-31:
Did you change anything in the end gcode in slicer?

bronsn, 2025-01-31:
nothing, after the print he do 80% of the End Code, he lower the Bed like in the End G code only the Temp of the BEd and Hotend stay at the print Temp.

I changed nothing else, i did the update , twice , cause after the first update the Display was still on 1.3 so i read that u fixed it and so i did the update again.
And now, after the update, i have this issue and have no clue why he ignore the M104 in the End Code.

pmbroth, 2025-01-31:
hello - post your end gcode  from the slicer

Phil1988, 2025-01-31:
Please attach the whole marco.cfg here.
Name it as .txt or zip it.
I am pretty sure it's a LF error or something like this.

bronsn, 2025-01-31:
> hello - post your end gcode from the slicer

in my Slicers End Gcode is only PRINT_END , in it.

here is my macro.cfg

[macrocfg_xsmart3.zip](https://github.com/user-attachments/files/18623904/macrocfg_xsmart3.zip)

Phil1988, 2025-02-01:
Please rename your existing macros.cfg on the printer to macros_bak.cfg.

Download this:
[macrocfg_xsmart3_modified.zip](https://github.com/user-attachments/files/18627927/macrocfg_xsmart3_modified.zip)

Extract it and upload the macros.cfg to your printer.

Then test it again and report back if it works for you :)

bronsn, 2025-02-01:
Hello and ty,

it took a while i printed today but now i have done a print and it dont work :(

[image]

Phil1988, 2025-02-01:
What slicer do you use?
Which version?
Can you save the project (in slicer) and attach it here?

bronsn, 2025-02-01:
i use Orca Slicer 2.2.0 like before i updated to the newest FreeDi Version.

Project? This issue is at every print, actually i print parts of the Dummy13, but testd it with other prints its not the projects fault, so i think its useless to upload the dummy 13 ?

Phil1988, 2025-02-01:
Well I don't have that problem.
I can only help you efficiently if you can provide the project file.

Do you know how to do that?
On the upper left on orca you can save the current slice job as project file.
I need it to open it and see the slicer settings you have.

bronsn, 2025-02-03:
sry i was very busy at the weekend so i cant answer.

Here is one of the files i printed:

[frame_g_1.zip](https://github.com/user-attachments/files/18640450/frame_g_1.zip)

Phil1988, 2025-02-03:
> sry i was very busy at the weekend so i cant answer.

Thanks.
It looks "OK" (could be optimized, but nothing problematic).

I think I shouldnt have listened to you :D

Please send your printer.cfg.

Thats the problem ;)

Phil1988, 2025-02-03:
You missed to use the provided printer.cfg from the config section or even deleted mine provided with the image ;)

Please replace your fan2 section (line 267) with mine:
https://github.com/Phil1988/FreeDi/blob/99a270990a484694021d32f8114db6586426dbac/config_section/X-Plus3/printer.cfg#L268

I will close this now to get it off my radar, but please report back that it works ;)

closed 2025-02-03 by Phil1988

bronsn, 2025-02-03:
but you posted the config fot the xplus3, i use the X-Smart3 ?

bronsn, 2025-02-03:
and no, it doesnt work, Temp wont shut off after a print or cancel a print

Phil1988, 2025-02-03:
> but you posted the config fot the xplus3, i use the X-Smart3 ?

You are right. I was on the road and just did quick search for your given PA8 pin.
wouldt hurt in that case, but taking the config from the X-Smart would be better.

>  it doesnt work

Does Mainsail show anything? 
"doesnt work" also "doesnt work" for helping you ;)

bronsn, 2025-02-03:
> > it doesnt work
> 
> Does Mainsail show anything? "doesnt work" also "doesnt work" for helping you ;)

i know sry, but there are no Error Warnings, no messages in Mainsail.
The printer does his job until the print is rdy, he starts the END_PRINT Macro normal but without the M104 and M140 and i have absolute no clue where i have too look cause this issue exist after the update to the newest FreeDi Version.

I can manually type M104 S0 into the Console and it works, but in the Macro it wont ?

Phil1988, 2025-02-03:
Tested it on my end and it works flawless:

[image]

bronsn, 2025-02-03:
my actual print lasts 29min , i will test it immediately after he is done.

Ty for your help

Phil1988, 2025-02-03:
Sorry @bronsn but I dont trust you 😅
You either are messing with the files you send me or you changed them afterwards.
Its completely impossible to get what you show me if you have the printer.cfg and the macro.cfg you send me here.
Did you took the template of Orca for the M106 by any chance?

bronsn, 2025-02-03:
i know its hard to believe but i changed nothing, i use those i have sent u before.
Here i send you both again, i downloaded them right now from the printer.

i´m confused by my own

[macrosandprintercfg.zip](https://github.com/user-attachments/files/18646584/macrosandprintercfg.zip)

Phil1988, 2025-02-03:
No worries, lets un-confuse together ;)

rename your macros.cfg-> macros_old.cfg or delete it.
I stoped checking your macros.cfg when I found 4 issues but I changed one thing on my macros.cfg.

Use this:
[macros.cfg](https://github.com/Phil1988/FreeDi/blob/master/config_section/X-Smart3/macros.cfg)

Fingers crossed that it works now

bronsn, 2025-02-03:
ty i will test it after dinner :)

One Question, your print_start in the macros.cfg is empty, it´s ok if i fill it with my code? 

gcode:
  SKEW_PROFILE LOAD=CaliFlower
  #### set defaults ####
   {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(150)|float %}
   {% set BED_TEMP = params.BED_TEMP|default(60)|float %}  
  #### end off definition  ####
  M104 S{EXTRUDER_TEMP}               ; set Hotend Temp
  M140 S{BED_TEMP}                    ; set Bed Temp 
  BED_MESH_CLEAR
  G28                                 ; Home Printer
  M83                                 ; relative Mode
  G92 E0                              ; Reset Extruder 
  #G1 X10 Y10 Z50 F3000                ; Move to wait position
  Smart_Park
  G90                                 ; absolute Mode
  BED_MESH_PROFILE LOAD="default"     ; load Mesh Profile
  M190 S{BED_TEMP}                    ; wait for Bed Temp
  M109 S{EXTRUDER_TEMP}               ; wait for Nozzle Temp  
  Line_Purge

bronsn, 2025-02-03:
it work now correct !

Nice, what did u changed ?

I will support your work , great job and thank you very much

Phil1988, 2025-02-04:
> Nice, what did u changed ?

I just replaced your macros.cfg with the one of my config section :)
On this I commented out 
https://github.com/Phil1988/FreeDi/blob/cfbe060caf710e631aa1f46f637c24b0b3f74000/config_section/X-Smart3/macros.cfg#L127C6-L127C10

This line is not needed for the X-Smart3 but I also doesnt cause an issue when its in there.
I just tested it on my side to verify:
[image]

So there was at least one (other) error in there responsible for your problem :)

Thanks for your support 🙏

---

Regarding the start macro:
Im not sure if `Smart_Park` and `Line_Purge` will work.

The rest should do, if you like to have your slicer start gcode with only call `PRINT_START`.

Just make sure to not do some of the things twice and test it thoroughly ;)