Phil1988/FreeDi #134: [Bug] Mismatch between mainsail reported FreeDi version and actual version

breathless19, 2025-02-01:
**Describe the bug**
Mainsail reports that FreeDi is v1.30-76-g3cd04218, when my machine has already been updated to 1.40, and it correctly shows as 1.40 on the machine "about" page

**Expected behavior**
Both version numbers should match

**To Reproduce**
Steps to reproduce the behavior:
1. Check software version on both mainsail and compare it with the version number shown on your Machines "about" page. 

[image]

[image]

 - Printer model: X-Max 3
 - X3series LCD firmware version: [ v1.40 stable]

Phil1988, 2025-02-01:
Tried the refresh button in the update manager?
It's intervals for checking that is not super high by default

breathless19, 2025-02-01:
Yea, I tried that several times:   https://youtu.be/r8jKyh2R3VE

rosenrot00, 2025-02-02:
Can you share the FreeDi section from your Moonraker.conf?

Phil1988, 2025-02-02:
Additionall that would help:
Open the terminal and do
```
cd ~/FreeDi  

git status  

git log -1 --oneline  

git remote show origin  

git log --oneline HEAD..origin/master  
```

Please show the outputs.

breathless19, 2025-02-03:
Here is my Moonraker.conf: 

**#####################################################################
# Announcments
#####################################################################
[announcements]  
subscriptions:  
  FreeDi  

#####################################################################
# Basics
#####################################################################
[server]
host: 0.0.0.0
port: 7125
klippy_uds_address: /home/mks/printer_data/comms/klippy.sock

[authorization]
trusted_clients:
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16
    FE80::/10
    ::1/128
cors_domains:
    *.lan
    *.local
    *://localhost
    *://localhost:*
    *://my.mainsail.xyz
    *://app.fluidd.xyz

#####################################################################
# Features
#####################################################################
[octoprint_compat]

[history]

# For kemp, see https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging
[file_manager]
enable_object_processing: True

#####################################################################
# Update Mangers
#####################################################################
[update_manager]
channel: stable
refresh_interval: 168

# Mainsail update_manager entry
[update_manager mainsail]
type: web
channel: stable
repo: mainsail-crew/mainsail
path: ~/mainsail

# FreeDi update_manager entry
[update_manager FreeDi]
type: git_repo
path: ~/FreeDi
channel: stable
origin: https://github.com/Phil1988/FreeDi
virtualenv: ~/klippy-env
requirements: requirements.txt
managed_services: FreeDi
info_tags:
	desc=FreeDi
	sparse_dirs:
	- FreeDiLCD
	- screen_firmwares

# Crowsnest update_manager entry
[update_manager crowsnest]
type: git_repo
path: ~/crowsnest
origin: https://github.com/mainsail-crew/crowsnest.git
managed_services: crowsnest
install_script: tools/pkglist.sh

[update_manager cartographer]
type: git_repo
path: ~/cartographer-klipper
channel: stable
origin: https://github.com/Cartographer3D/cartographer-klipper.git
is_system_service: False
managed_services: klipper
info_tags:
  desc=Cartographer Probe**

And attached is a picture of the outputs to the requested commands.   The last command didn't seem to do anything that I can see. 

Thanks!

[image]

Phil1988, 2025-02-03:
Thanks.
Its all looking good on your side 👍 

Please redo:
```
cd ~ && rm -rf FreeDi && git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh
```

I know you have been doing it, but your current version was intentionally deactivated to be sure every 1.40 user really has the working klipper_auto_flasher tool.

closed 2025-02-03 by Phil1988

breathless19, 2025-02-03:
Redo-ing that command fixed the discrepancy, and my FreeDi now shows the correct version.  Thanks!

I did see one error message during the process... I assume its nothing important:  "can't use storage command in..."

[image]

Phil1988, 2025-02-04:
That is normal.
Nothing to worry about :)