Phil1988/FreeDi #135: [Bug] Probe triggered prior to movement

MaZZly, 2025-02-02:
**Describe the bug**

Full bed mesh finishes with `Probe triggered prior to movement`

**Expected behavior**

Bed mesh calibration to work without throwing that error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to HeightMap
2. Click on Calibrate
3. Wait to finish
4. See error `Probe triggered prior to movement`

**Console output**
```
8:21 AM Probe triggered prior to movement
8:21 AM probe at 251.093,305.601 is z=1.854000
8:21 AM probe at 251.093,305.601 is z=1.854000
8:21 AM probe at 215.219,305.604 is z=1.814000
8:21 AM probe at 215.219,305.604 is z=1.804000
8:21 AM probe at 179.351,305.600 is z=1.856500
8:20 AM probe at 179.351,305.600 is z=1.856500
8:20 AM probe at 143.478,305.603 is z=1.871500
8:20 AM probe at 143.478,305.603 is z=1.871500
8:20 AM probe at 107.610,305.599 is z=1.901500
8:20 AM probe at 107.610,305.599 is z=1.899000
8:20 AM probe at 71.736,305.602 is z=1.906500
8:20 AM probe at 71.736,305.602 is z=1.904000
8:20 AM probe at 35.869,305.599 is z=1.916500
8:20 AM probe at 35.869,305.599 is z=1.911500
8:20 AM probe at 0.001,305.595 is z=1.949000
8:20 AM probe at 0.001,305.595 is z=1.944000
8:20 AM probe at 0.004,267.849 is z=1.819000
8:20 AM probe at 0.004,267.849 is z=1.811500
8:20 AM probe at 35.872,267.852 is z=1.809000
8:20 AM probe at 35.872,267.852 is z=1.801500
8:20 AM probe at 71.739,267.856 is z=1.811500
8:20 AM probe at 71.739,267.856 is z=1.804000
8:20 AM probe at 107.613,267.853 is z=1.809000
8:20 AM probe at 107.613,267.853 is z=1.801500
8:20 AM probe at 143.475,267.850 is z=1.799000
8:20 AM probe at 143.475,267.850 is z=1.791500
8:20 AM probe at 179.348,267.847 is z=1.784000
8:20 AM probe at 179.348,267.847 is z=1.779000
8:20 AM probe at 215.216,267.851 is z=1.766500
8:20 AM probe at 215.216,267.851 is z=1.761500
8:20 AM probe at 251.090,267.848 is z=1.736500
8:20 AM probe at 251.090,267.848 is z=1.734000
8:20 AM probe at 286.957,267.852 is z=1.896500
8:20 AM probe at 286.957,267.852 is z=1.896500
8:20 AM probe at 286.961,230.105 is z=1.751500
8:20 AM probe at 286.961,230.105 is z=1.749000
8:20 AM probe at 251.093,230.102 is z=1.779000
8:20 AM probe at 251.093,230.102 is z=1.771500
8:20 AM probe at 215.219,230.104 is z=1.804000
8:20 AM probe at 215.219,230.104 is z=1.794000
8:20 AM probe at 179.352,230.101 is z=1.821500
8:20 AM probe at 179.352,230.101 is z=1.811500
8:20 AM probe at 143.478,230.104 is z=1.821500
8:20 AM probe at 143.478,230.104 is z=1.809000
8:20 AM probe at 107.610,230.100 is z=1.829000
8:20 AM probe at 107.610,230.100 is z=1.821500
8:20 AM probe at 71.743,230.097 is z=1.821500
8:20 AM probe at 71.743,230.097 is z=1.811500
8:20 AM probe at 35.869,230.099 is z=1.816500
8:20 AM probe at 35.869,230.099 is z=1.809000
8:20 AM probe at 0.001,230.096 is z=1.799000
8:20 AM probe at 0.001,230.096 is z=1.791500
8:20 AM probe at 0.004,192.349 is z=1.734000
8:20 AM probe at 0.004,192.349 is z=1.734000
8:20 AM probe at 35.872,192.353 is z=1.741500
8:20 AM probe at 35.872,192.353 is z=1.741500
8:20 AM probe at 71.746,192.350 is z=1.756500
8:20 AM probe at 71.746,192.350 is z=1.754000
8:20 AM probe at 107.607,192.347 is z=1.774000
8:20 AM probe at 107.607,192.347 is z=1.766500
8:20 AM probe at 143.475,192.351 is z=1.774000
8:20 AM probe at 143.475,192.351 is z=1.769000
8:20 AM probe at 179.349,192.348 is z=1.774000
8:20 AM probe at 179.349,192.348 is z=1.769000
8:20 AM probe at 215.216,192.352 is z=1.764000
8:20 AM probe at 215.216,192.352 is z=1.761500
8:20 AM probe at 251.090,192.349 is z=1.744000
8:20 AM probe at 251.090,192.349 is z=1.744000
8:20 AM probe at 286.958,192.352 is z=1.731500
8:20 AM probe at 286.958,192.352 is z=1.731500
8:20 AM probe at 286.955,154.600 is z=1.706500
8:20 AM probe at 286.955,154.600 is z=1.704000
8:20 AM probe at 251.093,154.602 is z=1.726500
8:20 AM probe at 251.093,154.602 is z=1.726500
8:20 AM probe at 215.219,154.605 is z=1.744000
8:20 AM probe at 215.219,154.605 is z=1.744000
8:19 AM probe at 179.352,154.602 is z=1.744000
8:19 AM probe at 179.352,154.602 is z=1.744000
8:19 AM probe at 143.484,154.598 is z=1.744000
8:19 AM probe at 143.484,154.598 is z=1.741500
8:19 AM probe at 107.610,154.601 is z=1.731500
8:19 AM probe at 107.610,154.601 is z=1.731500
8:19 AM probe at 71.743,154.597 is z=1.704000
8:19 AM probe at 71.743,154.597 is z=1.696500
8:19 AM probe at 35.869,154.600 is z=1.671500
8:19 AM probe at 35.869,154.600 is z=1.664000
8:19 AM probe at 0.001,154.597 is z=1.641500
8:19 AM probe at 0.001,154.597 is z=1.636500
8:19 AM probe at 0.004,116.850 is z=1.564000
8:19 AM probe at 0.004,116.850 is z=1.564000
8:19 AM probe at 35.872,116.854 is z=1.594000
8:19 AM probe at 35.872,116.854 is z=1.594000
8:19 AM probe at 71.740,116.845 is z=1.631500
8:19 AM probe at 71.740,116.845 is z=1.629000
8:19 AM probe at 107.607,116.848 is z=1.669000
8:19 AM probe at 107.607,116.848 is z=1.664000
8:19 AM probe at 143.481,116.845 is z=1.694000
8:19 AM probe at 143.481,116.845 is z=1.686500
8:19 AM probe at 179.349,116.849 is z=1.699000
8:19 AM probe at 179.349,116.849 is z=1.691500
8:19 AM probe at 215.216,116.852 is z=1.701500
8:19 AM probe at 215.216,116.852 is z=1.694000
8:19 AM probe at 251.090,116.850 is z=1.699000
8:19 AM probe at 251.090,116.850 is z=1.694000
8:19 AM probe at 286.958,116.853 is z=1.696500
8:19 AM probe at 286.958,116.853 is z=1.691500
8:19 AM probe at 286.955,79.100 is z=1.676500
8:19 AM probe at 286.955,79.100 is z=1.671500
8:19 AM probe at 251.087,79.097 is z=1.689000
8:19 AM probe at 251.087,79.097 is z=1.681500
8:19 AM probe at 215.226,79.100 is z=1.699000
8:19 AM probe at 215.226,79.100 is z=1.691500
8:19 AM probe at 179.352,79.102 is z=1.684000
8:19 AM probe at 179.352,79.102 is z=1.676500
8:19 AM probe at 143.484,79.099 is z=1.659000
8:19 AM probe at 143.484,79.099 is z=1.651500
8:19 AM probe at 107.611,79.102 is z=1.626500
8:19 AM probe at 107.611,79.102 is z=1.626500
8:19 AM probe at 71.743,79.098 is z=1.586500
8:19 AM probe at 71.743,79.098 is z=1.586500
8:19 AM probe at 35.869,79.101 is z=1.529000
8:19 AM probe at 35.869,79.101 is z=1.524000
8:19 AM probe at 0.001,79.097 is z=1.479000
8:19 AM probe at 0.001,79.097 is z=1.476500
8:19 AM probe at 0.005,41.351 is z=1.389000
8:19 AM probe at 0.005,41.351 is z=1.384000
8:19 AM probe at 35.866,41.348 is z=1.431500
8:19 AM probe at 35.866,41.348 is z=1.431500
8:19 AM probe at 71.740,41.345 is z=1.494000
8:19 AM probe at 71.740,41.345 is z=1.491500
8:19 AM probe at 107.607,41.349 is z=1.551500
8:19 AM probe at 107.607,41.349 is z=1.544000
8:19 AM probe at 143.481,41.346 is z=1.589000
8:19 AM probe at 143.481,41.346 is z=1.589000
8:19 AM probe at 179.349,41.350 is z=1.614000
8:19 AM probe at 179.349,41.350 is z=1.616500
8:19 AM probe at 215.223,41.347 is z=1.634000
8:19 AM probe at 215.223,41.347 is z=1.634000
8:19 AM probe at 251.090,41.350 is z=1.656500
8:19 AM probe at 251.090,41.350 is z=1.651500
8:19 AM probe at 286.958,41.354 is z=1.661500
8:19 AM probe at 286.958,41.354 is z=1.656500
8:19 AM probe at 286.955,3.601 is z=1.636500
8:19 AM probe at 286.955,3.601 is z=1.641500
8:19 AM probe at 251.087,3.598 is z=1.646500
8:19 AM probe at 251.087,3.598 is z=1.646500
8:19 AM probe at 215.220,3.594 is z=1.626500
8:18 AM probe at 215.220,3.594 is z=1.634000
8:18 AM probe at 179.352,3.603 is z=1.626500
8:18 AM probe at 179.352,3.603 is z=1.634000
8:18 AM probe at 143.485,3.600 is z=1.579000
8:18 AM probe at 143.485,3.600 is z=1.576500
8:18 AM probe at 107.611,3.602 is z=1.536500
8:18 AM probe at 107.611,3.602 is z=1.529000
8:18 AM probe at 71.743,3.599 is z=1.441500
8:18 AM probe at 71.743,3.599 is z=1.446500
8:18 AM probe at 35.869,3.602 is z=1.376500
8:18 AM probe at 35.869,3.602 is z=1.369000
8:18 AM probe at 0.002,3.598 is z=1.306500
8:18 AM probe at 0.002,3.598 is z=1.304000
8:18 AM BED_MESH_CALIBRATE PROFILE="test"
```

**Base (please complete the following information):**
 - Printer model: X-max 3
 - X3seriesLCD firmware version: 1.30 (?)
 - Source: FreeDi image
 - Mainboard: Stock

MaZZly, 2025-02-02:
Updated to 1.40 and it still happens there also.

Phil1988, 2025-02-02:
It's not a bug but a mistake in your config.
And your bed looks very tilted (1.3-1.95).

I am missing information. 
Does your max3 has a inductive probe or bl touch?

As a temporary fix you can increase 
horizontal_move_z: 2.6   
To eg 4.0

But I recommend, doing my recommendations at the first steps section ;)

MaZZly, 2025-02-03:
I added the temporary fix and reran to see that there was a huge peak in the rear right corner... 
I took off the plate and made sure to clean so nothing was there under, and then it gave saner results (still need to do bed leveling, but "only" ~0.57mm variance now :) 

**Improvement suggestion**
About this section: https://github.com/Phil1988/FreeDi/wiki/Bed-leveling#3-adjust-the-screws, it could maybe have links to video instructions for various printers on where to find the screws?

For X-max3 (at least on mine), there are lock bolts underneath that need loosening, and the screws are located under the plate... There is a video showing how to adjust here: https://drive.google.com/drive/folders/1X7hf3yTCm68X2Xyft8-adGz3cFb6TUWa (basic manual leveling.. But at least shows the screw locations etc..)

Phil1988, 2025-02-03:
I like your suggestion.
However, as I don't own all the models and am only one person with a limited amount of time, I'm happy to return the suggestion ;)

Why don't you take this as an opportunity to take a video or pictures of the process so that others who are unsure can see how it is done? ;

pmbroth, 2025-02-03:
It seems if you do seraches on youtube, ther are several videos out there already.

Phil1988, 2025-02-04:
As its solved, I will close this now ;)

Happy printing!

@MaZZly If you do a video on this, contact me, so I can place it to the wiki.

closed 2025-02-04 by Phil1988