Phil1988/FreeDi #138: [Bug] [x-Smart 3] Catastrophic Upgrade to v1.40 (bricked)

theclue, 2025-02-03:
Tonite I decided to finally upgrade to v1.40.

I followed the procedure and everything seems to go ok, no error in the install.sh script execution.

I make a backup of my cfg files and also provided to soft recovery the dirty repository in update manager. Everything went ok there

I realized the screen was not flashing, so I reboot the host to make the screen starts flashing, and it does. But the flashing ended with "data error", Moonraker cannnot connect to Klipper anymore and it doesn't werk consistely anyway (after few seconds it goes down). I also cannot ssh to the machine.

If i reboot i sometimes get "Data Error" on screen, sometimes it starts flashing again but stops at 8%

Long story short, printer is bricked and I've no idea on where to start trying to fix it.

Hope you have some ideas, ad I have not :(

El-Keldo, 2025-02-03:
Hello.
I'm not sure it is related but I'm also experiencing issues with FreeDi 1.40 since I installed it 2 days ago.

My system : 
- X-Max-3 (a quite early model that recieved some upgrades, but I'm not the first owner, so I do not have full story)
- X4 v1.1 mainboard (MCU fan is always ON)
- Print head with newer PCB (with the 2 red connectors for little fans) and hot-end v2.5
- Has been running quite fine with latest QiDi firmware until last friday.
- Installed fresh FreeDi with the image file, using "install B"
- Sunplus USB webcam
- Connected via LAN
- EMMC 32G  (I still have the original QiDi firmware on the old 8G)

Symptoms : 
On first FreeDi install, the screen flashed sucessfully the 1.40 firmware but could not connect to the MCU. I could not reach the MCU by LAN (no ping, no SSH, etc.). MY DHCP server did not show any IP address request ...
A simple printer reboot did not fix the issue
So I suspected a network driver issue, removed all usb devices, including the wifi dongle and rebooted again --> success, I found the new IP in the DHCP server log and could connect via http and ssh.
Flashed klipper 12 on the print head RPi 2040 and on the mainboard STM32 using binaries included with FreeDi 1.40. --> no problem, adapted the print head ID in printer.cfg , rebooted --> all fine.
Re-plugged the webcam, the wifi and the cable for the top connector. OK.
I was happy it is was working, so I used kiauh to add some software : crownest, mobileraker, spoolman, Gcode viewer ...
Webcam configuration --> working well !
From that point, I started to work from the LCD screen to calibrate offset, vibrations, bed levelling etc.
The printer did NOT like that very much : I experienced a few system freezes, klipper crashes, some movement incidents with the bed crashing into the printhead and keep pushing ... the emergency stop worked a few times, but in one case I had to power off to stop the problem.
I also tried to print : the first layer was bad but on the right place (and very slowly printed) , then the machine moved to the 2nd layer, became much faster and completely lost the right XY position ... complete failure.
Today I decided to re-do a fresh install of FreeDi 1.40, with only crownest and spoolman. Re-install went very well, no system crash.
Calibration went fine, but I did all via the HTTP interface, I did not touch the LCD screen of the printer.
New slice, with default settings, new try at printing ... and same results : 1st layer very bad (but still sticky) like of the Z-Offset setting was completely useless and from the 2nd layer the XY position was completely wrong.
I suspected a Klipper bug and there was a newer version available on Kiauh (klipper 0.12.0-489 ?)   so I decided to give it a try, updated, compiled the new binaries, installed the new X_4.bin via sd-card and the new firmware for the printhead.
New printing test and ... same problem.

At this point, I am left with one possible test idea : there are some minor value changes between my old printer.cfg (QiDi) and the new one from FreeDi : some minimum are -2 instead of -6 and some maximum are 326 instead of 328 ... that kind of things. Could this be the cause of the problem ?
Could it be a problem because I have an older mainboard (X4 v1.1) ?
The other option is to go back to QiDi firmware for a while. It has been working fine, except for the "G17 unkown code" errors poping up regularly while printing. But I have no idea how to flash the LCD screen back with the QiDi firmware.

Note : I am not too bad with linux, firmware flashing, etc. , but I am still quite NOOB about 3D printing.

tjhart85, 2025-02-03:
I wasn't going to put in a ticket since I was able to get it back to working and didn't see others reporting the same issue, BUT, mine acted similarly going from 1.30 to 1.40 with that error on the screen of the Smart3.  Mine though didn't come back after leaving it off for a bit.  Maybe an issue with the Smart3's?

I pulled the eMMC, grabbed my configs off of it, re-flashed it to 1.4 this time, drilled a hole in the chassis to allow the network cable to come out without causing the printer to wobble while I was waiting for it to flash, put the eMMC back, put my configs back on and then went through the upgrade procedure doc to manually edit in all the updates to the config file that this update included (and now actually backed up my modifications!).

Additional details:  The first version of FreeDi I used was 1.30 and that was the version I had flashed to my eMMC originally and where I had started the upgrade from.

pmbroth, 2025-02-03:
Hello i have one of the first X-Max 3 as well with the same
Motherboard and. I have everything working.  Did you validate your original config  and the new one as the installation guide states. Phil had several areas of the config to validate.

theclue, 2025-02-03:
> Hello i have one of the first X-Max 3 as well with the same Motherboard and. I have everything working. Did you validate your original config and the new one as the installation guide states. Phil had several areas of the config to validate.

Yes, I diff'ed all the cfg files and applied the changes. But I would say that nothing you can put in *.cfg files could eventually brick the flashing. I don't think is _technically_ possible, I mean.

pmbroth, 2025-02-04:
> From "Gabriele Baldassarre" ***@***.***>
> Cc "pmbroth" ***@***.***>; "Comment"
> Date 2/3/2025 6:00:29 PM
> Subject Re: [Phil1988/FreeDi] [Bug] [x-Smart 3] Catastrophic Upgrade to
> v1.40 (almost bricked) (Issue #138)

Hello

I had to manually flash my display with a microsd card with the lcd 
firmware for Freedi 1.40,, because the flash did not work.  I grabbed 
Phils lcd firmware put it on the microsd card, then put it in the slot 
on the lcd, and turned on the machine.  I then was able to continue  
with the installation.

Brian

rosenrot00, 2025-02-04:
Thanks for the reports to everyone here who takes the time to report back.

@theclue and to all the others commenting here - sorry for the inconvenience during the update process. The title ‘almost bricked’ seems a bit drastic, as in expert terminology, ‘bricking’ means that the device is no longer recoverable. Basically, the system is not brickable at all because in the worst case, you could simply recover the emmc, flash the MCU, tool head, and the screen using an SD card. That aside, we have had rare incidents in which the TFT update process freezes and the system is not accessible anymore (web UI, ssh). In these cases powering off and on usually helps. We are currently investigating the reasons, so far we understand that the RK3328 is rather low performance and sometimes tends to freeze on heavy UART usage. We are currently figuring out countermeasures but incidents where this behavior persists through several power cycles have been rare. We hear your feedback and we are looking into it.

Phil1988, 2025-02-04:
### Hey there! 👋

First of all, I appreciate your detailed feedback and the effort you put into troubleshooting the issue. I completely understand that running into such a problem can be frustrating, and I want to assure you that FreeDi is meant to **enhance** your system—not cause headaches. 😊  

That being said, I’d like to address a few points regarding the way this issue was reported and what actually happened.  

---

### About the issue  
We are **aware of this behavior**, and it is clearly mentioned in the **update instructions**:  

[image]  

It seems this part might have been **overlooked** during the update process. **rosenrot00** has already explained the technical details.

As a side note, I want to clarify some wordings:

- The **MCU** is the microcontroller responsible for handling I/O tasks (sensors, heaters, etc.).  
- You **cannot** unplug USB from the MCU (because it's part of the mainboard) or connect to it via LAN.
- The correct terms would be:  
  - "Unplug USB from the **mainboard**" ✅  
  - "Could not access the **host**" ✅  

---

### About the troubleshooting approach  
I noticed that you attempted **several unnecessary steps** (e.g., unplugging USB, reflashing eMMC, adding a wired Ethernet connection etc.). None of these steps were mentioned in my update instructions.  

I’m not sure why you felt the need to **deviate from the documented process** and then open a **critical bug report** with a title like *"system almost bricked"*. That kind of wording can be quite alarming and might not accurately reflect the situation.  

---

### The actual solution  
As noted, the simplest fix was just to **turn the printer off, take a deep breath to chill down (you and the CPU 😉), and turn it back on**.  

That’s all that was needed. No drilling, no re-flashing, no panic.  

In fact, it’s very likely that the time spent on these extra steps allowed the **RK3328** to cool down enough for the screen flashing process to complete successfully on the next boot.  

---

### 🙏 A general note to all users  
I **truly appreciate** that many of you take initiative and try to solve problems on your own. That curiosity and dedication **help improve** the project and your findings can help to fix issues! But sometimes, a quick check-in with us would be more effective.

Instead of jumping into **panic mode**, I encourage everyone to **report issues right away** and follow the instructions carefully. That way, we can **collaborate** and find the best solutions **together**. If there are bugs or problems it's most of the time just a little fix on the software side that could be ironed out better this way. Most of the time it's simply causes by doing something "unforeseen or different" ;)

---

Now, @pmbroth is in a **different page** - he updated even before the release was finished because he’s *100% on the bleeding edge* 😆. Unfortunately, that means he ran into issues that havens been wiped out yet. Sorry, Brian and thanks for being the guinea pig  😉  

---

### 🚀 Final thoughts  
FreeDi is here to **make your experience better**, not to cause stress. Let’s work together and get in touch as soon as you see something "weird".

Thanks again for your feedback - I hope this helps clarify things. 😊

Phil1988, 2025-02-04:
@El-Keldo 
Everything starting from "The printer did NOT like that very much" should be discussed seperately as it is not connected to the update at all.

I need your backup config and your current,
detailed information when "system freezes, klipper crashes, some movement incidents" happened exactly.
I need to see when they are happening to reproduce it or to see what you are doing. 
Short videos definetly help.
Some of them might actually only "appear for you" as freezes or klipper crashes" as you are a klipper novice and might not understand that the system is eg. saving the config after input shaping (which ends on a klipper restart that is displayed on screan and might look like a unwanted klipper restart, while it is in fact normal when changing/saving the config).

More on this better in a seperate thread or even better in discord, as you might need more background information as a new klipper user.
The normal behavior can be seen here:
https://youtu.be/_ZmSTJBWUus?t=1886

Phil1988, 2025-02-04:
I'll leave this open, untill there is a way to put less stress to the CPU while updating the display.

pmbroth, 2025-02-04:
@Phil1988 I did not blame you or anyone else about my specific situation when upgrading.  I do not have an issue in being the first to try things out.  I can always roll back if there is a major issue.  When things are new, there is bound to be a hiccup.  Thank you again for all your doing!!  Brian

Phil1988, 2025-02-04:
@pmbroth all good mate!
I never felt you blaming anyone.
Just wanted to underline that you were my unwanted guinea pig :D
Im happy to have you on board 👍

pmbroth, 2025-02-04:
@Phil1988 , I'm going on vacation from the 8th through the 17th, after that if there are tests needed, I will help.  As I will be away out of the country I won't be able to test  :-)  Regards, Brian

rosenrot00, 2025-02-04:
Anyone here who sees the problem and is stuck on flash. Try to quickly login into ssh and run this command:
`echo performance | sudo tee /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor`
This is changing the frequency scaling of the host to performance. I have read that the rk3288 could freeze if there is a lot of scaling involved. If this should help anyone, please report

theclue, 2025-02-04:
I'm terribly sorry,  I never wanted to be unfair. I'm your bigger fan here!

But for a series of unfortunate circumstances this is my only available printer atm so I went offtrack and...ehm...ok...I went to panic mode I'm afraid :|

Said that. sorry also for the catastrophic wording. The fact is that the printer actually __looked__ bricked to me. Not being able to ssh even with RJ45, reboot, access to moonraker gave me too few options to try to troubleshoot and to put it up and running again. But to transition from a working state to an unworking one without debug information to try to recover IMHO could be labelled as "bricked". There is no intent to put shame on the project!

I followed the guide, I noticed the box you quoted above, but I toggled the power countless times but the flashing process stucked everytime at 8%.  Everytime. It looked too deterministic and sistematically reproducible to me for being one of those occasional system freezes referred above.

I plugged off from the host (ehm, yes, the host not the controller) everything not strictly needed for booting (camera, widi...), suspecting something could go wrong during the host lookup (I'm using a cheap generic webcam). Actually, the camera was my first candidate. It worked, but ofc it could be a coincidence. But high CPU usage corrupting the stuff on the UART bus is still my best hypothesis (altough...systematically crashing at 8%?!? Still unexplained)

About testing, I would be happy to help, but as I said it's the only working kiddo I've atm, so I don't think I can put it offline for long time. But if the crash tests won't run for days (and we can do them during weekends), I'll be happy to help!

I think I ow you a beer! Perhaps during the first freeDi party meeting :D

Phil1988, 2025-02-04:
Its all good mate!
I do understand that it was shocking for you.

Thank you for being a part of the project and for your involvement.
I think next time you'll find us more easily and we can calm you down -
And hey, I’ll gladly accept that beer offer when the time comes! 😄