Phil1988/FreeDi #139: Doc: How to flash mmc from usb (on printer that do not boot from usb)

potens1, 2025-02-04:
> Sorry, this should be put in the wiki, but I can wrap my head on how to contribute a wiki on github.
> In the end, you can just copy paste the following in a wiki page (if you thing it has some value there, I know it's quite dirty, but it's a work  base)

# Disclaimer:

* This is a huuuge draft, it should be cleaned and completed when other people experiment
* This method worked for _me_, I can not guarantee it will for you, be ready to need the emmc to usb adapter, after all
* I used this on a Qidi X-MAX 3 that refused to boot from USB
* This is NOT clean, the end result is clean but the whole process is not.
* I repeat, it was tested only once, on my printer...
* I sent the final FS through the network, if it can not work fro you, maybe you will need a second USB key, and to connect two keys at the same time, needing to open the printer

# Preparation

* Write the latest image from release on a USB drive
* Ensure you have ssh access on the stock firmware

# Process

* plug the USB drive in the printer
* ssh to the printer (qidimax3 on my network)

```
❯ ssh mks@qidimax3
mks@qidimax3's password: 
┌─────────────────────────────────────────────┐
│  ___  ___ ____ ___   _____ _____ ____ _   _ │
│ / _ \|_ _|  _ \_ _| |_   _| ____/ ___| | | |│
│| | | || || | | | |    | | |  _|| |   | |_| |│
│| |_| || || |_| | |    | | | |__| |___|  _  |│
│ \__\_\___|____/___|   |_| |_____\____|_| |_|│
│                                             │
└─────────────────────────────────────────────┘
+-------------------------------------------------------------------------------+
| Warning: Modifying system files and installing unofficial plugins means that  |
| the customer is waiving their expectations of official support. They will be  |
| solely responsible for the security and safety of their printer. Any firmware |
| issues arising from these modifications will not be covered under warranty.   |
+-------------------------------------------------------------------------------+

Welcome to Armbian 22.05.0-trunk  with bleeding edge Linux 5.16.20-rockchip64

No end-user support: built from trunk

System load:   36%              Up time:       1 min
Memory usage:  18% of 976M      IP:            192.168.666.222
CPU temp:      45°C             Usage of /:    76% of 6.6G   
storage/:      35% of 256M   

```

* check if the USB disk has been mounted, and unmount it if it was

```
mks@mkspi:~$ mount |grep /home/mks/gcode_files
/dev/sda1 on /home/mks/gcode_files/sda1 type vfat (rw,nosuid,nodev,noexec,relatime,gid=100,fmask=0111,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,utf8,flush,errors=remount-ro)
mks@mkspi:~$ sudo umount /home/mks/gcode_files/sda1 
[sudo] password for mks: 

```

* become root
```
sudo -s
```

* unmount /boot and logs

```
root@mkspi:/home/mks# umount /boot
root@mkspi:/home/mks# umount /var/log.hdd

```

* break everything...and flash the USB image onto the eMMC

Yes the system will not be happy, and it can result into a "bricked" (until you have emmc to usb hardware)

```
root@mkspi:/home/mks# dd if=/dev/sda of=/dev/mmcblk1 bs=4M
Message from syslogd@mkspi at Feb  4 13:18:10 ...
 systemd[1]: Caught <ILL>, core dump failed (child 2446, code=killed, status=4/ILL).

Broadcast message from systemd-journald@mkspi (Tue 2025-02-04 13:18:10 CET):

systemd[1]: Caught <ILL>, core dump failed (child 2446, code=killed, status=4/ILL).

Broadcast message from systemd-journald@mkspi (Tue 2025-02-04 13:18:10 CET):

systemd[1]: Freezing execution.

Message from syslogd@mkspi at Feb  4 13:18:10 ...
 systemd[1]: Freezing execution.

dd: error writing '/dev/mmcblk1': No space left on device
1865+0 records in
1864+0 records out
7818182656 bytes (7.8 GB, 7.3 GiB) copied, 255.78 s, 3d+01 MB/s
Connection to qidimax3 closed.

```

* poweroff and on again 
 leave the USB key in the printer !

# FreeDi running (from USB)

* ssh in the system again
  you will have an error, about the key, that's normal and it means you booted from the usb drive

```
❯ ssh mks@qidimax3
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the ED25519 key sent by the remote host is
SHA256:BlAbKabKaBlaBla/dEAdCat.
Please contact your system administrator.
Add correct host key in /home/nic/.ssh/known_hosts to get rid of this message.
Offending ECDSA key in /home/nic/.ssh/known_hosts:5275
Host key for qidimax3 has changed and you have requested strict checking.
Host key verification failed.

```

correct the situation and ssh again
```
❯ ssh mks@qidimax3
The authenticity of host 'qidimax3 (192.168.69.222)' can't be established.
ED25519 key fingerprint is SHA256:bloubloubou/d4rthv4der.
This key is not known by any other names.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'qidimax3' (ED25519) to the list of known hosts.
mks@qidimax3's password: 
           _              _ 
 _ __ ___ | | _____ _ __ (_)
| '_ ` _ \| |/ / __| '_ \| |
| | | | | |   <\__ \ |_) | |
|_| |_| |_|_|\_\___/ .__/|_|
                   |_|      
Welcome to Armbian-unofficial 24.2.0-trunk Bookworm with Linux 6.6.17-current-rockchip64

No end-user support: built from trunk

System load:   54%              Up time:       1 min
Memory usage:  23% of 911M      IP:            192.168.69.222
CPU temp:      40°C             Usage of /:    70% of 6.7G   
RX today:      10.0 MiB  

```
*Yay !*

Now, the system should tell you it is updating the display firmware, let it finish it.
When done the screen of the printer should show you nice illustration of the printer with a "booting" message 

# Replace (again) the root filesystem from ...somewhere

I sent the image from my computer through ssh, but if you have windows I don't know if it works.
So from the directory I uncompressed the image:

```
cat FreeDi_v1.40_X-Max3_Armbian24.2.0_Bookworm6.6.17.img |ssh root@qidimax3 "dd status=progress of=/dev/mmcblk1"
root@qidimax3's password: 
7810626048 bytes (7.8 GB, 7.3 GiB) copied, 882 s, 8.9 MB/s 
15269888+0 records in
15269888+0 records out
7818182656 bytes (7.8 GB, 7.3 GiB) copied, 883.862 s, 8.8 MB/s
```

I guess, but this must be confirmed, you can also do it from the booted USB (I did nbot since the FS is mounted, so the resulting copy is not consistent, the filesystem check at boot should correct it but unsure)

* Ensure everything is written on disk
```
mks@mkspi:~$ sudo sync
[sudo] password for mks: 
```

Final reboot, now you should be able to continue the guide from [here](https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#creating-the-perfect-match-with-your-hardware)

Phil1988, 2025-02-04:
Awesome work mate!
I'll bring it to the wiki but leaving this thread to other can contribute, ask, feedback 👍

closed 2025-02-04 by Phil1988

potens1, 2025-12-05:
Just to let you know I used that again to upgrade to 2.0, and it worked without a problem (for me)

Phil1988, 2025-12-06:
Thanks for confirming 👍 

But I am wondering: 
Was this done on a another stock QIDI printer?
Because if it was done on a "FreeDi printer", wasn't it possible to directly boot from USB?
On my testings it worked flawless.

potens1, 2025-12-06:
Hey @Phil1988 no, it was done on previous FreeDi version, but I was never able to make it boot from the USB stick. The only hurdle I had with this was you have to put a sdcard on the motherboard for the firmware, and I'm pretty sure it could be done without it.

TBH, I think if I find some time soon, I will have a look to check if there is no way to implement [RAUC](https://rauc.io/) on your image (not with A/B partition, but with updater partition and installer partition [like this](https://rauc.readthedocs.io/en/latest/scenarios.html#asymmetric-slots) the only unknown to me right now is the bootloader). This way it would be easier to do atomic update of full FS, without needing opening the printer, removing the emmc and so on. (but I guess I will need to buy a emmc reader, because, the risk for me to nuke the whole thing experimenting is quite high)