Phil1988/FreeDi #143: Maybe a bug or not

pmbroth, 2025-02-06:
Hello Phil,

As you are aware I updated my system to 1.4, display shows freedi 1.4 stablel, but wihtin Fluid or Mainsail, look at the screen shot below for what  diplayed:  Frredi V1.3-74g16f6f141.  Thanks you Brian

[image]

Phil1988, 2025-02-06:
Its not a problem.
You are on this [stage](https://github.com/Phil1988/FreeDi/commit/16f6f1417900d8f584cadbea634df35f17f9f5b2) which was deactived on a later fix.
The fix includes path and chmod changes for klipper auto flasher and is not at all important for you.
You can simlpy skip it and wait for the next real release. (you can redo the v1.40 update again to pull the latest 2 files and making it appear as v1.40 in update manager if you like to).

closed 2025-02-06 by Phil1988

pmbroth, 2025-02-06:
@Phil1988     I did redo the 1.4 install, and it still says 1.3 in fluid and mainsail,,  I will wait till the next release, but it is strange:  Just wanted to let you know
Brian