Phil1988/FreeDi #146: [Bug] Klipper auto flasher (KAF) not flashing the MCU

Phil1988, 2025-02-07:
On [some systems](https://github.com/Phil1988/FreeDi/discussions/140) KAF is unable to flash the MCU on a booting system.
Reason is probably that klipper hasnt fully loaded when KAF is being executed (slightly slower EMMC, additional software booting etc.).

I have detailed logs and will be under investigation when I am back.
Most likely I found a solution already but it needs to be tested.

Everyone facing this issue can use KAF by invoking the start manually.
Do do so:

1. Bring your toolhead in boot mode
  Like shown [here](https://youtu.be/_ZmSTJBWUus?t=1204)
1.1 Press and hold the „boot“ button
1.2 Press and release the „reset“ button.
1.3 Release the „boot“ button.
 
2. Execute this via terminal:
    ```
    sudo ~/FreeDi/helpers/klipper_auto_flasher.sh
    ```

    You should see something like this

    ```
    Device /dev/sda1 size is exactly 128M
    Copying firmware to /dev/sda1...
    Firmware successfully copied to /dev/sda1
    
    +-----------------------------------------------------------------------+
    |         HID-Flash v2.2.1 - STM32 HID Bootloader Flash Tool            |
    |     (c)      2018 - Bruno Freitas       http://www.brunofreitas.com/   |
    |     (c) 2018-2019 - Vassilis Serasidis  https://www.serasidis.gr/      |
    |   Customized for STM32duino ecosystem   https://www.stm32duino.com/    |
    +-----------------------------------------------------------------------+
    
    Trying to open the [ttyS0]...
    Open the [ttyS0] successfully
    Sending <reset pages> command...
    rcv 2 data:OK
    rcv 2 data:OK
    Flashing firmware...
     1024 Bytes
     29696 Bytes
    rcv 8 data:BTLDCMD
     30720 Bytes
    rcv 8 data:BTLDCMD
     31744 Bytes
    rcv 8 data:BTLDCMD
     32768 Bytes
    rcv 8 data:BTLDCMD
     33792 Bytes
    rcv 8 data:BTLDCMD
     34816 Bytes
    rcv 8 data:BTLDCMD
    
    [...]
    
    Starting Klipper service...
    ```

rmx-it, 2025-02-26:
Hi,
I'm affected by this issue with my X-Plus-3.
Toolhead is on: Version: v0.12.0-289-g08a1c9f1
MCU is still on: Version: v0.10.0-530-g3387a9c2-dirt
Host is on: Version: v0.12.0-289-g08a1c9f12-dirty
I did an update today by replacing the original EMMC with a 32GB one flashed with Freedi 1.40.
BTW: Mainsail still shows in update manager:
  FreeDi v1.30-85-g64e22fd0

If you want me to test something...

rosenrot00, 2025-02-26:
@rmx-it could you post the output of `journalctl -u AutoFlasher.service` here?

rmx-it, 2025-02-26:
....
Jan 28 19:18:12 mkspi systemd[1]: Starting AutoFlasher.service - Toolhead & MCU Flash Service...
Jan 28 19:18:12 mkspi (asher.sh)[866]: AutoFlasher.service: Failed to locate executable /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 28 19:18:12 mkspi (asher.sh)[866]: AutoFlasher.service: Failed at step EXEC spawning /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 28 19:18:12 mkspi systemd[1]: AutoFlasher.service: Main process exited, code=exited, status=203/EXEC
Jan 28 19:18:12 mkspi systemd[1]: AutoFlasher.service: Failed with result 'exit-code'.
Jan 28 19:18:12 mkspi systemd[1]: Failed to start AutoFlasher.service - Toolhead & MCU Flash Service.
-- Boot afa806b8313841b6b1c75191e4197a03 --
Jan 28 20:08:28 mkspi systemd[1]: Starting AutoFlasher.service - Toolhead & MCU Flash Service...
Jan 28 20:08:28 mkspi (asher.sh)[930]: AutoFlasher.service: Failed to locate executable /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 28 20:08:28 mkspi (asher.sh)[930]: AutoFlasher.service: Failed at step EXEC spawning /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 28 20:08:29 mkspi systemd[1]: AutoFlasher.service: Main process exited, code=exited, status=203/EXEC
Jan 28 20:08:29 mkspi systemd[1]: AutoFlasher.service: Failed with result 'exit-code'.
Jan 28 20:08:29 mkspi systemd[1]: Failed to start AutoFlasher.service - Toolhead & MCU Flash Service.
-- Boot 3e88873780ad48c69978b4580dd5aef6 --
Jan 28 20:15:22 mkspi systemd[1]: Starting AutoFlasher.service - Toolhead & MCU Flash Service...
Jan 28 20:15:22 mkspi (asher.sh)[934]: AutoFlasher.service: Failed to locate executable /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 28 20:15:22 mkspi (asher.sh)[934]: AutoFlasher.service: Failed at step EXEC spawning /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 28 20:15:22 mkspi systemd[1]: AutoFlasher.service: Main process exited, code=exited, status=203/EXEC
Jan 28 20:15:22 mkspi systemd[1]: AutoFlasher.service: Failed with result 'exit-code'.
Jan 28 20:15:22 mkspi systemd[1]: Failed to start AutoFlasher.service - Toolhead & MCU Flash Service.
-- Boot 51217c4a872446318c45d3b1d78a7ad0 --
Jan 28 20:22:53 mkspi systemd[1]: Starting AutoFlasher.service - Toolhead & MCU Flash Service...
Jan 28 20:22:53 mkspi (asher.sh)[893]: AutoFlasher.service: Failed to locate executable /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 28 20:22:53 mkspi (asher.sh)[893]: AutoFlasher.service: Failed at step EXEC spawning /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 28 20:22:53 mkspi systemd[1]: AutoFlasher.service: Main process exited, code=exited, status=203/EXEC
Jan 28 20:22:53 mkspi systemd[1]: AutoFlasher.service: Failed with result 'exit-code'.
Jan 28 20:22:53 mkspi systemd[1]: Failed to start AutoFlasher.service - Toolhead & MCU Flash Service.
-- Boot a950d29ed84c4fceb2960415acecddfe --
Jan 28 23:16:37 mkspi systemd[1]: Starting AutoFlasher.service - Toolhead & MCU Flash Service...
Jan 28 23:16:37 mkspi (asher.sh)[875]: AutoFlasher.service: Failed to locate executable /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 28 23:16:37 mkspi (asher.sh)[875]: AutoFlasher.service: Failed at step EXEC spawning /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 28 23:16:37 mkspi systemd[1]: AutoFlasher.service: Main process exited, code=exited, status=203/EXEC
Jan 28 23:16:37 mkspi systemd[1]: AutoFlasher.service: Failed with result 'exit-code'.
Jan 28 23:16:37 mkspi systemd[1]: Failed to start AutoFlasher.service - Toolhead & MCU Flash Service.
-- Boot dbdb0155fe0e465cb83bd9d6d0ff6cd3 --
Jan 29 10:42:58 mkspi systemd[1]: Starting AutoFlasher.service - Toolhead & MCU Flash Service...
Jan 29 10:42:58 mkspi (asher.sh)[912]: AutoFlasher.service: Failed to locate executable /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 29 10:42:58 mkspi (asher.sh)[912]: AutoFlasher.service: Failed at step EXEC spawning /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 29 10:42:58 mkspi systemd[1]: AutoFlasher.service: Main process exited, code=exited, status=203/EXEC
Jan 29 10:42:58 mkspi systemd[1]: AutoFlasher.service: Failed with result 'exit-code'.
Jan 29 10:42:58 mkspi systemd[1]: Failed to start AutoFlasher.service - Toolhead & MCU Flash Service.
-- Boot 54a9311a6c694dc7a40a5d137bcbfd76 --
Jan 29 10:49:43 mkspi systemd[1]: Starting AutoFlasher.service - Toolhead & MCU Flash Service...
Jan 29 10:49:43 mkspi (asher.sh)[935]: AutoFlasher.service: Failed to locate executable /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 29 10:49:43 mkspi (asher.sh)[935]: AutoFlasher.service: Failed at step EXEC spawning /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
Jan 29 10:49:44 mkspi systemd[1]: AutoFlasher.service: Main process exited, code=exited, status=203/EXEC
Jan 29 10:49:44 mkspi systemd[1]: AutoFlasher.service: Failed with result 'exit-code'.
Jan 29 10:49:44 mkspi systemd[1]: Failed to start AutoFlasher.service - Toolhead & MCU Flash Service.
-- Boot dcbf95ef6b304de39e22fcbc61951b6f --
Jan 30 08:36:23 mkspi systemd[1]: Starting AutoFlasher.service - Toolhead & MCU Flash Service...
Jan 30 08:36:23 mkspi (asher.sh)[963]: AutoFlasher.service: Failed to locate executable /home/mks/FreeDi/helpers/klipper_auto_flasher.sh: Permission denied
[... cut ...]
Feb 26 13:35:44 mkspi systemd[1]: AutoFlasher.service: Deactivated successfully.
Feb 26 13:35:44 mkspi systemd[1]: Stopped AutoFlasher.service - Toolhead & MCU Flash Service.
-- Boot d326fe12bfc1479e907f145c2b1fdf64 --
Feb 26 13:35:54 mkspi systemd[1]: Starting AutoFlasher.service - Toolhead & MCU Flash Service...
Feb 26 13:35:55 mkspi klipper_auto_flasher.sh[958]: Device /dev/sda1 size is 3.6M (not 128M)
Feb 26 13:35:55 mkspi systemd[1]: Finished AutoFlasher.service - Toolhead & MCU Flash Service.
....

pmbroth, 2025-02-26:
- When I was updating my toolhead, I had to  use the steps  in the screen shot below, as the steps you outlined  did not work for me as my tool head never went into the the boot mode:
 
-Bring your toolhead in boot mode
Like shown [here](https://youtu.be/_ZmSTJBWUus?t=1204)
1.1 Press and hold the „boot“ button
1.2 Press and release the „reset“ button.
1.3 Release the „boot“ button.

This worked for me:

[image]

rmx-it, 2025-02-26:
In my case the toolhead where updated fine but not the main MCU...

pmbroth, 2025-02-26:
did you try putting the x_4.bin on the micro sd card and try it that way?

rmx-it, 2025-02-26:
No,
because above rosenrot00 stated:
...
It is worth mentioning that anyone for whom it doesn't work is very welcome to contact us as we have a possible solution in place that would need testing :)
...

rosenrot00, 2025-02-26:
Can you run this

`cd ~ && rm -rf FreeDi && git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh`

and then redo the flashing automated flashing and post the output again??

rmx-it, 2025-02-26:
just to be sure:
For the main MCU to get flashed is it needed to use the boot button on the toolhead during poweron
or is it sufficient to just powercycle the printer?

rosenrot00, 2025-02-26:
> just to be sure: For the main MCU to get flashed is it needed to use the boot button on the toolhead during poweron or is it sufficient to just powercycle the printer?

For now the MCU gets only flashed when the toolhead is in DFU mode, meaning the boot button has to be held during poweron for 3 seconds (from power off to power on)

rmx-it, 2025-02-26:
The install went through, the only thing what was unexpected during build:
...
Script /home/mks/FreeDi/helpers/usbmounter.sh is now executable.
==== AUTHENTICATING FOR org.freedesktop.systemd1.reload-daemon ====
Authentication is required to reload the systemd state.
Authenticating as: MKS,,, (mks)
Password: Reload daemon failed: Method call timed out
polkit-agent-helper-1: pam_authenticate failed: Authentication failure
...

rmx-it, 2025-02-26:
I'm currently doing a new install after a git pull

rmx-it, 2025-02-26:
No, still old version:
...
Feb 26 14:57:22 mkspi systemd[1]: AutoFlasher.service: Deactivated successfully.
Feb 26 14:57:22 mkspi systemd[1]: Stopped AutoFlasher.service - Toolhead & MCU Flash Service.
-- Boot b414ef336292482aa9d3f93d5c65b4c0 --
Feb 26 14:57:33 mkspi systemd[1]: Starting AutoFlasher.service - Toolhead & MCU Flash Service...
Feb 26 14:57:34 mkspi klipper_auto_flasher.sh[989]: Device /dev/sdb1 size is exactly 128M
Feb 26 14:57:34 mkspi klipper_auto_flasher.sh[989]: Copying firmware to /dev/sdb1...
Feb 26 14:57:34 mkspi sudo[1030]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/cp /home/mks/FreeDi/helpers/../mainboard_and_toolhead_firmwares/v0.12.0-289/Toolhead_RP2040.uf2 /dev/sdb1
Feb 26 14:57:34 mkspi sudo[1030]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Feb 26 14:57:34 mkspi sudo[1030]: pam_unix(sudo:session): session closed for user root
Feb 26 14:57:35 mkspi klipper_auto_flasher.sh[989]: Firmware successfully copied to /dev/sdb1
Feb 26 14:57:35 mkspi sudo[1117]:     root : PWD=/ ; USER=root ; COMMAND=/usr/sbin/service klipper restart
Feb 26 14:57:35 mkspi sudo[1117]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Feb 26 14:57:35 mkspi sudo[1117]: pam_unix(sudo:session): session closed for user root
Feb 26 14:57:35 mkspi klipper_auto_flasher.sh[989]: Klipper service has been running for at least 10 seconds.
Feb 26 14:57:38 mkspi sudo[1345]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/tee /home/mks/printer_data/comms/klippy.serial
Feb 26 14:57:38 mkspi sudo[1345]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Feb 26 14:57:38 mkspi klipper_auto_flasher.sh[1347]: FIRMWARE_RESTART
Feb 26 14:57:38 mkspi sudo[1345]: pam_unix(sudo:session): session closed for user root
Feb 26 14:57:38 mkspi klipper_auto_flasher.sh[1352]: +-----------------------------------------------------------------------+
Feb 26 14:57:38 mkspi klipper_auto_flasher.sh[1352]: |         HID-Flash v2.2.1 - STM32 HID Bootloader Flash Tool            |
Feb 26 14:57:38 mkspi klipper_auto_flasher.sh[1352]: |     (c)      2018 - Bruno Freitas       http://www.brunofreitas.com   |
Feb 26 14:57:38 mkspi klipper_auto_flasher.sh[1352]: |     (c) 2018-2019 - Vassilis Serasidis  https://www.serasidis.gr      |
Feb 26 14:57:38 mkspi klipper_auto_flasher.sh[1352]: |   Customized for STM32duino ecosystem   https://www.stm32duino.com    |
Feb 26 14:57:38 mkspi klipper_auto_flasher.sh[1352]: +-----------------------------------------------------------------------+
Feb 26 14:57:38 mkspi klipper_auto_flasher.sh[1352]: > Trying to open the [ttyS0]...
Feb 26 14:57:38 mkspi klipper_auto_flasher.sh[1352]: > Open the [ttyS0] successfully
Feb 26 14:57:38 mkspi klipper_auto_flasher.sh[1352]: > Sending <reset pages> command...
Feb 26 14:57:38 mkspi sudo[1351]:     root : PWD=/ ; USER=root ; COMMAND=/usr/sbin/service klipper stop
Feb 26 14:57:38 mkspi sudo[1351]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Feb 26 14:57:41 mkspi klipper_auto_flasher.sh[1352]: [13B blob data]
Feb 26 14:57:43 mkspi sudo[1351]: pam_unix(sudo:session): session closed for user root
Feb 26 14:58:41 mkspi klipper_auto_flasher.sh[1352]: [13B blob data]
Feb 26 14:58:44 mkspi klipper_auto_flasher.sh[1352]: [13B blob data]
Feb 26 14:58:47 mkspi klipper_auto_flasher.sh[1352]: rcv 2 data:
Feb 26 14:58:50 mkspi klipper_auto_flasher.sh[1352]: [13B blob data]
Feb 26 14:58:53 mkspi klipper_auto_flasher.sh[1352]: [13B blob data]
Feb 26 14:58:56 mkspi klipper_auto_flasher.sh[1352]: rcv 2 data:
Feb 26 14:58:59 mkspi klipper_auto_flasher.sh[1352]: [13B blob data]
Feb 26 14:59:02 mkspi klipper_auto_flasher.sh[1352]: [12B blob data]
Feb 26 14:59:05 mkspi klipper_auto_flasher.sh[1352]: > Error, no ok receive, timeout and exit!!!
Feb 26 14:59:05 mkspi klipper_auto_flasher.sh[1352]: > Finish
Feb 26 14:59:05 mkspi klipper_auto_flasher.sh[989]: Starting Klipper service...
Feb 26 14:59:08 mkspi sudo[1626]:     root : PWD=/ ; USER=root ; COMMAND=/usr/sbin/service klipper start
Feb 26 14:59:08 mkspi sudo[1626]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Feb 26 14:59:13 mkspi sudo[1626]: pam_unix(sudo:session): session closed for user root
Feb 26 14:59:13 mkspi systemd[1]: Finished AutoFlasher.service - Toolhead & MCU Flash Service.
...

rosenrot00, 2025-02-26:
So it still doesn't work, could you pull and install once more? I changed the timing again.

rmx-it, 2025-02-26:
Now it behaves differently:

After some minutes and also after a firmware restart I get:

Klipper reports: ERROR

mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

rmx-it, 2025-02-26:
journalctl -u AutoFlasher.service
...
-- Boot fb787c40df974fb1b08812962f194b92 --
Feb 26 15:21:59 mkspi systemd[1]: Starting AutoFlasher.service - Toolhead & MCU Flash Service...
Feb 26 15:21:59 mkspi klipper_auto_flasher.sh[902]: Device /dev/sd*[0-9] not found
Feb 26 15:21:59 mkspi systemd[1]: Finished AutoFlasher.service - Toolhead & MCU Flash Service.
...

dmesg states:
...
[   21.123120] rk_gmac-dwmac ff550000.ethernet end1: Link is Up - 100Mbps/Full - flow control off
[   21.196812] systemd-journald[570]: Data hash table of /run/log/journal/b316eee227b246f4a05a837026b08d9f/system.journal has a fill level at 75.5 (1568 of 2076 items, 1196032 file size, 762 bytes per hash table item), suggesting rotation.
[   21.196852] systemd-journald[570]: /run/log/journal/b316eee227b246f4a05a837026b08d9f/system.journal: Journal header limits reached or header out-of-date, rotating.
[   27.364031] platform hdmi-sound: deferred probe pending
[   40.961671] systemd[1549]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[   64.169172] irq 42: nobody cared (try booting with the "irqpoll" option)
[   64.169204] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C O       6.6.17-current-rockchip64 #4
[   64.169224] Hardware name: Makerbase mkspi (DT)
[   64.169235] Call trace:
[   64.169243]  dump_backtrace+0x94/0x114
[   64.169267]  show_stack+0x18/0x24
[   64.169283]  dump_stack_lvl+0x48/0x60
[   64.169301]  dump_stack+0x18/0x24
[   64.169316]  __report_bad_irq+0x38/0xe4
[   64.169336]  note_interrupt+0x328/0x378
[   64.169354]  handle_irq_event+0x9c/0xb0
[   64.169370]  handle_level_irq+0xdc/0x1d4
[   64.169388]  generic_handle_domain_irq+0x2c/0x44
[   64.169404]  rockchip_irq_demux+0xa0/0x220
[   64.169425]  generic_handle_domain_irq+0x2c/0x44
[   64.169441]  gic_handle_irq+0x44/0xcc
[   64.169456]  call_on_irq_stack+0x24/0x4c
[   64.169474]  do_interrupt_handler+0xd4/0xd8
[   64.169493]  el1_interrupt+0x34/0x68
[   64.169511]  el1h_64_irq_handler+0x18/0x24
[   64.169528]  el1h_64_irq+0x64/0x68
[   64.169543]  cpuidle_enter_state+0xc4/0x4bc
[   64.169562]  cpuidle_enter+0x38/0x50
[   64.169582]  do_idle+0x1fc/0x270
[   64.169601]  cpu_startup_entry+0x34/0x3c
[   64.169618]  kernel_init+0x0/0x1e0
[   64.169638]  arch_post_acpi_subsys_init+0x0/0x8
[   64.169658]  start_kernel+0x6c4/0x8fc
[   64.169676]  __primary_switched+0xb4/0xbc
[   64.169692] handlers:
[   64.169699] [<00000000b9987b11>] irq_default_primary_handler threaded [<000000004e2ce727>] regmap_irq_thread
[   64.169741] Disabling IRQ #42
...

rmx-it, 2025-02-26:
Output of the install script:
...
mks@mkspi:~/FreeDi$ ./install.sh 
Do you use the stock Mainboard? (y/n)
y
Starting the installation...
Sparse checkout only the required folders...
Configuring git to fetch tags automatically...
Removing freedi_update.sh from git index because it's already tracked...
Disabling freedi_update.sh git tracking for future modifications...
Successfully ignoring freedi_update.sh
Local ignore setup completed. The file freedi_update.sh will now be ignored locally by git.
Creating a symbolic link for freedi.py from /home/mks/FreeDi/klipper_module to /home/mks/klipper/klippy/extras...
Successfully installed freedi.py to /home/mks/klipper/klippy/extras.
Restarting Klipper service...
[sudo] password for mks: 
Klipper service restarted successfully.
Installation complete.
Setup dtbo for serial communication...
dtbo install done!
Customize the armbianEnv.txt file for serial communication...
Line found. Replacing the line.
armbianEnv.txt file modified successfully!
Setup the toolhead serial path in printer.cfg...
Error: no serial devices found in /dev/serial/by-id
Activating Klipper virtual environment and installing Python packages...
Klipper environment python version: 3.11.2
Arranging Python requirements...
Requirement already satisfied: pip in /home/mks/klippy-env/lib/python3.11/site-packages (25.0.1)
Requirement already satisfied: configparser in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 1)) (7.1.0)
Requirement already satisfied: requests in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 2)) (2.32.3)
Requirement already satisfied: pyserial>=3.5 in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 3)) (3.5)
Requirement already satisfied: Pillow in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 4)) (11.0.0)
Requirement already satisfied: numpy in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 5)) (2.2.0)
Requirement already satisfied: matplotlib in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 6)) (3.10.0)
Requirement already satisfied: psutil in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 7)) (6.1.1)
Requirement already satisfied: websocket-client in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 8)) (1.8.0)
Requirement already satisfied: charset-normalizer<4,>=2 in /home/mks/klippy-env/lib/python3.11/site-packages (from requests->-r /home/mks/FreeDi/requirements.txt (line 2)) (3.4.0)
Requirement already satisfied: idna<4,>=2.5 in /home/mks/klippy-env/lib/python3.11/site-packages (from requests->-r /home/mks/FreeDi/requirements.txt (line 2)) (3.10)
Requirement already satisfied: urllib3<3,>=1.21.1 in /home/mks/klippy-env/lib/python3.11/site-packages (from requests->-r /home/mks/FreeDi/requirements.txt (line 2)) (2.2.3)
Requirement already satisfied: certifi>=2017.4.17 in /home/mks/klippy-env/lib/python3.11/site-packages (from requests->-r /home/mks/FreeDi/requirements.txt (line 2)) (2024.12.14)
Requirement already satisfied: contourpy>=1.0.1 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (1.3.1)
Requirement already satisfied: cycler>=0.10 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (0.12.1)
Requirement already satisfied: fonttools>=4.22.0 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (4.55.3)
Requirement already satisfied: kiwisolver>=1.3.1 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (1.4.7)
Requirement already satisfied: packaging>=20.0 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (24.2)
Requirement already satisfied: pyparsing>=2.3.1 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (3.2.0)
Requirement already satisfied: python-dateutil>=2.7 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (2.9.0.post0)
Requirement already satisfied: six>=1.5 in /home/mks/klippy-env/lib/python3.11/site-packages (from python-dateutil>=2.7->matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (1.17.0)
Python requirements installed from requirements.txt.
Installing required packages for input shaping (if not already installed)...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libatlas-base-dev is already the newest version (3.10.3-13).
libopenblas-dev is already the newest version (0.3.21+ds-4).
ntfs-3g is already the newest version (1:2022.10.3-1+deb12u2).
0 upgraded, 0 newly installed, 0 to remove and 15 not upgraded.
System dependencies installed successfully.
Adding FreeDi section to moonraker update manager...
Moonraker configuration file /home/mks/printer_data/config/moonraker.conf found
The section [update_manager FreeDi] already exists in the file.
Permit Moonraker to restart FreeDi service...
[... cut ...]
Hit:2 http://deb.debian.org/debian bookworm-updates InRelease            
Hit:3 http://deb.debian.org/debian bookworm-backports InRelease          
Hit:4 http://security.debian.org bookworm-security InRelease             
Hit:5 http://armbian.systemonachip.net/apt bookworm InRelease            
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
15 packages can be upgraded. Run 'apt list --upgradable' to see them.
Setup complete!
Please restart your system for the changes to take effect.

rosenrot00, 2025-02-26:
Could you tell the output of autoflasher when you hit the boot button on the toolhead during power on?

I assume it still didn't flash the MCU?

wewentmoon69, 2025-02-27:
Not helpful to finding an automated solution for everyone, but I was having the same issue as @rmx-it with the Bootloader Flash Tool not working.  So, I followed the steps in the Installation Guide for flashing the MCU and tool head titled ""On FreeDi lower than v1.40".  Furthermore, I can confirm that if you only have a micro-SD card larger than 32GB, then you can use Disk Management or any similar partition software to create a FAT32 partition with 32GB and it should work.

[image]

That worked flawlessly for my machine, and everything is now running smoothly on Klipper v0.12.0-289.

rmx-it, 2025-02-27:
As it seemed the mcu got not flashed properly with the flashing tool
and was bricked afterwards.
Using the SDcard method worked fine afterwards.
So I have to wait now for the next clipper update to test the update tool again...

BTW: Many thanks for providing FreeDi!

rosenrot00, 2025-02-27:
Thanks for the feedback and your support. Once we come up with a solution I will post here again to see if some of you having the issue are happy to try.

jawzx2, 2025-03-08:
Just adding that my experience was the same as rmx-it, FreeDi 1.4 image on 32gig emmc, KAF flashed the tool head fine using boot button at power on as instructed. Main board MCU remained on v0.10 until I used the microSD card and x_4.bin file. Now tool head and MCU show the correct matching kipper version.

FuzzyLotus, 2025-03-27:
> On [some systems](https://github.com/Phil1988/FreeDi/discussions/140) KAF is unable to flash the MCU on a booting system. Reason is probably that klipper hasnt fully loaded when KAF is being executed (slightly slower EMMC, additional software booting etc.). For most systems it works.
> 
> I have detailed logs and will be under investigation when I am back. Most likely I found a solution already but it needs to be tested.
> 
> Everyone facing this issue can use KAF by invoking the start manually. Do do so:
> 
> 1. Bring your toolhead in boot mode
>    Like shown [here](https://youtu.be/_ZmSTJBWUus?t=1204)
>    1.1 Press and hold the „boot“ button
>    1.2 Press and release the „reset“ button.
>    1.3 Release the „boot“ button.
> 2. Execute this via terminal:
>    ```
>    sudo ~/FreeDi/helpers/klipper_auto_flasher.sh
>    ```
>    
>    
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    You should see something like this
>    ```
>    Device /dev/sda1 size is exactly 128M
>    Copying firmware to /dev/sda1...
>    Firmware successfully copied to /dev/sda1
>    
>    +-----------------------------------------------------------------------+
>    |         HID-Flash v2.2.1 - STM32 HID Bootloader Flash Tool            |
>    |     (c)      2018 - Bruno Freitas       http://www.brunofreitas.com/   |
>    |     (c) 2018-2019 - Vassilis Serasidis  https://www.serasidis.gr/      |
>    |   Customized for STM32duino ecosystem   https://www.stm32duino.com/    |
>    +-----------------------------------------------------------------------+
>    
>    Trying to open the [ttyS0]...
>    Open the [ttyS0] successfully
>    Sending <reset pages> command...
>    rcv 2 data:OK
>    rcv 2 data:OK
>    Flashing firmware...
>     1024 Bytes
>     29696 Bytes
>    rcv 8 data:BTLDCMD
>     30720 Bytes
>    rcv 8 data:BTLDCMD
>     31744 Bytes
>    rcv 8 data:BTLDCMD
>     32768 Bytes
>    rcv 8 data:BTLDCMD
>     33792 Bytes
>    rcv 8 data:BTLDCMD
>     34816 Bytes
>    rcv 8 data:BTLDCMD
>    
>    [...]
>    
>    Starting Klipper service...
>    ```

Running this solution got me up and running - thanks!

Phil1988, 2025-03-28:
You are welcome @FuzzyLotus :)

c10l, 2025-07-02:
I'm facing this issue on a Plus4. I got the toolhead in DFU mode but there's no `/dev/sd*` device.

```shell
mks@mkspi:~$ sudo ~/FreeDi/helpers/klipper_auto_flasher.sh
[sudo] password for mks:
Device /dev/sd*[0-9] not found
mks@mkspi:~$ ls -l /dev/sd*
ls: cannot access '/dev/sd*': No such file or directory
```

I tried both methods: booting with the `boot` button pressed, or the "hold `boot`, press `reset`, release `boot`" methods. Both yield similar results, the toolhead is unresponsive to the firmware and there's no `/dev/sd*` device.

Phil1988, 2025-07-04:
We talked about that already in discord but for others that may come across this thred: v2.00 has no KAF any more.

closed 2025-07-04 by Phil1988