Phil1988/FreeDi #148: [Bug] Fans stay on after print is completed

breathless19, 2025-02-09:
**Describe the bug**

When a print is completed, the hotend fan and filter fan stay on instead of shutting off.   I do not know yet if it happens with every print or only certain prints. 

**Expected behavior**
The fans should shut off after the print is done

**To Reproduce**
Steps to reproduce the behavior:
1. Print a file
2. When the print complete screen shows, fans are still on.  

**Screenshots or picture**

[image]
[image]

**Base (please complete the following information):**
 - Printer model: X-Max 3
 - X3seriesLCD firmware version: 1.40 stable
 - Source: FreeDi image
 - Mainboard: stock Mainboard

Phil1988, 2025-02-09:
Probably a LF error in your printer end macro. 

Can you share yours? 
Does your slicer call the PRINT_END macro?

Compare with this
https://github.com/Phil1988/FreeDi/blob/5237040676ab8f9626c36b2b40f4d70ad619a96f/config_section/X-Max3/macros.cfg#L104

breathless19, 2025-02-09:
As far as I can tell it should be turning off the fans...  Your print end section looks like mine:

#################################################
# Nozzle PID Tuning             
#################################################
# For more information, see https://www.klipper3d.org/Config_checks.html#calibrate-pid-settings
[gcode_macro NOZZLE_PID_TUNE]
description: "Perform PID tuning for the nozzle heater"
gcode:
    PID_CALIBRATE HEATER=extruder TARGET=220                                                                    ; PID tune the nozzle

#################################################
# Bed PID Tuning             
#################################################
# For more information, see https://www.klipper3d.org/Config_checks.html#calibrate-pid-settings
[gcode_macro BED_PID_TUNE]
description: "Perform PID tuning for the bed heater"
gcode:
    PID_CALIBRATE HEATER=heater_bed TARGET=80                                                                   ; PID tune the bed

#################################################
# Probe calibration              
#################################################
[gcode_macro PROBE_CALIBRATE]
gcode:
    CARTOGRAPHER_CALIBRATE

[force_move]

enable_force_move: False

#enable_force_move: True

    
########################################
# Basic Macros
########################################
[gcode_macro PRINT_START]
# Use PRINT_START for the slicer starting script																						   
description: Start printing
gcode:
    CLEAR_PAUSE

[gcode_macro PRINT_END]
gcode:
    {% set max_x = printer.configfile.config["stepper_x"]["position_max"]|float %}
    {% set max_y = printer.configfile.config["stepper_y"]["position_max"]|float %}
    {% set max_z = printer.configfile.config["stepper_z"]["position_max"]|float %}

    {% if (printer.toolhead.position.z + 12) < max_z %}
        {% set z_hop = 2  %}
        {% set z_safe = max_z - 10 %}
    {% else %}
        {% set z_hop = 0 %}
        {% set z_safe = max_z %}
    {% endif %}
    
    G91                                                                                                             ; Relative positioning
    G0 Z{z_hop} E-4.0 F3600                                                                                         ; Move nozzle up and retract filament
    G90                                                                                                             ; Absolute positioning
    G0 X{max_x/2} Y{max_y-40} F20000                                                                                ; Move nozzle to remove stringing 
    G0 Z{z_safe} F3600                                                                                              ; Lower bed 
    M400                                                                                                            ; Wait for buffer to clear
    G92 E0                                                                                                          ; Zero the extruder

    M106 P0 S0                                                                                                      ; Turn off part cooling fan
    M106 P2 S0                                                                                                      ; Turn off side fan
    M106 P3 S0                                                                                                      ; Turn off activated charcoal fan
    
    M104 S0                                                                                                         ; Turn off hotend
[... 292 lines cut ...]
[gcode_macro M204]
rename_existing: M99204
gcode:
    {% if params.S is defined %}
        {% set s = params.S|float %}
    {% endif %}
    {% if params.P is defined %}
    {% if params.T is defined %}
        {% set s = [params.P|float ,params.T|float] | min %}
    {% endif %}
    {% endif %}

    SET_VELOCITY_LIMIT ACCEL={s}
    SET_VELOCITY_LIMIT ACCEL_TO_DECEL={s/2}

########################################
# Macros for display functionallity
########################################
[gcode_macro INPUT_SHAPING_CALIBRATE]
description: "Perform Input Shaping calibration and save the results to printer.cfg"
gcode:
    SHAPER_CALIBRATE                                                                                            ; Step 1: Perform the Input Shaping calibration
    PAUSE                                                                                                       ; Step 2: Pause to allow the calibration to complete
    SAVE_CONFIG                                                                                                 ; Step 3: Save the configuration to printer.cfg
    RESTART                                                                                                     ; Step 4: Restart the firmware to apply the new settings

TopheC, 2025-02-09:
Similar behavior as @breathless19 here, but only the MCU Fan stay ON.
Same config :

    Printer model: X-Max 3
    X3seriesLCD firmware version: 1.40 stable
    Source: FreeDi image
    Mainboard: stock Mainboard

I noticed that the target Temp for MCU Fan stays at 45°C, even on machine cold start :

[image]

And the standby temp of my MCU is about 45°C +-2°, on my printer, and then the MCU Fan can run several times, several minutes, by an hour, even on standby.

Target temp can be set to 0°C (disabled) manually in Mainsail as workaround, but is 45°C a "normal" temp for MCU on standby, with ambient at ~19-20°C ?

Phil1988, 2025-02-09:
@breathless19 
> As far as I can tell it should be turning off the fans... Your print end section looks like mine:

Im not too sure about that.
Can you share the gcode where this happened (Top Cart Tall Ironed.gcode)?
Did the toolhead move to the mid back of your printer and the bed lowered?

btw: please use code tags next time ("```") 😉

@TopheC 
Similar behavior as @breathless19 here, but only the MCU Fan stay ON.

Then its not the same behavior because your fans (except the MCU fan) turns off 😉
MCU fan cools the CPU and is controlled via the CPU temp. 
Its doing what it should for you.

TopheC, 2025-02-09:
OK, my mistake about fans role.
That's better clear for me now. Thks ;-)

I haven't noticed this behavior (MCU fan running sometimes in standby mode) with the stock firmware...

Phil1988, 2025-02-09:
> I haven't noticed this behavior (MCU fan running sometimes in standby mode) with the stock firmware...

This is becuase on stock, the fan is only on, when the nozzle is hot.
You can decide if this approach is smart and how a hot nozzle is connected to the actual CPU temp 😉

breathless19, 2025-02-09:
> EDIT by Phil1988: please use code tags ;)

My bad, still learning the ropes :)

Here's the file as requested: 

https://drive.google.com/file/d/1nHNC7ObtirwwPhfE2Gmna7pBaVBNoKsH/view?usp=sharing

Phil1988, 2025-02-09:
@breathless19 

Took me 5min scrolling on my phone...

Well if you don't call the PRINT_END macro and don't turn off the fans from the slicer, how should FreeSi turn it off?

Please set it right in your slicer.

[image]

It's not a bug and I will close it.

closed 2025-02-09 by Phil1988

breathless19, 2025-02-09:
Sorry about this, feel free to delete this thread.   I thought the PRINT_END command was already in my slicer :/

Phil1988, 2025-02-10:
No problem mate :)
Thread is closed so all good.