Phil1988/FreeDi #149: Load filament / resume print after gcode pause

billkenney, 2025-02-14:
I've recently had two prints where I have inserted a pause in the slicer (Qidi slicer) at a certain layer. When that pause occurs, the freedi firmware doesn't recognize that the print has been paused. So the option to load filament or resume the print is greyed out. I've gotten around this with console commands like G1 E100 F200 to purge the old filament and load the new filament, and RESUME to resume the print, but figured I'd let you know about the issue. I'm running a Max 3 and am pretty sure I have all of your changes implemented to the configuration file. Aside from this one issue, your firmware and configuration changes have been great. Thanks for your contribution!

Phil1988, 2025-02-14:
Thanks mate for your detailed feedback.

You said that FreeDi didnt recognize the pause, but that the buttons where grayed out.
You can only enter that menu, if the pause is detected.
The reason why you were unable to use these buttons is very likely because the nozzle temp was not above 180°C.

So your observations were right, that something is not logical or the usability can be improved, but your assumption were a little bit off ;)

Ofc the Load/Unload macro should be able to be called and I will change that.
Thanks for that.
It will be improved in like 3 weeks when I am back.

billkenney, 2025-02-14:
Interesting that it has to do with the temp, I hadn't thought of that but it makes sense.

Another thing I've noticed is if I adjust the z offset at the beginning of a print it doesn't save unless I click save config in fluidd after the print. I'm wondering what the logic is behind updating the z offset, i.e., under what circumstances does the updated z offset save to printer.cfg, or is there a way other than clicking save config to keep the current z offset (aside from manually editing printer.cfg).

I've already updated it in the printer.cfg, so there's no rush in responding. Enjoy your time off.

Phil1988, 2025-02-14:
If you adjust the z-offset, you will get a "save z-offset" button on the print finished screen.

As a Klipper restart is needed for that, there is no other way and it follows the logic that Mainsail/Fluidd has.
I hope this works for you.

billkenney, 2025-02-14:
Yeah that works for me. I figured that's how it worked. Just wanted confirmation. Thanks again!

billkenney, 2025-02-18:
With regards to the save z offset button after a print, it doesn't work for me. When I tap it, the graphic shows the button has been pressed but nothing happens (it remains on the same screen) and the z offset isn't saved. 

Another issue I noticed, when I stop a print and press confirm (and the print does stop), the nozzle, bed, and chamber temperatures all remain whatever they were set at for the print. They should stop revert to 0 degrees.

Phil1988, 2025-02-18:
When you click the "Save z-offset" button it should show you something like this:
[image]

This does not pop up for you?

billkenney, 2025-02-19:
It looks like you made some changes in your macro configuration recently. I'll update my config and follow up if it still doesn't work. 

[image]

Phil1988, 2025-02-19:
I'll have a look into that in Detail in one week.

Lets not mix the issues here ^^
I created a new thread for that.

Phil1988, 2025-03-01:
This is included now.

Please update to v1.41 to get this 😉
https://github.com/Phil1988/FreeDi/releases/tag/v1.41

billkenney, 2025-03-01:
> This is included now.
> 
> Please update to v1.41 to get this 😉
> https://github.com/Phil1988/FreeDi/releases/tag/v1.41

Awesome!

billkenney, 2025-03-04:
> This is included now.
> 
> Please update to v1.41 to get this 😉 https://github.com/Phil1988/FreeDi/releases/tag/v1.41

So I updated and this worked for me once today. I paused the print for a color change, loaded the new filament, set the nozzle temp, pressed the load button (twice) and it worked as expected. It seems to become available when the nozzle is around 180 degrees. Unfortunately that print broke due to a poor design, so I am printing it again. This time I followed the same process, however, after loading the filament and pressing the play button (ended up pressing it six times) I noticed my print head moving to the left slightly every time I pressed the play button. And this was in the console six times:
G91 
G1 X-1 F4000 
G90

I looked at the macro and it seemed like it stores the location of the print head in case it is moved during pause so I entered RESUME in the console and it resumed without any issue. The only difference perhaps is that I was on a different screen when I pressed play, but I can't be sure.

billkenney, 2025-03-04:
I’ve also had this happen three times in the last 2-3 days. A print will randomly pause for no apparent reason. I had the same print pause for no reason twice and couldn’t find a pause command in the gcode. There is nothing about a pause in the klipper logs. 

The console says:
// idle_timeout: Timeout set to 43200.00 s

When I resume:
$ RESUME
// idle_timeout: Timeout set to 5400.00 s

At first I thought it was an anomaly, but it keeps happening.

Phil1988, 2025-03-04:
Thats "interessing".
Can you share your macros.cfg and your printer.cfg please?

I will review this tonight.
I think a F4000 is nowhere in the macros I provide, so I'm wondering if my memory is wrong if where this is coming from.

billkenney, 2025-03-04:
> Thats "interessing". Can you share your macros.cfg and your printer.cfg please?
> 
> I will review this tonight. I think a F4000 is nowhere in the macros I provide, so I'm wondering if my memory is wrong if where this is coming from.

I did a ripgrep (recursive grep search) for F4000 in printer_data, which returned nothing in the config or logs. I did get results from some of my gcode files, but they are different from what occurred yesterday, for instance: 
```
40205-G1 X113.139 Y155.759 E.04464
40206:G1 F4000.918
40207-G1 X112.822 Y155.427 E.03181
--
391905-G1 X222.517 Y208.704 E.03679
391906:G1 F4000.422
391907-G1 X223.167 Y208.951 E.04819
```

Here's a link to my printer.cfg and macros.cfg. They should be the same as your config with some additional macros. 
[https://filebin.net/y3845yjxn4o6ewkc]

Phil1988, 2025-03-05:
I had a look at them and didn't found anything substantially wrong.

It just me, but I don't think screw tilt adjust and input shaping should be located at the macros :D

Besides that there are a lot of additional includes and I don't understand why you have/need them at all.

Especially:
[include MKS_THR.cfg]
[include fluidd.cfg]
[include KAMP_Settings.cfg]

But you have also shell extensions and:
[include moonraker_obico_macros.cfg]
[include plr.cfg]
[include time_update.cfg]
[include timelapse.cfg]
[include chopper_tune.cfg]

So it's pretty complex on what is going on exactly.

### Getting back to track this down:
You said you pressed load button twice: 
Why?

It became available when the nozzle was above 180°C?
It should be available anytime now.

You said you hit the "resume" button 6 times.
I am sure this function was executed 6 times instead:
[image]

Can you please verify that you are on v1.41 at the screen (in the "about" page) and can you re-check this resume again?
There was a error on this menu (https://github.com/Phil1988/FreeDi/issues/125#issuecomment-2635144532), but I cant see any errors any more.

If you really touch the resume button (play sign) and instead of resuming it does a -X move, can you change the distance setting (the one left from the extruder picture) from 1mm to eg. 10mm and see if it acually is moving 10mm to the left?

billkenney, 2025-03-05:
> It just me, but I don't think screw tilt adjust and input shaping should be located at the macros :D

Seems to me that the printer.cfg and macros.cfg are the same. If you include an external file it's no different from adding what is in that file to printer.cfg. So I add to macros things that I may need to change in the future. I add to printer.cfg things that I don't think I will change in the future. 

> Besides that there are a lot of additional includes and I don't understand why you have/need them at all.
>
> Especially:
> [include MKS_THR.cfg]
> [include fluidd.cfg]
> [include KAMP_Settings.cfg]
>
> But you [...]

I've had all of these for 6+ months (except for chopper_tune which was maybe 2 weeks ago) and never had any issues. 

> ### Getting back to track this down:
> You said you pressed load button twice: 
> Why?

Because pressing the load button once doesn't seem to purge the new filament. If I press twice then I see the new filament come out of the extruder. 

> It became available when the nozzle was above 180°C?
> It should be available anytime now.

It isn't for me. It doesn't become available until after the nozzle is heated up to around 180.

> You said you hit the "resume" button 6 times.
> I am sure this function was executed 6 times instead:
> [image]

I use the settings (cog wheel) screen to load filament. It doesn't have the movement options on the image you posted. 

> Can you please verify that you are on v1.41 at the screen (in the "about" page) and can you re-check this resume again?

I'm in the middle of a big print (partially to test whether it will pause again), but I checked the fit logs and am on 1.41
[image]

> There was a error on this menu (https://github.com/Phil1988/FreeDi/issues/125#issuecomment-2635144532), but I cant see any errors any more.
>
> If you really touch the resume button (play sign) and [...]

I'll do a test print tomorrow and give this a try on different screens.

billkenney, 2025-03-05:
This was a pause for a filament runout, same as color change. Shows that the filament load doesn't become available until around 180. https://filebin.net/py3lp7jm8jpjj8wj

Phil1988, 2025-03-05:
> So I add to macros things that I may need to change in the future. I add to printer.cfg things that I don't think I will change in the future.

Whatever works for you the best. Its your printer :)

> > Besides that there are a lot of additional includes and I don't understand why you have/need them at all.
> > Especially:
> > [include MKS_THR.cfg]
> > [include fluidd.cfg]
> 
> I've had all of these for 6+ months (except for chopper_tune which was maybe 2 weeks ago) and never had any issues.

I'm not thinking they might be the cause (most likely not) but it was just too much to say for sure.
I'm still wondering what MKS_THR.cfg and fluidd.cfg are used for.
On FreeDi everything is nice and clean so these clusered configs were one goal to get rid off :)

> 
> > ### Getting back to track this down:
> > You said you pressed load button twice:
> > Why?
> 
> Because pressing the load button once doesn't seem to purge the new filament. If I press twice then I see the new filament come out of the extruder.

Gotcha! It works for me with this workflow: 
Inserting new filament all the way to the toolhead until I feel it hits the end.
Pressing the "Load" button.
It gets loaded and I can see its getting purged.
But if you change colors - especially from black to white it might not be enough.

If you constantly experience this, you can adjust the amount at the "LOAD_FILAMENT" macro.
I dont recommend hitting it twice because the first part of the filament gets charged quicker which may lead to stress to the extruder mechanics. Better use the "extrude" (down arrow) button in this case.

> > It became available when the nozzle was above 180°C?
> > It should be available anytime now.
> 
> It isn't for me. It doesn't become available until after the nozzle is heated up to around 180.

It would be handy if you could do a short video next time you test that.
You may have entered the hidden god mode menu (that Im not even aware of it existence 😅 )
A video will definetly help me understanding it.

> > Can you please verify that you are on v1.41 at the screen (in the "about" page) and can you re-check this resume again?
> 
> I'm in the middle of a big print (partially to test whether it will pause again), but I checked the git logs and am on 1.41 [image]

Not in a rush. I just want to make sure the screen firmware is also on v1.41 - so after that print totally works for me 👍

Phil1988, 2025-03-05:
> This was a pause for a filament runout, same as color change. Shows that the filament load doesn't become available until around 180. https://filebin.net/py3lp7jm8jpjj8wj

That is super strange.

Phil1988, 2025-03-05:
@billkenney 
While checking that, I run into a rare issue that is now pretty persistant.
This may sound bad but its actually good as I can and want work on this for quite some time.

I think this one is not critical for you (as you can workaround with resume) and im sorry for the delay on this.
Thanks for your understanding!

billkenney, 2025-03-05:
I took a video of the issue showing the printhead movement when the play button is pressed. It only occurs when you are on the screen with the load/unload buttons. When you are on the main screen the print resumes as expected. I tested with a color change pause and a regular pause and the behavior was the same. I also tested with 1 mm, 20 mm, and 50 mm. Here's a link to the video (I cut out like 15 seconds when the nozzle was warming up): https://www.youtube.com/watch?v=4UuHUcr4BLQ

Here's the output on the console:
$ G91 
G1 X-1 F4000 
G90
$ G91 
G1 X-20 F4000 
G90
$ G91 
G1 X-50 F4000 
G90

billkenney, 2025-03-05:
> [@billkenney](https://github.com/billkenney) While checking that, I run into a rare issue that is now pretty persistant. This may sound bad but its actually good as I can and want work on this for quite some time.
> 
> I think this one is not critical for you (as you can workaround with resume) and im sorry for the delay on this. Thanks for your understanding!
> 
> In the meantime: Please send me the the version of your tft firmware.
> 
> The issue I confirmed was actually at tft version 1.40, so Im not sure if its really still existent for me on v1.41:
> 
> [image]

I updated to tft version 1.41 and the resume function works normally on both screens. 

I wasn't aware a new screen firmware had been released, and I also thought new screen firmware would be automatically installed. I downloaded the release source code, copied the tft file to a micro sd card, and updated that way.

Phil1988, 2025-03-06:
Normally it should flash the tft automatically.

Can you please show the output of
```
ls -la ~/FreeDi/screen_firmwares
```

billkenney, 2025-03-06:
It has all the current versions. Does the FreeDi service handle the updates or is another service supposed to handle the updates?

rw-r--r-- 1 mks mks 14837288 Mar  1 18:01 x-plus3_x-max3_beta_v1.41.tft
-rw-r--r-- 1 mks mks 14837304 Mar  1 18:01 x-plus3_x-max3_stable_v1.41.tft
-rw-r--r-- 1 mks mks  8861544 Mar  1 18:01 x-smart3_beta_v1.41.tft
-rw-r--r-- 1 mks mks  8861560 Mar  1 18:01 x-smart3_stable_v1.41.tft

Phil1988, 2025-03-07:
The FreeDi software handles that when connecting to the screen.

As you have the latest TFT files in your system I wonder why it didn't update that.
The software v1.41 should have recognized the TFT being on v1.40 and auto update.

Next time if you have an issue like this, it can help starting the software with the debug option:
https://github.com/Phil1988/FreeDi/wiki/Bugs!

This way we can see the cause better.

We can continue the discussion if there is anything, but I will close this issue as it's solved.

closed 2025-03-07 by Phil1988