Phil1988/FreeDi #156: [Help]Cannot connect to printer after Update Through IP Address

66gtfastback, 2025-02-18:
If you are facing issues (especially during the first installation), please share important information to let me understand your current status.

This may include...

**How did you install it?** (Did you use the full image? Did it on your own? ...)
**What have you done?** (describe exactly please. step by step)
**What kind of messages/responses/feedback did you get?** (command line responses, error messages in the Screen/on Mainsail etc.)
And sometimes a **printer.cfg** is handy, which can be included here as renamed to *.txt or zipped.

In general: 
**Anything you think is helpful for me to understand your problem and how you got there :)**
I started by flashing the FreeDi software onto my stock X Max 3 Emmc. I inserted the emmc and the update installed and the new software booted to the Klipper screen. I plugged into the printer through the ethernet port but cannot connect using the old IP address and am unable to check the IP address of the printer now on the screen due to the update. I completely missed the backup section of the writeup and started here. Hopefully this has not killed my printer as I am unsure how to go back to stock to confirm my IP. Please let me know if there is anything I can do from here. Thank you.

Phil1988, 2025-02-18:
Please see the wiki section here:
https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems
and the video.

Its all step by step and I did the wrote up to not explain every step every time ;)
Seems like you missed a few steps already at the beginning.

Dont worry, your printer is not killed, you simply didnt followed the instructions.
First: Get your current printers IP (from your router or an IP scanner).
Next: follow the guide :)

We are here to help, but please try to follow the guide.

El-Keldo, 2025-02-21:
@66gtfastback 
Hi.
I got exactly the same behaviour when I flashed FreeDi over the original QiDi firmware : my printer recieved a new/different IP address.

WHY ?
Unless you gave a fixed IP to your printer, in most of the cases, the IP address is given to your printer at the end of the printer boot process by something called a "DHCP server" which runs on your local network.
This "DHCP server" can be part of the services delivered by your ISP "box", or by your local/home router/wifi box, or, in some more complex cases by a server (Windows / Linux / ...) inside your local network.
A DHCP server is using the MAC-address of the ethernet interface of the requesting device (here, your printer asking for an IP) to identify it.
Based on the MAC address of the requestor, a DHCP server tries to always give the same IP address to the same device, but it is not always possibl ; technically, with DHCP,  your printer could recieved a different IP address each time it has been rebooted.
Ethernet interfaces of PCI-E network cards (including the on-board ones on PC motherboards) have unique and "hard-written" MAC addresses, usually stored in an EEPROM, so they are stable and not supposed to be modified
ARM-based SBC's the mainboard of our printers are different, their MAC address is usually stored in the boot configuration settings of the boot partition, inside the microSD or the EMMC flash drive.
The problem we both faced is most probably due to the fact that re-flashing the EMMC of our printers also wiped the original MAC address given by QiDi and replaced it with something different value ; so, for the DHCP server on our networks, those re-flashed printers appear as new devices and, logically , recieved a different, new IP address.

WHAT NOW ?
As explained in the FreeDi documention, you need to find which device on your nework is actually acting as DHCP server / is distributing the IP addresses locally. This device should provide you a list of currently assigned / distributed IP addresses, with the MAC addresses of the devices and also the time when the lease begun. (IP addresses are not "given", they are only "leased" for a limited time, usually a few hours, by the DHCP server)
Just reboot your printer, then have a look at the refreshed IP list on the DHCP server and find the most recent lease : bingo, this is the the IP address of your printer.

66gtfastback, 2025-02-21:
Thank you both for the quick and helpful replies. For those who may have the same issue in the future, El-Keldo is spot on. I connected my printer via ethernet to the router and used the devices connected to my router (found by typing in the my router's IP into my search bar on Microsoft Edge. To find which IP my printer was as it was the only one connected through Ethernet.

I strongly recommend connecting your printer to your router with an ethernet cable prior to performing any update/modification to the software of the printer. You then can write down your ethernet IP address from your printer's network settings. (Hopefully it stays the same after flashing the emmc) 

My printer is now fully functioning minus the stock Qidi webcam. I downgraded Qidi Slicer to v1.1.6 as the latest versions will not allow me to connect to the printer using the correct IP due to different port issues in the software.

I have followed countless guides including the one included on how to get the Qidi webcam to connect but I am still getting no signal. I have Crowsnest installed into mainsail with the camera code information active in the crowsnest.conf file yet I cannot figure it out. Also is Qidi Link supported on FreeDi or not yet?

Phil1988, 2025-02-21:
@El-Keldo Thanks for supporting here 👍 

@66gtfastback 
Im glad that it was just as I said: A problem of finding your printers IP address :)

Please dont mix questions / issues.
Its just harder for others to find related questions later.

For the camera and not FreeDi related topics, feel free to open a discussion.
The "issue section" is not the right place :)

closed 2025-02-21 by Phil1988