Phil1988/FreeDi #157: [Feature request] FreeDiLCD set nginx password

chrizbee, 2025-02-18:
I added a nginx password according to this guide:
https://forum.vorondesign.com/threads/protect-your-klipper-printer-mainsail-password-ssl-firewall-rules.469/#-password

Everything still works - except the FreeDiLCD, which is stuck at "Connecting to Klipper".
As your FreeDiLCD main.py seems to be closed source, I can't really tell which changes I need to make to support the nginx password.

Can you help with that?
PS: In OrcaSlicer I simply use "username:password@ipaddress" which works fine.

Phil1988, 2025-02-18:
I cant help atm, but I will have a look into that, when I am back.

rosenrot00, 2025-02-26:
@chrizbee could you set the port in the freedi section in printer.cfg to 7125? With this setting please check again if the screen is working even though you have a password in place.

rosenrot00, 2025-02-28:
Feel free to reopen if this issue still remains after changing the port to 7125.

closed 2025-02-28 by rosenrot00

chrizbee, 2025-03-17:
@rosenrot00 Sorry for the late reply - I have tested your solution just now:
```md
# URL of the printer service
url: 127.0.0.1:7125
```

But it's still stuck at "Connecting to Klipper".

Phil1988, 2025-03-17:
@chrizbee Please show us the debug output by doing this:
```
sudo systemctl stop FreeDi.service
cd ~/FreeDi/FreeDiLCD && ~/klippy-env/bin/python3 start.py -debug
```

chrizbee, 2025-03-17:
Using `:7125` as port:
```md
# URL of the printer service
url: 127.0.0.1:7125
```

```bash
mks@mkspi:~$ cd ~/FreeDi/FreeDiLCD && ~/klippy-env/bin/python3 start.py -debug
Connecting to Moonraker at ws://127.0.0.1:7125:7125/websocket
Exception in Websocket run_forever: Port could not be cast to integer value as '7125:7125'
Using selector: EpollSelector
TFT baudrate detection...Detected baudrate: 921600
Display response: comok 1,1281-0,TJC8048X250_011R,165,10201,C0D73401CEA3F630,33554432-0
Display firmware: v1.42
Printer model: x-max3
Printer volume: None
Klipper version: None
Software version: v1.42 stable
FreeDiLCD start
FreeDi sw-version sent to screen!
pushing data to tft. Last update: (0.93 sec)
```

But after I updated to v1.42 (from v1.40) and removed the port...
```md
# URL of the printer service
url: 127.0.0.1
```

... the LCD could connect to Klipper again.
Sorry for the confusion and thanks for the update!

rosenrot00, 2025-03-17:
> Using `:7125` as port:
> 
> # URL of the printer service
> url: 127.0.0.1:7125
>  mks@mkspi:~$ cd ~/FreeDi/FreeDiLCD && ~/klippy-env/bin/python3 start.py -debug
> Connecting to Moonraker at ws://127.0.0.1:7125:7125/websocket
> Exception in Websocket run_forever: Port could not be cast to integer value as '7125:7125'
> Using selector: EpollSelector
> TFT baudrate detection...Detected baudrate: 921600
> Display response: comok 1,1281-0,TJC8048X250_011R,165,10201,C0D73401CEA3F630,33554432-0
> Display firmware: v1.42
> Printer model: x-max3
> Printer volume: None
> Klipper version: None
> Software version: v1.42 stable
> FreeDiLCD start
> FreeDi sw-version sent to screen!
> pushing data to tft. Last update: (0.93 sec)
> But after I updated to v1.42 (from v1.40) and removed the port...
> 
> # URL of the printer service
> url: 127.0.0.1
> ... the LCD could connect to Klipper again. Sorry for the confusion and thanks for the update!

So it works now even with your password set?

Phil1988, 2025-03-17:
> [@rosenrot00](https://github.com/rosenrot00) Sorry for the late reply - I have tested your solution just now:
> 
> # URL of the printer service
> url: 127.0.0.1:7125
> But it's still stuck at "Connecting to Klipper".

Only enter the URL _without_ the port.
For the port, enter it as the port ;)

Check if this works please