Phil1988/FreeDi #158: [Help] LCD Flashing / Boot up on v1.4 image does not happen

Ludn, 2025-02-18:
Tried an initial install on stock printer X-Smart3: FreeDi_v1.40 with new 32GB EMMC

**How did you install it?** 
 full image Way B: FreeDi_v1.40_X-Smart3_Armbian24.2.0_Bookworm6.6.17.img

**What have you done?**
Flashed image on new 32GB EMMC with makerbase USB stick, same PCB layout and marking as original:
flashes at ~40Mb/s, flashed file system also looks ok on PC:
 1st time with balena, 2nd try Win32DiskImager

**What kind of messages/responses/feedback did you get?** 
Qidi LCD boot screen as normal, after minutes: 'systems starts abnormal'
->Expected flashing process as shown on video, tried >10 times with long power down pause

EMMC access (LEDs behind) flashing stops 1..2s after power up. (But keeps on flashing when USB-Stick with image is connected, understood this is not needed)
No LAN access possible, no debugging :(

Printer still fully works with original EMMC (this i'll want to keep as backup).

Anything i can check? -Tried disconnect Cam, WiFi-Stick: no change.
(also tried putting head into boot mode, but not necessary at this stage, right?)

Diskimager 'Verify' reports after a while:
[image]
Can anyone confirm this ok?

Phil1988, 2025-02-19:
Hello and welcome to you!

I dont see that you are doing anything wrong.

My best guess is that image file was unpacked with errors or that your usb ports may cause a problem when writing.
Can you try to flash the EMMC on a different computer?
Normally the LCD getting flashed in the first 30-60s.
Long power down times are not needed.

Phil1988, 2025-02-19:
LEDs should look like this when its working:
[image]

Can you backup your stock 8GB EMMC and write this backup to the 32GB EMMC?
Just to make sure if you may have a faulty EMMC/Reader/PC issue.

Ludn, 2025-02-19:
Thx: i get 32GB EMMC to work by just reading original 8GB and write to new 32 GB EMMC.
Than, verify is also ok (Win32 Disk Imager).

Seems a compatibility issue of provided image-file to a non-8GB EMMC.
Any idea, like manual copying files to make it work on 32GB?

Phil1988, 2025-02-19:
It is a plain readout with Win32 Disk Imager based on a 8GB EMMC to not lock out all the 8GB EMMC users.

It is not a compatibility issue of the provided image file, as it runs on many 32GB systems without any issues.
The problem is definetly on your side.
It could be a download corruption or unpacking issue from *.rar to *.img.

Please re-check that - maybe with a different browser etc. 😊

Ludn, 2025-02-19:
Ja, TMC did un-rar *.img only half-sized! -rechecking with win-rar

Ludn, 2025-02-19:
Next, restarting screen came up and on LAN4 Printer is detected now.
But IP in browser doesn't show Mainsail, and no SSH connect possible, no matter which restart i do on LCD.

EMMC access looks kind of repeating twice flashing yellow LED, so system might be finished boot up: 'Underlying issue'

Phil1988, 2025-02-20:
Are you sure your IP is correct?

You need to know your IP address and I invite you to read my wiki or watch the video.
So far you are focusing the wrong direction of possible issue causes ;)

The whole process is very well documented and you can follow this step by step.

When you follow it, I recommend using the tools I recommend, because... you experienced now why :)

Ludn, 2025-02-20:
Thx, ja, IP is from router information, port according your documents a non-100xx.

I've done a complete Klipper Installation new from scratch on Sonic Pad with a self build printer quite easily, but here I am a bit stuck now. :)
Might need to read your manual process also and what's behind to get an idea.

Problem is that in your video and wiki, Mainsail is next step is intended just to work, and on my setup it's not, actually.
So if you can kindly point me to your well-known work, wiki- or source code position where I can look up more details to check myself what's blocking network access and how to debug.

What else can I check: is there a LCD debug possibility, should Head flashing as mentioned in your wiki work in this stage, etc.?

Phil1988, 2025-02-20:
Yes I used winrar as seen in my video.

Toolhead flashing will work, but it doesnt help you much as you need access to Mainsail to enter your toolhead ID.

There is nothing on the printers side to do/check.
You only need to 

1. flash the **non corrupted** *.img to the EMMC
2. insert that to the printers mainboard
3. connect an ethernet cable between router and printer
4. turn on
5. access via browser by entering the IP adress (no port specified).

There might be one case where this doesnt work:
Your network IP address is not in this range:
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16

But in any case a direct ssh connection via IP:22 (Port 22) will work (what you said is not possible for you).

So currently these are the only possible reasons that come to my mind:
- a corrupted img file (download/decompressing issue)
- a EMMC writing error
- a network/router error 
- ethernet wire is faulty
- you

If you show screenshots of your routers network or mesh screen, your browser while accessing the printer and a picture of the display when booted, this will help to guide you.

Ludn, 2025-02-20:
I appreciate your detailed help on this!

Quite an easy fix: LAN4 on Fritzbox was a guest network with limited rights.
However, makerbase USB flashing and verifying is not 100% kosher (eg. 1st run ok, 2nd not).

After that got Mainsail and ssh mks user to run right away. Thx again

Phil1988, 2025-02-20:
You are welcome mate !

I appreciate that you appreciate my help 😉

Closing it now and changed the title accordingly.

closed 2025-02-20 by Phil1988