Phil1988/FreeDi #159: [Bug] LCD does not display thumbnail correctly

teddeebare, 2025-02-18:
When a print job starts, the LCD display correctly shows the currently running print. However, if this print happens to be the 2nd print or more after a cold boot, the screen does not display the thumbnail correctly. See image.

- X-Max 3
- v1.40
- FreeDi image
- stock X-4 mainboard

[image]

Phil1988, 2025-02-19:
Thanks for the report.

Can you please share your gcode file?

teddeebare, 2025-02-19:
I also thought there maybe something wrong with the G-Code. Manually decoded the thumb and it is complete and correct. 

[Test.zip](https://github.com/user-attachments/files/18865581/Test.zip)

Phil1988, 2025-02-19:
Thanks @teddeebare 

I will check that in Detail when I am back (~1-2 weeks).
As it is a minor esthetic issue, I hope you can live with this meanwhile :)

teddeebare, 2025-02-19:
Yes. All good.

Your work is a genius effort. This is something minor and doesn’t affect operation.

Regards,
Glen

Phil1988, 2025-03-01:
@teddeebare 
I am fixing this now, but I have a question to your GCODE.
While analyzing, I saw this is start section:

```
M73 P0 R529
M117 Time Left 8h49m40s
M107
;TYPE:Custom
; custom gcode: start_gcode
PRINT_START
BEEP_ON
M141 S0
G0 Z50 F600
M190 S90
RESPOND TYPE=error MSG='Waiting 8 minutes for stability.'
G4 P480000
G28 Z
```

I am guessing that you modified your PRINT_START macro right?
Your BEEP_ON call would cause the printer to constantly beep from the time starting.

What is going on there? 😅

Phil1988, 2025-03-01:
@teddeebare
I fixed it.
Please update to the latest version and check if it still persists:
https://github.com/Phil1988/FreeDi/releases/tag/v1.41

Im closing this to get it off my radar, but please give me feedback in all cases :)

teddeebare, 2025-03-02:
Thanks for the update.

I’ll update the software when back in the office, 2 days time.

Yes, the start code is custom. On the original config from Qidi, the beeper never worked. Regardless of how it was used. I probed the board and noticed they had configured the beeper enable inverted. Simple change. Now the beep works. Don’t know if that was something unique with the revision of boards I had or not, but both machines were identical.

I updated my routines to create warning beeps when something is wrong, but that’s only needed when the machine is running. Hence I disable the beeper until it starts.

I’ll let you how it goes.

https://github.com/Phil1988/FreeDi/releases/tag/v1.41

https://github.com/Phil1988/FreeDi/releases/tag/v1.41

teddeebare, 2025-03-03:
@Phil1988, I have just updated and found/created an issue. I was reading through the posts on the release and followed the command line to update. I also followed the instructions to restart the system at the end of the update. But I did not realise that the screen had already started updating. Thus when I went to look at the machine, the screen was frozen. I powered off and restarted only to see the DATA Error ! notice. But the screen has now automagically updated after it being on for a while.

Now after a restart with screen working, I checked the G-code file already stored on the printer. The image attached is what was shown after it generated the thumbnail. Also, I have included the beginning of the g-code file for you to check. The whole file is 17 MB compressed. So only the relevant part included.

[image]
[G-Code.zip](https://github.com/user-attachments/files/19061783/G-Code.zip)

teddeebare, 2025-03-03:
Just to be sure, I uploaded another file fresh now that is completely never been used previously. The thumbnail issue remains. Looks like there is more to this bug than initially thought. Has no one else reported the issue? Could this be an issue related to hardware on my two machines?

Phil1988, 2025-03-04:
Thanks @teddeebare 
I'll check your gcode when I'm back home on my machine (tomorrow).
Im pretty sure its an allocating problem on the display side as your thumbnails are relatively detailed (with the bed rendered) and therefore big in size.

Can you check on the "About" page for the versions the display reports?

teddeebare, 2025-03-04:
Both versions 1.41.

That makes sense given the one thumbnail that is a smaller data size works.

Having the bed in the  background is pretty normal. I can easily turn it off, but it helps to have an idea of what parts should be where on the build plate from the image.

Is there a limitation on how big the images can be, pixel dimensions, etc.?

Phil1988, 2025-03-04:
There is a limitation but you can not do much there.
The thumbnail gets converted and sent to the display in a different format.

I'll check your files tomorrow.
Problem is we are already around the limit the picture can be.

Phil1988, 2025-03-06:
@teddeebare can you please check and show your tft version at the about menu on your display?

teddeebare, 2025-03-06:
@***@***.***>,

I won’t be back to the machine till next week. But I checked after the update,  the TFT showed v1.41 for the display and board. I didn’t get a pic, but I will when I return Monday.

Sorry.

Phil1988, 2025-03-08:
I tested your files:
1. [Test.zip](https://github.com/user-attachments/files/18865581/Test.zip)
2. [DVR Tray 4 Sets.zip](https://github.com/user-attachments/files/19061848/DVR.Tray.4.Sets.zip)
and 
3. [G-Code.zip](https://github.com/user-attachments/files/19061783/G-Code.zip)

1&2 do work now on v1.41:
[image]
Thumbnail size is 46800B.

3 doesnt work
[image]
It's 73800B and this is the problem.
The size is too big.
We are now at about the limit I can allocate.
There is no way I can increase the size another 54%.
So these super detailed thumbnails cant be supported.

There are a few options:
A: reduce the thumbnail size in the first row from the slicer 
    A1: disabling the bed preview
    A2: simplify the bed preview
    A3:  decreasing the resolution
B: FreeDi gets an integrated image resizer to reduce image size in these cases

B wont happen in the soon future as its a non critical function for very rare cases (only you 😅)

I will close this now as it's not a bug (and never was).
Thumbnails do work up to the limit the memory can take.
If you need this function, feel free to open a feature request 🙂

closed 2025-03-08 by Phil1988

teddeebare, 2025-03-08:
Thanks for your effort.

I am glad it isn’t a bug and not something greater causing issues for others.

I can live without the background and just the parts showing in the thumbnail. It’s my coworkers who don’t use the machines that often who rely on what the screen shows them.

You have done a marvellous job. I’ll disable the background in the slicer. Superslicer doesn’t have more granular control than dimension and format. I tried JPG instead of PNG, but as you said, you convert to what the screen can handle anyway.

Cheers
Glen

image.png (view on web)<https://github.com/user-attachments/assets/fe4571fe-49e4-4133-89fe-e5211bffb2f4>

image.png (view on web)<https://github.com/user-attachments/assets/1c6c2e94-7642-4587-acfc-cd164f662ad3>

image.png (view on web)<https://github.com/user-attachments/assets/fe4571fe-49e4-4133-89fe-e5211bffb2f4>

image.png (view on web)<https://github.com/user-attachments/assets/1c6c2e94-7642-4587-acfc-cd164f662ad3>