Phil1988/FreeDi #161: [Bug] Save Z-Offset button doesnt work

Phil1988, 2025-02-19:
> With regards to the save z offset button after a print, it doesn't work for me. When I tap it, the graphic shows the button has been pressed but nothing happens (it remains on the same screen) and the z offset isn't saved. 

 _Originally posted by @billkenney in [#149](https://github.com/Phil1988/FreeDi/issues/149#issuecomment-2664654136)_

Phil1988, 2025-02-19:
When you click the "Save z-offset" button it should show you something like this:
[image]

This does not pop up for you?

Phil1988, 2025-02-19:
From @billkenney

> Yeah nothing pops up, I tried pressing it like 10 times. I click the button, it shows the button is pressed, but it stays on the same screen and does nothing

Phil1988, 2025-02-19:
I'll have a look into that in Detail in one week.

billkenney, 2025-02-19:
I will say I am still using my image with the latest Qidi firmware (not the screen firmware), but I'm pretty sure that shouldn't interfere with your screen firmware. I've disable the makerbase-client (xindi) software, so aside from some random tweaks here and there it should be mostly the same. I update all of my software on a regular basis with no issue. If nobody else has this problem I'll install your image on my other emmc and see if it persists.

billkenney, 2025-02-19:
Actually rather than ask you to troubleshoot this, I will install your image on my other emmc and see if I have the same problem. The Qidi firmware is trash anyways and I don't need it. I'll post back if it happens after I've updated to your image.

Phil1988, 2025-02-19:
Thanks mate, that is very kind.

I'll have more time in about 1-2 weeks to dive deeper at reports.
Currently I cant even test if this is working on my image/firmware.
So if this issue still persists, I will wipe it then.

ChrisInMilwaukee, 2025-02-19:
As an observation on this, I noticed that with v1.4, when I do touch the "Save Offset" button, I do receive the Yes/No pop up as shown above.  After I touch "Yes", if I monitor the console it does, indeed, save the offset and restart Klipper.  But the front panel does not acknowledge the button push.  I have to touch "No" to leave the dialogue.  It did accept the changes, however, and it did restart the system as expected.

Phil1988, 2025-03-01:
Thanks @ChrisInMilwaukee for your feedback.
@billkenney I tested it today and it worked for me.

The bug that caused no forwarding to the print complete screen when pushing "Yes" is fixed now.

Please update to the latest version to benefit:
https://github.com/Phil1988/FreeDi/releases/tag/v1.41

closed 2025-03-01 by Phil1988