Phil1988/FreeDi #162: [Help] Print cancel doesnt work

Phil1988, 2025-02-19:
> I compared my cancel macro with yours and it's the same: 
> 
> ```
> [gcode_macro CANCEL_PRINT]
> rename_existing: BASE_CANCEL_PRINT
> gcode:
> 
>     {% set max_z = printer.configfile.config["stepper_z"]["position_max"]|float %}
> 	{% if (printer.toolhead.position.z + 12) < max_z %}
>         {% set z_hop = 2  %}
>     {% else %}
>         {% set z_hop = 0 %}
>     {% endif %}
> 
>     G91                                                                                                            ; Relative positioning
>     G1 Z{z_hop} F600                                                                                               ; Do a tiny z-hop
>     G90                                                                                                            ; Absolute positioning
> 
>     G1 X{printer.toolhead.axis_maximum.x/2} Y{printer.toolhead.axis_maximum.y-40} F12000
>     CLEAR_PAUSE
>     SDCARD_RESET_FILE
> 
>     PRINT_END
>     BASE_CANCEL_PRINT
> ``` 

 _Originally posted by @billkenney in [#149](https://github.com/Phil1988/FreeDi/issues/149#issuecomment-2667553204)_

Phil1988, 2025-02-19:
Can you please explain or show a video of what happens, when you click the stop button and confirm?
We have to track down, what is executed and what not.

Please also share your PRINT_END macro.

My best guess is, that there might be a LF error.

billkenney, 2025-02-19:
I'll compare the logs from my image and your image and see if there are any differences.

billkenney, 2025-02-26:
Print cancellation does work. It just doesn't reset the chamber, bed, or nozzle temps until the bed reaches the bottom. You can close this issue

Phil1988, 2025-02-27:
Thanks for the feedback,
I took this feedback to do some changes that will be published with the next release.

closed 2025-02-27 by Phil1988