Phil1988/FreeDi #164: Help

Bigsbie1, 2025-02-19:
Hello I installed FreeDi and love it so far. I have had to issues with my machine just shutting off, anyone have any idea what would cause this? I will link both the klippy files as well.

[klippy (2).log](https://github.com/user-attachments/files/18874670/klippy.2.log)
[klippy (1).log](https://github.com/user-attachments/files/18874669/klippy.1.log)

rosenrot00, 2025-02-19:
Hi, great that you joint the FreeDi community :)

Could you elaborate a bit more besides sending the logs? I see a very high definition bed mesh and some beacon entries. I guess you could elaborate a bit in the direction if your printer was working right after you installed it and what happened after you did certain things to the fresh installation.

Bigsbie1, 2025-02-19:
Yes sorry, the first time i was heating up my bed and chamber preparing for a print then it shutdown. Same thing with the second print. It has just happened 2 times so far. The printer seemed to be working properly as I got a few prints off before the first one and another 1 or 2 before the 2nd. Thanks for any information you can give to help.

rosenrot00, 2025-02-19:
So from the logs I can see that you have a x-max3 printer. I do have one as well, and mine doesn't shutdown. Thus, my question is, what have you changed so that it does crash on a regular basis.

How did you install FreeDi, did you flash the 1.40 image to the emmc? If so, was it working with the stock config that is provided?

rosenrot00, 2025-02-19:
I mean, don't worry we get it fixed but currently I don't know where the fault comes from. I guess that your beacon bed mesh settings crash the mcu but I don't know yet for sure. I would start setting the probe_count in the bed_mesh section to something reasonable like 15,15 or simply start with a fresh beacon installation with standard values. I believe that this would solve your issues - otherwise we will find the issue.

Bigsbie1, 2025-02-19:
I flashed the emmc with the provided file, the 1.4. I did the beacon install as fresh, and that was the best settings i got according to the users from the annex discord. I will try changing the settings on it to see if it will change anything. I was thinking if i put a raspberry pi in my printer instead of the emmc i wonder if this would help prevent the crashing?

rosenrot00, 2025-02-19:
In my opinion, issues should be solved in a structured manner. First, find the root cause and then identify an appropriate solution. 

Here someone has the beacon running on FreeDi seemingly without isses: https://github.com/Phil1988/FreeDi/discussions/121

Bigsbie1, 2025-02-19:
ok ill go look at what they have on there and post back in a few.

Bigsbie1, 2025-02-19:
Ok rosenrot, i have looked at the information and it seems that we are running very simialr on everything. My mesh is rougly the same. I did lower my mesh count. i am going to start the heatup process and see if anything happens.

Bigsbie1, 2025-02-19:
Ok it shutoff again, i might comment out all of the stuff like the beacon, and the btt filament sensor and try again

Bigsbie1, 2025-02-19:
the new log says something about webhooks request to reset, isn't that moonraker sending a shutdown signal?

Bigsbie1, 2025-02-19:
it done it again with all the previous mentioned commented out.

rosenrot00, 2025-02-20:
Could you confirm through the printer's web ui that all 3 components are on klipper v0.12 as shown here?

[image]

Phil1988, 2025-02-20:
Hello Bigsbie1,

please take your time to give a full report with important information and a meaningful title.

I dont like to pull pieces of information one by one as its very time consuming if everyone handles it that way.
I hope you understand that.

You want us to help you, so please try to make it easy for us 😊

Bigsbie1, 2025-02-20:
Hello again, sorry phil i didn't know the proper procedure when asking for help in here, thanks for the information. Rosenrot00 It seems my mcu didn't update to the .12 it is still on .10. How do i update this? I think i know how but i don't want to mess something up.

Bigsbie1, 2025-02-20:
I have it updated now sorry for the bother going to test it again, i thought i flashed to mcu but i didn't. Again sorry phil and rosen, i will give more information next time i have issues.

rosenrot00, 2025-02-20:
The important question is, does it work now?

Phil1988, 2025-02-20:
Im still wondering what this issue is about 😅

@Bigsbie1 
What have you updated?
Which MCU did you flash? On the Mainboard or on the beacon or what else?

I hope that it is solved for you now but I would like to know the cause of this problem and the fix.
This way I can organize this issue better.
Would be nice, if you can report the open questions back, when you finished your testings.

Bigsbie1, 2025-02-20:
> Im still wondering what this issue is about 😅
>
> [@Bigsbie1](https://github.com/Bigsbie1) What have you updated? Which MCU did you flash? On the Mainboard or on the beacon or what else?
>
> I hope that [...]

Sorry again, i updated toolhead etc, installed the beacon, I had forgot to update the mcu on the mainboard, done that now it seems to be working well. Again i am new to github, and I am sorry for not giving enough information in the start, next time i have an issue i will get all the information that i can so it isn't just a guessing game.

Phil1988, 2025-02-21:
Thanks @Bigsbie1 for clarifying.

Now I know what happened 😉

closed 2025-02-21 by Phil1988