Phil1988/FreeDi #167: [Bug] Print head XY position changes erratically while printing

El-Keldo, 2025-02-25:
**Describe the bug**
Since I installed FreeDi, I'm unable to have any correct print, they all fail during the first layer or when starting the 2nd layer.
The printer behaves normally while starting a print (homing, local bed mesh, heating up bed and nozzle), it then prints the first few lines or, at best, the full first layer , all this at normal speed.
Then, the head suddently accelerate at max speed and moves to a wrong XY location ( a few cms away), then speed comes back to normal and the head continues to print the next lines or the next layer +/- normally  but at a wrong location.
And it cycles like this every lines/few seconds/layer (depending on the model sent to print), randomly moving away to another XY location.

**Expected behavior**
Normal printing.

**To Reproduce**
Steps to reproduce the behavior:
1. Start the printer, load filament
2. Print any GCode, I even tested the "Fast Benchy 17m ..." demo file from the QiDi website.

**Screenshots or picture**
If applicable, add screenshots to help explain your problem.

**Console output**
If applicable, add the debug console output here (here is a [how to](https://github.com/Phil1988/FreeDi/wiki/Bugs!))

**Base (please complete the following information):**
 - Printer model: X-Max-3 
 - X3seriesLCD firmware version:  v1.40 stable
 - Source:  I went from from QiDi v. 4.3.15  to  FreeDi 1.40 image
 - Mainboard: stock  X4 v.1.1 mainboard, 32G emmc

Checks I made : 
- The printer was printing correctly with the QiDi firmware v. 4.3.15 and Klipper v. 0.10.
- Installed FreeDi 1.40 from the full image on Github, flashed on a new 32G emmc.
- I did a full compare between the original printer.cfg file that was working well on the QiDi firmware and the config file from FreeDi, adapted values when needed.
- Klipper firmware v.0.12.0-289 was correctly install on mcu (STM32F402) and print head (RP2040)
- It is NOT a "first layer adhesion" issue, all the failed prints are still firmly glued on the print plate when I cancel the print. Note : I'm using a degreased "Gold" PEI plate from QiDi with a thin layer of 3DLak ... everything stays nicely glued in place until the bed is back to room temp.

I also tested with two more recent version of Klipper  (0.12.0-429  and  0.12.0-439), upgraded via Kiauh , also upgrading both mcu firmwares, but the printer behaviour stays exactly the same.

Installed (via Kiauh) : 
[image]

"Machine"  page of Mainsail : 
[image]

I'm attaching both "printer.cfg" files (before FreeDi and after) and also 2 pictures of failed prints : 
- The QiDi fast benchy : here the first layer is partially printed on different locations, at least 
4.
- The rocket parts : the bottom circle and the "radioactive fan" have their full 1st layer normally printed (but z-offset could be better), and then the printer starts the 2nd layer on a wrong XY place (top circle, on the left). The second layer of the circle is still glued on the bed plate, even if it is visible that this second layer has been printed "in the air".

[image]
[image]

[Curerently used by FreeDi - printer.cfg.txt](https://github.com/user-attachments/files/18951573/Curerently.used.by.FreeDi.-.printer.cfg.txt)
[Original from QiDi firmware - printer.cfg.txt](https://github.com/user-attachments/files/18951572/Original.from.QiDi.firmware.-.printer.cfg.txt)

pmbroth, 2025-02-25:
It looks like zoffset is way off

El-Keldo, 2025-02-25:
> It looks like zoffset is way off

If you speak about the first layer of the circle and the fan, I agree, but I did MANY "z-offset" setting corrections (with the paper test) , using the printer screen and also using the marcro button in mainsail ... no changes, like if the z-offset value setting was not used by the printer.
But that is is not the first problem to fix ...   at the moment it is unusable , no matter the z-offset.

00stag, 2025-02-25:
After the Z-Offset setting with paper and the automatic print bed levelling, you should print something like the picture.

[image]

Then you can use the baby steps to adjust the height of the Z-offset during printing. Wait until the print is complete and then press save. Repeat this until you are satisfied. Normally 2 to 3 times.

[image]

[image]

[image]

rosenrot00, 2025-02-25:
Can we rule out mechanical issues (loose belts or pulleys) at this point?

Could you do a video so that we could see/hear the movement?

El-Keldo, 2025-02-25:
> Can we rule out mechanical issues (loose belts or pulleys) at this point?
Definitely yes : 
- The printer was printing correctly with the QiDi firmware
- With FreeDi, the printer behaves normally for many operations like homing, bed levelling, etc.

It looks like some specific G-code would be misinterpreted by the MCU and results into a sudden very fast move to another XY location, or that the relative XY origin is suddently corrupted.
 
> Could you do a video so that we could see/hear the movement?
I'll try to make one with my phone (the printer cam does not have a good angle to show the first layers correctly).
Where am I supposed to upload it ?
Which codec and resolution are preferred ?

By the way, thanks for making FreeDi (even if it is not usable on my printer at the moment) and for looking into my issue.

El-Keldo, 2025-02-25:
> After the Z-Offset setting with paper and the automatic print bed levelling, you should print something like the picture.
> Then you can use the baby steps to adjust the height of the Z-offset during [...]

Hi 00stag.
Thanks for the tip, but it is impossible to follow at the moment, as the printer is generally unable to finish a first layer without randomly jumping to wrong XY locations during the print. Look at the picture of the "QiDi fast benchy" : the first layer is completely de-structured.

pmbroth, 2025-02-25:
Did you follow all of the steps?  I used the precompiled files that Phil provided, try using those instead of compiling your own as you show above.

El-Keldo, 2025-02-25:
> Did you follow all of the steps? I used the precompiled files that Phil provided, try using those instead of compiling your own as you show above.

Yes, for Klipper v.0.12.0-289 I used the pre-compiled files in the FreeDi image , using SD-card flash for the STM32F402 and the "boot button" flash for the RP2040.  KFA did not work, even after 3 or 4 different reboots.
The picture showing the failed "rocket parts" print (the circle and "fan") in my initial post have been printed with Klipper 0.12.0-289 firmwares installed.

To flash v.0.12.0-429 on the mainboard MCU and tool head, I also used the SD-card and "boot button" method.

To flash v.0.12.0-439, I used different ways : 
- I used Kiauh "compile&flash" for the toolhead (RP2040) without the need to use the "boot" button.
- I manually used some parts of the KFA script to flash the X_4.bin on the STM32F402 (= pushing "firmware reset" on the MCU, stopping klipper service, flashing with "hid-flash my_firmware_file ttyS0" and restarting klipper service).
Both worked.

Side note : When flashing the STM32F402 with the SD-Card, my X_4.bin file has never been renamed X_4.cur , and I am quite sure KFA did NOT flash the MCU  ;  because, otherwise, it would have flashed the 0.12.0-289 version (hard coded in the KFA script) , and not my 0.12.0-429 version  ...

Maybe there is a different / older bootloader code on my STM32 MCU ?
I do not have the full history of my printer because I am only the second owner, but I have reasons to believe it is one of the more early models (initially came with BL-touch, strong magnets of the bed have been replaced by a magnetic pad, exhaust fan enclosure does not look like the one on the latest printer version, print head front and back covers were still the early version without extra little fans, etc.)

Note : 
I used to program PIC16/24/33 and have the tools for that.
I never worked with STM32 chips, but I do have one of those cheap chinese "USB ST-link V2 STM8/STM32 programmer" stick, so, with some guidence, I may be able make a dump of the STM32F402 chip, or re-flash a newer bootloader on it if needed.

rosenrot00, 2025-02-25:
@El-Keldo forget about the bootloader, as this part of the memory doesn't impact the operation of the MCU. If the version reported in the web ui changes after flashing (sd card or any other method), your MCU is flashed.

What I would suggest is to go back to the latest qidi firmware and report if the issue persists. This way we could tell if it has anything to do with the new klipper version or any config changes.

As you are a flash veteran now, this shouldn't take you long and we can proceed with further ideas once we know if the issue persists. What do you think?

El-Keldo, 2025-02-26:
OK, I'm now back on stock QiDi firmware, using my old 8G emmc and the klipper 0.10.0-530 MCU firmwares from : 
GitHub - FreeDi/old_mainboard_and_toolhead_firmwares/v0.10.0-530_old_stock/

The screen is not working at the moment, so z-offset is a problem, but using Fluidd I could home the printer and launch a fast benchy print ... and the first 15 layers of the benchy now stack up correctly, no more "destructuration" or "modern art" ;-)

:-/

El-Keldo, 2025-02-26:
It would be nice if someone could explain how to revert the screen firmware back to the QiDi version. ;-)

rosenrot00, 2025-02-26:
Very interesting I would say.

Did you use the plain FreeDi 1.40 emmc image to run the prints that didn't work out?

Reverting the TFT:
Extract that file on a fat32 sd card and insert it into the tft sd card slot just behind the tft. Turn on the printer and it will flash the firmware.

[800_480_MAX.tft.zip](https://github.com/user-attachments/files/18982078/800_480_MAX.tft.zip)

El-Keldo, 2025-02-26:
> Very interesting I would say.

I'm not too sure if I'm happy or not with what that means ...  :-)

OK, thanks for the screen firmware.

What I propose : 
- I first go back with a working screen on stock QiDi firmware
- I double check the printer is working as expected
- Then I wipe my 32G emmc , download the latest FreeDi 1.40 image again and re-install everything exactly as the guide says, using 0.12.0-289 mcu firmwares.
If the "destructuration" issue re-appears, then I'll do a M112 command and extract the klippy.log for review.

Phil1988, 2025-02-26:
I would say qidi stock is now proven to work with 15layers and you can start on a fresh FreeDi 1.40 system.

When you install it and run into any issues (especially at flashing the mcu), please get in touch with us before doing anything.

I totally believe that you are capable of finding a workaround, but thinking together about next steps could be beneficial :)

For the mainboard MCU, flash it the old style with the X_4.bin on a FAT32 SD card.
Verify it was flashed successfully via mainsail and check if it was renamed to *.CUR

Wouldn't be the first time I'm getting surprised by a special hardware peculiarity on the X3 series :D

So far you are the only one with this problem, so I'm very curious about the reason :)

pmbroth, 2025-02-28:
@Phil1988   Hello Phil   What do you mean by "I would say qidi stock is now proven to work with 15layers but another test won't hurt." Just for my own insight.  Thank you,   Brian

pmbroth, 2025-02-28:
Thank you Phil...

El-Keldo, 2025-03-03:
Hi all.
Sorry, I'm not always reactive, but I only have limited time available for 3D Printing.

So, here is the situation until now : 

I went back on QiDi version 4.3.15 by : 
- plugging the old 8Gb emmc card back on the printer
- flashing klipper 0.10.0 (from FreeDi github) on the print head, using the "boot" button push
- flashing klipper 0.10.0 (from FreeDi github) on the mainboard MCU using the SD-Card ... **and got the X-4 file renamed to ".CUR" after the flash**
- flashing the screen using a SD-card, using the "QD_Max3_UI5.0" file from the QiDi update 4.3.15, but renaming it "800_480_MAX.tft" (because the file previously provided by @rosenrot00 is for version 4.3.13)

I slightly modified the printer.cfg that was still on the 8GB emmc by using the fastest value in the "[printer]" section : 
```
[printer]
kinematics:corexy
max_velocity: 600
max_accel: 20000
max_z_velocity: 20
max_z_accel: 500
square_corner_velocity: 8
```
instead of the values that were previously used in my "QiDi firmware" config using v.4.3.15 
```
[printer]
kinematics:corexy
max_velocity: 300
max_accel: 5000
max_accel_to_decel: 2500
max_z_velocity: 10
max_z_accel: 250
square_corner_velocity: 8
```
I made this because I want to rule out any effect of the speed and/or acceleration settings on the issue.
The printer can now go up to full speed, like with the FreeDi default settings.
I aslo disconnected the webcam to remove any CPU overload influence.

Then I made some test prints : 
- Test 1 : fast benchy 17m from QiDi, using PLA --> Failed  : layer 1 and 2 printed OK, then (only) one erratic XY move , so entire layer 3 started to print to a wrong location, but all other layers printed correctly right above the layer 3. I stopped the print after about 25 layers.
- Test 2 : Rebooted (power off / power on)  the printer , then again the fast benchy 17m from QiDi, using PLA --> Full success, no errors,I have a nice benchy.
--> Here , I reconnected the camera and added the config in Fluid. Limiting to 5fps.
- Test 3, 4 and 5 : various PLA parts, using the default Generic PLA and 0.14mm layers HQ profile in QiDi Studio. Success, not a single error for more than 12 hours of printing , parts with 250+ layers.
- Test 6 and 7 : my first prints with ABS+, using Qidi studio with modified generic ABS and 0.14mm layers HQ profiles, heated chamber at 50°C. Success, even if one of the parts is ugly, this is due to the lack of filament calibration, the second part is quite nice and dimentionally perfect. I am very happy for my first try with ABS+ and, again, no error.

So, even if there is place for improvement (better "shake and tune", maybe double checking the belts tension, etc.) it seems that the stock QiDi firmware is working correctly.

What now ?
Maybe I could try some very high speed and high acceleration torture tests with the QiDi firmware, if you have any suggestion ?
Or do I go for a FreeDi 1.40 fresh install ?
In that case, I'll start again without the camera, to rule out any influence from there on the CPU.

Anyway, I'm really wondering why the X_4.bin file has been correctly renamed to X_4.CUR when flashing the 0.10.0 firmware, but never when flashing a 0.12.0 firmware.

El-Keldo, 2025-03-03:
... And I have no idea why the issue appeared only once on QiDi stock/0.10.0 after the re-flash and never again.

As illustration, I took a video while the printer was still using FreeDi 1.40 + Klipper 0.12.0-439 and printing the fast benchy from QiDi.
Find it here [https://keldermans.eu/VID_20250226_100618.mp4](url)
The issue starts @ 1:07 , you can hear a specific noise while the print heat moves very quickly to another XY location and, as my hands and the phone were on printer, the image becomes blurry for half a second each time it happens.

Phil1988, 2025-03-03:
The video was very important!

Ok quite a few point that I am wondering:

1. You said your stock `max_accel` was 5000. I also have a X-Max3 and my was at 20000 by default (on the stock system).
2. Your video shows a short stop at the purge line. This is suspicous to me. Might have been a very old slicer option. I also see that no thumbnail is integratted.
3. I can hear, that **your motors are skipping steps**. Its not be an "erratic" move but **maybe** your stepper / motor is different and cant handle these high acceleration values -> loses steps -> print looks bad.

**If that is the case it explains everything:**

- You had the problem at least once on your **"Test 1"** with the stock system (so it also happens there).
- It happens more often on FreeDi because FreeDi is simply faster in every way. It happens more often that your (maybe not capable) hardware get to /over it limits.

We have to make sure about this (you, by testing towards this theory), but if this is the case there are 2 solutions:

- Change your hardware (replace motors, steppers or mainboard)
- Decrease the acceleration to the point your hardware can handle it. Even if you lower it, you will benefit from FreeDi in regard to print time and quality.

I'm still wondering about this `max_accel: 5000`. 
**Can you also share your motor run current?**
 **When did you buy your X-Max 3**?

El-Keldo, 2025-03-03:
Quick answer, I'll come back with more later.

> **When did you buy your X-Max 3**?
I bought it used in november 2024.
I have no idea when the previous owner (first) bought it, but the printer initially came with the bl-touch system (later replaced).

>     1. You said your stock `max_accel` was 5000. I also have a X-Max3 and my was at 20000 by default (on the stock system).
I think the change from 20000 to  5000 has been done by the previous owner to get higher quality prints.
Anyway, the tests I made lately were all done with 20000, and I got only one issue at the start of the tests.
 
>     2. Your video shows a short stop at the purge line. This is suspicous to me. Might have been a very old slicer option. I also see that no thumbnail is integratted.
There is no thumbnail because I am printing the original G-Code from QiDi for their fats benchy, so it does not have the right thumbnail for FreeDi.
I suppose the short stop at the purge line is also part of the GCode of QiDi.
The original file is available here : https://drive.google.com/file/d/1RXvGRzcXcLWeGYBQKQ4IpFqCfFOw_C5_/view?usp=sharing

>     3. I can hear, that **your motors are skipping steps**. Its not be an "erratic" move but **maybe** your stepper / motor is different and cant handle these high acceleration values -> loses steps -> print looks bad.
> **If that is the case it explains everything:**
> We have to make sure about this (you, by testing towards this theory), 
Very interesting ... I need to dig into this.   What are the next steps ?

>     * Decrease the acceleration to the point your hardware can handle it. 
Sure this is something to be tested on FreeDi

> **Can you also share your motor run current?**
If it is in the printer.cfg, I shared the files in the first post.
Or is it something I can measure in some way ... then how ?

And, agian , thanks for helping me to troubleshoot this ;-)

El-Keldo, 2025-03-04:
Just in case, I just found someone with the same symptoms, but it seems the reason is not aligned at all with my case ...

Phil1988, 2025-03-04:
> I bought it used in november 2024 . [...] the printer initially came with the bl-touch system (later replaced).

So its one of the very first batches with some of the initial problems.
Does it have detachable stepper drivers on the mainboard?

> I think the change from 20000 to 5000 has been done by the previous owner to get higher quality prints [...] I got only one issue at the start of the tests.

Yes that might have been his workaround for this problem. 
Even if it happened "rarely" for you on the stock system: It happened.
And dont forget: the ~4 year old klipper on the stock system (0.10) is a lot slower.

> Very interesting ... I need to dig into this. What are the next steps ?

Clear answer: It depends 😄

We first have to know what exactly is the cause.
I guess its the stepper drivers.
If you answer the question about the detachable drivers we know more.

In general: We need to know as much as possible:

- Power supply voltage and power
- Stepper drivers
- Smoothness of the XY mechanism
- Maybe even motors
- etc.

> > **Can you also share your motor run current?**
> 
> If it is in the printer.cfg, I shared the files in the first post.

Yes its in that printer.cfg and I didnt check (and can't right now as I am on my mobile and scrolling on this is a pain).

For your other questions:
Skipping steps loses steps, you are right.
But are you 100% sure about all the movements the toolhead is supposed to do at any time?

**I think this is the case:**
It travel moves mith max acceleration -> looses steps there.
I can definetly hear that its skipping (cause is not 100% clear but this is my "experienced guess").
You cant see it, but maybe the head is trying to travel to the right, but doesnt actually move as the acceleration is too high and looses step at this. This is where we hear the skipping.
Then moves to the left  - which in this case works. 

You assume it simply went too far left, but it acutally may have tried to move to the right (but almost stayed at the spot) then moved to the left which worked.
The outcome is the same -> too far to the left, but the reason/cause is completely different.

Decrease the `max_accel` to say 3000 and I think it wont do these "erratic moves" any more.
This would be a test you should do to approach this further (besides the mentioned hardware information that would help).

El-Keldo, 2025-03-04:
> Does it have detachable stepper drivers on the mainboard?

Yes, it does. 
Please have a look at my main board here (I made the pictures) : 
https://openqidi.com/books/upgrades-updates/page/adding-usb-20-ports-on-the-x-max-3

Motor current values on my original printer.cfg were the same at the ones in the Freedi 1.40 printer.cfg : 

```
[tmc2209 stepper_x]
uart_pin: PD2
run_current:1.07
#hold_current: 0.5
interpolate: True
stealthchop_threshold: 0
diag_pin:^PB8
driver_SGTHRS: 85
#driver_SGTHRS: 130

[tmc2209 stepper_y]
uart_pin: PB9
run_current: 1.07
#hold_current: 0.5
interpolate: True
stealthchop_threshold: 0
diag_pin:^PC0
driver_SGTHRS: 85
#driver_SGTHRS: 145

[tmc2209 stepper_z]
uart_pin: PC5
run_current: 0.95
#hold_current: 0.6
interpolate: True
stealthchop_threshold: 1200
```

So, the plan : 
Step 1 ; collect all possible refs/model/etc. from the hardware
Step 2 : flash FreeDi 1.41 and klipper 0.12.0-289
Step 3 : test with downgraded speed and accel values, trying to find the max working values without issues.

Phil1988, 2025-03-04:
I'm not sure which stepper drivers these are, but you may benefit to use two TMC2209 like these:
https://a.aliexpress.com/_EwUdvwY

No guarantee at all that they might improve it or even work, but if I were you, I would give it a go for ~7 bucks.

El-Keldo, 2025-03-13:
Hello.

I finally found some time to look under the hood ...
So here is what I found inside my Max3:
 - Power supplies are 2x  MeanWell  LRS-450-24
 - Motors drivers are 3x replaceable TMC2209
 - For X and Y axis, the motors are BJ42D29-28V07
 - For the Z axis, the motor is a BJ42D29-22V08

The best link I found regarding the motors : https://fr.kelimotorgroup.com/stepper-motors/printer-hb-bj42d-stepper-motor.html

It looks like those motors are quite usual suspects, as they are known into the "motor_database.cfg" file of  "klipper_tmc_autotune" 

```
[motor_constants qidi-BJ42D29-28V07]
# used for X/Y-axis on QIDI X MAX 3
resistance: 1.4
inductance: 0.0026
holding_torque: 0.41
max_current: 1.5
steps_per_revolution: 200

[motor_constants qidi-BJ42D29-22V08]
# used for Z-Axis on QIDI X MAX 3
resistance: 4.7
inductance: 0.011
holding_torque: 0.6
max_current: 1.2
steps_per_revolution: 200
```

El-Keldo, 2025-03-13:
For information, the datasheet for the X-Y motor : 
[(BJ42D29-28V07)-drawing-1.pdf](https://github.com/user-attachments/files/19238247/BJ42D29-28V07.-drawing-1.pdf)

Phil1988, 2025-03-14:
I don't recommend going for the TMC2226.
Yes the heat dissipation can be a pro, but the dissipation design of the BTT drivers I linked minimizers that difference.

Other than that you will have to play around with the the differences like different driver_SGTHRS and such.

An improvement would be the TMC2240 because of the internal temp sensor, but it is not applicable for you as it needs SPI and the X-4 to X-6 Mainboard is only capable of UART.

But it's your decision.

El-Keldo, 2025-07-02:
I just installed the official stable FreeDi 2.00 on my X-Max 3.
Everything went smoothly, according to the documentation.

Very nice job from COco/Phil1988 !!!

I am currently using the untouched stock config built by the wizzard.
I did some auto tunings as expected by the install guide (bed leveling, PID tunings, z-offset, etc.).
The only "side" modification I made was to install spoolman via kiauh.
Speed and accel settings are untouched and I did not backport any settings from my old config files.

Until now, zero strange move, everything running as expected. Great !
Printed the included z-offset test file : no error ... but more fine tuning need to the offset --> done.
Tested the fast pre-sliced QiDi benchy ... printed in 17minutes as expected, no erratic move, good result. Even greater !

This is looking good for now.   :-))
I'll check in the coming days to make some useful prints and come back with results.