Phil1988/FreeDi #168: [Bug] (maybe) After changing the port in printer.conf to 10088 and restarting, the display does not come out of the start screen and hangs.

00stag, 2025-02-25:
After changing the port in printer.conf to 10088
and restarting, the display does not come out of the start screen and hangs. Back to port 80 and everything works again.
It is not possible to print directly via the Qidi Studio without port 10088.

Now you can ask yourself the question, why Qidi-Studio and not Orca-Slicer? Because the Qidi-Studio has the ‘.STEP-Import function, where you can set the “Linear Deflection and Angle Deflection”. This results in a much better look for round components.

Note: This is not absolutely necessary because you can of course also use gcode export. 

Does not hurry..

X-Plus 3 -stock hardware
FreeDi display v 1.40
FreeDi v1.30-85-g64e22fd0
Klipper v0.12.0-289-g08a1c9f1

rosenrot00, 2025-02-25:
which port do you intend to change? the freedi port setting is to connect to moonraker. Did you change the moonraker port accordingly?

00stag, 2025-02-25:
I have changed the port in the printer.cfg to 10088.

"Also, why are you not on FreeDi v1.40 if your screen is?"

I ask myself the same question. I have the 
FreeDi_v1.40_X-Plus3_Armbian24.2.0_Bookworm6.6.17.img on the EMMC, but the dashboard shows FreeDie 1.30..
Then I ran the terminal command you gave me yesterday:
‘rm -rf ~/FreeDi && cd ~ && git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh’
This also runs through but the version in the dashboard does not change.

rosenrot00, 2025-02-25:
For moonraker to work on that port you would have to change that first, right?

Did you refresh the update manager after running the command?

00stag, 2025-02-25:
"For moonraker to work on that port you would have to change that first, right?"

Correct. I have changed this in the printer.cfg, do I need to change anything else, or in another file?

"Have you updated the update manager after executing the command?"

No, I didn't, but I've done it now. Sorry ;-(
Now the version is set to v1.40-2-g52370406

rosenrot00, 2025-02-25:
I‘m not a printing expert but I thought you would have to change the moonraker port in moonraker.conf.

Can you connect to the 10088 port with whatever application you want to without changing moonraker.conf?

I‘m very surprised that the application doesn‘t let you change the port in first place.

pmbroth, 2025-02-25:
I had tried qidi studio, it would not find the printer, so I use orca slicer.

00stag, 2025-02-25:
Exactly, but the custom firmware for QIDI should be compatible with the QIDI software. Orca works, QIDI-Slicer works. But I described above that the QIDI Studio has the .STEP import function and this is a game changer as far as the quality of round objects is concerned. In my opinion.

pmbroth, 2025-02-25:
qidi studio won't work because Qidi hard coded a port for the printer, which is a terrible idea.  All other slicers Will work.. Look At orca slicer 2.3beta to see if they have that function your needing.. I use it,   Orca slicer does import step files

[image]

pmbroth, 2025-02-25:
what exactly are you looking for, I imported a step file into orca slice 2.3

00stag, 2025-02-25:
Yes, the Orca slicer does this, as do many other slicers. The point is, the QIDI-Studio recalculates the ‘Linear Deflection and Angle Deflection’ according to your requirements when you drag a .STEP into the slicer workspace. This leads to a finer resolution of the arcs in the print file. You are right, this is hard coded in the studio, but the open cliff should offer the possibility to adapt the port to the conditions of the QIDI software.

pmbroth, 2025-02-25:
you may want to try prusa slicer

pmbroth, 2025-02-25:
There may be additional functions, again look at orca slicer 2.3, as it has so many new features. Orcaslicer is based on Pruissa as well

00stag, 2025-02-25:
The question is, how and where does the clipper/moonraker have to be configured for the QIDI-Studio to work?
and the display of the FreeDi does not freeze when using the Studio.

The point is, the QIDI-Studio recalculates the ‘Linear Deflection and Angle Deflection’ according to your requirements when you drag a .STEP into the slicer workspace.

Try it out. Do the other slicers have this too?

pmbroth, 2025-02-25:
[image]

this is from bambu stuudio

I used this before with my qidi x-max 3, you may want to try that as well

00stag, 2025-02-25:
The BS work with FreeDi? Then i will tray it.

00stag, 2025-02-25:
But the studio should still work with FreeDi. It's just not nice to constantly switch between slicers when you have a lot of projects, don't you think?

pmbroth, 2025-02-25:
I dont use bambu studio anymore, I liked orcasliscer better.  Again, qidi hard coded the port, rather than just the IP. I can use simplify3d , as they don't expect a given port.

rosenrot00, 2025-02-25:
I still don’t get the initial problem. changing the port setting in the freedi section in printer.cfg wouldn’t impact any port needed by any slicer. 

So where is the issue with freedi? of course if you change the port in freedi section without changing the moonraker port the display wouldn’t show anything. neither would a slicer work on the 10088 port.

pmbroth, 2025-02-25:
QIdi did a lot of things wrong with their software configurations...

pmbroth, 2025-02-25:
its an issue with qidi, hard coding their software to look for a  port...

colin-small, 2025-02-25:
Qidi studio is based on Bambu studio.. so it should have the same option you are looking for.

pmbroth, 2025-02-25:
Bambu is based on Prussia.   :-)

00stag, 2025-02-25:
Then I ask for a short instruction on how to correctly configure the access of the QIDI-Studio on port 10088 on the FreeDi/Klipper system.

> Am 25.02.2025 um 22:22 schrieb rosenrot00 ***@***.***>:
> 
> changing

pmbroth, 2025-02-25:
I just instakleld qidi studio, and all works after I setup the machine for my ip and my poort of :80.  The only issue is the the device tab doe snot work, nbut I was bale to send a job perfectly.  Whule it wa sprirntign I went into Orcalsiecer to look athe machine

[image]

colin-small, 2025-02-25:
I have the same experience. I can use Qidi studio to print, but the device tab doesn't work.. I don't know why. I had a similar issue in Qidi slicer.

pmbroth, 2025-02-25:
its because they hard code the ports in the device area...   But this is a work around if  ‘Linear Deflection and Angle Deflection’ functions are needed and cannot be found in the other slicers by another name.

rosenrot00, 2025-02-25:
> Then I ask for a short instruction on how to correctly configure the access of the QIDI-Studio on port 10088 on the FreeDi/Klipper system.
> […](#)

I would say you have to change the moonraker port. The instructions are found on google.

If that makes your slicer work we can check for freedi afterwards

00stag, 2025-02-25:
> pmbroth

Can we please stop discussing which slicer or whatever? The point here is: QIDI Studio does not work with FeeDi. This problem should be solved, nothing else in this enquiry:

rosenrot00, 2025-02-25:
> Can we please stop discussing which slicer or whatever? The point here is: QIDI Studio does not work with FeeDi. This problem should be solved, nothing else in this enquiry:
> […](#)

100% what I would suggest. So please check if the moonraker port is what is needed. this isn’t freedi related at least from what I know until now.

pmbroth, 2025-02-25:
got this from searching:
To configure the Moonraker printer port, access the "moonraker.conf" configuration file on your system, usually located in the "printer_data/config" directory, and modify the "port" value within the "server" section, which defaults to 7125; this setting defines the port on which your Moonraker instance will listen for connections from your slicer or other applications

00stag, 2025-02-25:
"got this from searching:
To configure the Moonraker printer port, access the "moonraker.conf" configuration file on your system, usually located in the "printer_data/config" directory, and modify the "port" value within the "server" section, which defaults to 7125; this setting defines the port on which your Moonraker instance will listen for connections from your slicer or other applications"

 I have tried this, but no success.

rosenrot00, 2025-02-25:
so I checked it on my printer and this should be doing it (revert any changes made to moonraker):

edit this file and change the port at the top:

`nano /etc/nginx/sites-available/mainsail`

then call `sudo systemctl restart nginx`

El-Keldo, 2025-02-26:
Maybe a totally crazy idea but ... : 

Original QiDi firmware uses Fuidd on port 10088, FreeDi uses Mainsail on port 80 ;  why not install both Fluidd and Mainsail on FreeDi, via Kiauh, and then configure Fluidd to use port 10088 (and keep Mainsail like it is today on port 80 ?
Would this kind of setup work ?
Because, if yes, then QiDi studio would work as expected and I suppose FreeDi and the new screen firmware too.

But I have no idea if Mainsail  and Fluidd can work at the same time without ill effects.

rosenrot00, 2025-02-26:
I would say it isn't needed. Once the port for web access is set to 10088 the slicer will work and then freedi can be set to port 7125 which is the standard moonraker port. Everything will work this way.

00stag, 2025-02-26:
> I would say it isn't needed. Once the port for web access is set to 10088 the slicer will work and then freedi can be set to port 7125 which is the standard moonraker port. Everything will work this way.
> 
The connection from QIDI-Studio works, but on my x+3 the display restarts when you press the STOP button on the display, for example.
After restarting the display, it then hangs.
Can anyone confirm/test this?

rosenrot00, 2025-02-26:
Which stop button? stop printing or emergency stop?

Please share as much detail as possible or even better record a short video so that I know exactly what you mean.

00stag, 2025-02-26:
Stop button, while printing. Also occurred when I wanted to switch off the housing light.

This topic is not urgent from my point of view. Basically, I can work well with FreeDi. Of course I can also work with OrcaSlicer and I know that you developers have a lot of work with the development of FreeDi. Test the compatibility with QIDI-Studio if your time allows it and fix the urgent things first. We can leave it as it is and when you have found a solution I will be happy to continue testing.

pmbroth, 2025-02-26:
I press stop button on the screen, and all is fine.  I do not have any issues.

pmbroth, 2025-02-26:
I have mainsail and fluid nstalled - Mainsail I have at port 80 and fluid at port 85 - both work perfectly.

00stag, 2025-02-26:
So all back to zero.
All ports back to 80.
Fluidd installed on port 10088
reboot
DIDI-Studio finds the printer
Print file sent to the printer with ‘upload and print’
Printer starts printing.
When touching the display it crashes.

rosenrot00, 2025-02-26:
Can you run the following and post the output once the crash happens?

`sudo systemctl stop FreeDi.service
`
`cd ~/FreeDi/FreeDiLCD && ~/klippy-env/bin/python3 start.py -debug`

00stag, 2025-02-27:
Connecting to Moonraker at ws://127.0.0.1:80/websocket
Using selector: EpollSelector
TFT baudrate detection...Websocket connected
Moonraker connection opened at ws://127.0.0.1:80/websocket
Starting new HTTP connection (1): 127.0.0.1:80
http://127.0.0.1:80 "GET /server/files/metadata?filename=H%C3%BClse-Bel%C3%BCftungsseite-neu.gcode HTTP/11" 200 None
G-Code file path is: ~/printer_data/gcodes
Found z_offset = 1.139 in variable "z_offset" in section "probe"
Detected baudrate: 921600
Display response: comok 1,1281-0,TJC8048X250_011R,165,10201,0FB1340175502444,33554432-0
Display firmware: v1.40
Printer model: x-plus3
Printer volume: 280x280x280
Klipper version: v0.12.0-289-g08a1c9f12-dirty
Software version: v1.40 stable
FreeDiLCD start
FreeDi sw-version sent to screen!
self.lcd.data_update()
printjob.name.txt="Hülse-Belüftungsseite-neu.gcode"
data.klipper_state != self.printer.klipper_state
klippy ready
Printer state: printing
Ongoing print detected
extract_printjob_values: self.printer.file_name: Hülse-Belüftungsseite-neu.gcode
http://127.0.0.1:80 "GET /server/files/metadata?filename=H%C3%BClse-Bel%C3%BCftungsseite-neu.gcode HTTP/11" 200 None
Exception in thread Thread-4 (periodic_update):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "build/main.pyx", line 289, in main.FreeDiLCD.periodic_update
  File "build/main.pyx", line 280, in main.FreeDiLCD.update_lcd
  File "build/lcd.pyx", line 988, in lcd.LCD.data_update
  File "build/main.pyx", line 924, in main.FreeDiLCD.lcd_callback
  File "build/main.pyx", line 461, in main.FreeDiLCD.extract_printjob_values
KeyError: 'filament_type'

Phil1988, 2025-02-27:
Your latest output shows that its not a problemwith the port per se.
It's a known bug from QidiStudio that does something wrong with the `filament_type` (reported multiple times in discord).
Please show interest on fixing it here:
https://github.com/QIDITECH/QIDIStudio/issues/22

Still waiting for Qidi to implement custom ports for their slicer:
https://github.com/QIDITECH/QIDIStudio/issues/60
Feel free to show your interest there.

We included a workaround and will check why it doesnt work for you.

00stag, 2025-02-27:
Well, that doesn't work for me. Can you please provide me with the original x-Plus3 firmware for the mainboard, toolhead and display?

pmbroth, 2025-02-27:
https://github.com/Phil1988/FreeDi/discussions/73

rosenrot00, 2025-02-27:
> Well, that doesn't work for me. Can you please provide me with the original x-Plus3 firmware for the mainboard, toolhead and display?

Solved.

Run `cd ~ && rm -rf FreeDi && git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh`

00stag, 2025-02-27:
Ok, i test it tomorrow.

00stag, 2025-02-28:
unfortunately no

cd ~/FreeDi/FreeDiLCD && ~/klippy-env/bin/python3 start.py -debug
Connecting to Moonraker at ws://127.0.0.1:80/websocket
Using selector: EpollSelector
TFT baudrate detection...Websocket connected
Moonraker connection opened at ws://127.0.0.1:80/websocket
G-Code file path is: ~/printer_data/gcodes
Found z_offset = 1.139 in variable "z_offset" in section "probe"
Detected baudrate: 921600
Display response: comok 1,1281-0,TJC8048X250_011R,165,10201,0FB1340175502444,33554432-0
Display firmware: v1.40
Printer model: x-plus3
Printer volume: 280x280x280
Klipper version: v0.12.0-289-g08a1c9f12-dirty
Software version: v1.40 stable
FreeDiLCD start
FreeDi sw-version sent to screen!
Exception in thread Thread-4 (periodic_update):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "build/main.pyx", line 300, in main.FreeDiLCD.periodic_update
  File "build/main.pyx", line 238, in main.FreeDiLCD.update_lcd
AttributeError: 'PrinterData' object has no attribute 'klipper_state'. Did you mean: 'klipper_status'?

rosenrot00, 2025-02-28:
I will fix that within an hour.

rosenrot00, 2025-02-28:
Could you run `cd ~ && rm -rf FreeDi && git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh`

Let me know if it works.

00stag, 2025-02-28:
That's much better. Display works.
Printing from QIDI-Studio works. (2 components tested)
Buttons work.

There is a warning:
Klipper runtime warning
MCU ‘mcu’ configured for 84Mhz but running at 83Mhz!

I will do more tests later.

Phil1988, 2025-02-28:
@00stag 
Please give more information:
Where do you get that (I assume at the bell on mainsail) -> please show a screenshot.

Please share your klippy.log :)

rosenrot00, 2025-02-28:
I would say we close that one because the issue is solved. If ever the MCU warning comes back let's have a new issue.

00stag, 2025-02-28:
ok, close it.

closed 2025-02-28 by rosenrot00