Phil1988/FreeDi #169: After hard recovery errors reported

elmo71, 2025-02-26:
Everything was perfect for a while.
As a routine, after checking the update Klipper showed invalid. I tried soft recovery, but it did not work, after hard recovery I can not get Klippy to work:

Here are the last log entries:

Config error
Traceback (most recent call last):
  File "/home/mks/klipper/klippy/klippy.py", line 130, in _connect
    self._read_config()
  File "/home/mks/klipper/klippy/klippy.py", line 127, in _read_config
    pconfig.check_unused_options(config)
  File "/home/mks/klipper/klippy/configfile.py", line 305, in check_unused_options
    raise error("Section '%s' is not a valid config section"
configparser.Error: Section 'freedi' is not a valid config section
webhooks client 281473838619472: New connection
webhooks client 281473838619472: Client info {'program': 'Moonraker', 'version': 'v0.9.3-0-g71f9e67'}

rosenrot00, 2025-02-26:
After a hard recovery of klipper you have to run install.sh again to get freedi going again. This is the case for every addon that copies files in the klipper folder to interact.

elmo71, 2025-02-26:
Thnak you, so I did and backup and running again! Thank you!