Phil1988/FreeDi #171: [Help] How to revert back to Qidi Stock firmware

dgauche, 2025-02-27:
If you get stuck somewhere in the middle of the upgrade process or just want to revert back to the stock system, how would you do that on an X-Max 3?

colin-small, 2025-02-27:
https://github.com/Phil1988/FreeDi/discussions/73

closed 2025-02-27 by dgauche

Phil1988, 2025-02-27:
@dgauche You can also tell us where you gut stuck.
I am 100% sure we can guide you through it :)

dgauche, 2025-02-27:
Thanks Phill,

I am having issues with the webcam.  When I run KIAUH I just get a cursor and no options to select anything.

Great work btw.

dgauche, 2025-02-27:
I just tried after leaving the printer off for a few hours and it worked... wierd.

Phil1988, 2025-02-28:
Im glad that it worked, and welcome to the familiy ;)

I wish to have more time on exanding the wiki on those topics but right now I need to focus on improving the project itself.
Its still on the to do list tho.

In your case a simple installation of crowsnest is needed.
Then the webcam has to be included to the crowsnest.cfg.
Sometimes a restart/reboot is needed to proberly start all the services (that might have been the case for you).

If you are having any problems in the future, get in touch with us.
Fastes way is likely discord:
https://github.com/Phil1988/FreeDi/wiki/Community